pbatard/rufus #1999: Where do I see the option to remove TPM/Secure boot on version 3.19?

mrjayviper, 2022-07-26:
Supposedly these options only show up when "These customization options are only proposed when using a Windows 11 image."

But I selected Win11 21H2 and 22H2 ISOs and no dialog was presented to me.

a bug maybe? thanks

thefirethirteen, 2022-07-26:
Same issue here, even after downloading an iso through rufus.

Also, this is funny, but I literally saw this issue 37 seconds after you posted it :).

pbatard, 2022-07-26:
That's because you're not following the checklist and chose not to look at the [relevant entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Help_I_no_longer_see_the_option_to_bypass_TPMSecure_Boot_with_Windows_11). It's really annoying that the very people who think the checklist should not apply to them are the ones who very much should follow it to get their problem solved.

closed 2022-07-26 by pbatard