pbatard/rufus #2002: Fedora Rawhide (F37) ISOs not detected as BIOS/CSM bootable

marcosfrm, 2022-07-27:
Fedora Rawhide recently ditched Syslinux from its ISOs in favour of GRUB 2.

https://github.com/weldr/lorax/pull/1226

https://dl.fedoraproject.org/pub/fedora/linux/development/rawhide/Server/x86_64/iso/

The new ISOs are BIOS/CSM bootable, but Rufus does not think so:

```
Rufus x86 v3.19.1911
Windows version: Windows 10 Pro, 64-bit (Build 19044.1826)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Fedora-S-dvd-x86_64-rawh'
  Size: 650.8 MB (Projected)
  Uses: EFI
Using image: Fedora-Server-netinst-x86_64-Rawhide-20220725.n.1.iso (661.3 MB)
```

It is still early in Fedora 37 development... do not know if more changes will come. FYI.

pbatard, 2022-07-27:
Thanks for the report. I'll have a look when I get a chance...

closed 2022-07-28 by pbatard

marcosfrm, 2022-07-28:
269 downsteam patches... pfff

commit eda1f59a (2022-07-28): [iso] fix GRUB version detection for Fedora Rawhide

* How nice of "Open Source proponent" IBM/Red-Hat/Fedora to fix double space typos while making sure the
  provenance of the software they are using is hidden:
  https://src.fedoraproject.org/rpms/grub2/blob/rawhide/f/0024-Don-t-say-GNU-Linux-in-generated-menus.patch
* Long story short: Fedora fixed the double space in "GRUB  version", but of course they didn't upstream
  this change since it is part of a patch that removes every possible mention of GNU. This made our GRUB
  version detection break, since it relies on finding a "GRUB  versio [... cut]