pbatard/rufus #2003: no x64 release?

zedd-0, 2022-07-29:
hi,
i discovered a native win64-x86 version in the microsoft store.
sadly, it is not released on github. :(

i want create a automatic downlad script for all 4 versions.
(arm, arm64, x86, x86-64)
sadly there is no link to the win64-x86.

why is that so?
can you change it, please?

i create myself an automatic generated file archive of important apps with update option.

pbatard, 2022-07-29:
> why is that so?

Because the cost/benefit ratio of maintaining a separate 86_64 download (as well as answering questions as to what versions should most people use, when the x86_32 will do just fine) is simply too high.

> can you change it, please?

Not planning to at this stage. If you __absolutely__ want an x64 version, just use the App Store version.

closed 2022-07-29 by pbatard

zedd-0, 2022-07-29:
you use FossHub, too.
is there no free space to put the x64 version on?

pbatard, 2022-07-29:
It's not a matter of space. It's a matter of __time__.
It takes __time__ to maintain an extra download, whereas the demand for it is exceedingly limited, and there's very little reason for most people to use an x86_64 version (apart from irrational demands of _"But I must have it!!!"_).

My time being limited, it is my choice to dispense with elements that will take parts of it that I'd rather use for other things.

zedd-0, 2022-07-29:
than remove the arm(32) version
this will nobody ever use.

i see other github project with tonnes of links.
like i do, can you not automatic this process with a upload script.
i thik the githib-api can do this.

"thake time now, save time later"
does not windows-store cost time, too?
or is this process easier than github?

pbatard, 2022-07-29:
Please explain exactly how you __must__ have an x86_64 version of Rufus and why you cannot use the x86_32 version?

pbatard, 2022-07-29:
> than remove the arm(32) version

If I remove it, ARM32 users don't have a native version that runs on their CPU.
If I don't produce an x86_64 version, and because x86_64 is compatible with x86_32, x86_64 users do have a native version that runs on their CPU.

>  can you not automatic this process with a upload script?

Not for a file that runs elevated and therefore needs to be signed.

zedd-0, 2022-07-29:
because windows is drifting to, x64.
windows 11 do not have an x86 version.
i know the compatibility layer, but when it will be disabled by microsoft?

i will not that you remove arm32,
but are there any people left, which they use that slow SoC?

pbatard, 2022-07-29:
> but when it will be disabled by microsoft?

And when that I happens, I will switch to procuding a single x86_64 version of Rufus with no x86_32. That's the plan, so that I don't have to waste time with 2 x86 versions.

I can __save__ time by limiting the number of x86 versions of Rufus I produce. And it is still too early to produce an x86_64 only version of Rufus.

> Windows PE/Setup runs only native apps

That's not a mode I want to support for using Rufus. There are other issues there (regarding UI elements), so if you wanna use Rufus in a PE environment, you're on your own.

I strongly suggest you also read [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

zedd-0, 2022-07-29:
"so that I don't have to waste time with 2 x86 versions"
but you already waste time on 2 x86 versions.

why is there an x64 on the windows store,
if you do not have to. (even on the windows store)

that do not go in my mind.

pbatard, 2022-07-29:
> so that I don't have to waste time with 2 x86 versions

so that I don't have to waste __more__ time with 2 x86 versions

I don't think you quite understand how much more time is being taken away by producing 2 applications where one suffice, due to users being confused as to which one they should pick.

The store version hides that choice, so it's not an issue.
The downloads do not, so it __is__ an issue.

I already waste enough time with the portable and non portable version, and that is precisely why I want to limit the number of additional versions I publish. It's not that hard to understand when you actually make an effort to try to see things from a global perspective, rather than a _"But I want it!"_ one.

zedd-0, 2022-07-29:
you trust the users too little.
i have little sympathy when users don't understand the basics of computers.

p.s.:
for DAU's i write me a little starter script, witch automatically selects the right version.
%PROCESSOR_ARCHITECTURE%