pbatard/rufus #2006: Rufus crashed in the middle of an Windows on the go installation and now my pendrive does not work anymore

Veneiro, 2022-07-31:
I was trying to install a Windows 10 on the go version on my pendrive and rufus crashed almost at the end of the installation. Now when I connect the drive it crash my windows explorer, disk manager and even rufus or anything that is related to the disks until I disconnect the usb drive itself from my pc. Anyone know how to solve it or is my pendrive lost forever

pbatard, 2022-07-31:
By the looks of it, Rufus crashed because your USB was defective and it couldn't write to it. Especially, considering that simply connecting the drive crashes your PC indicates that this is a pure hardware issue, and it is simply [not possible for Rufus to damage a flash drive](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive).

Also, are all these applications actually crashing? Or are they simply freezing (i.e. the cease responding but they do not exist forcefully, which is what a crash is)? My guess is that they are freezing on account that your flash drive is unresponsive due to a hardware issue, because an actual crash of all these applications would be very surprising.

At any rate, your flash drive is most likely dead and you will need to replace it.

Veneiro, 2022-07-31:
> By the looks of it, Rufus crashed because your USB was defective and it couldn't write to it. Especially, considering that simply connecting the drive crashes your PC indicates that this is a pure [...]

This apps freeze until I disconnect the USB, meanwhile they just keep freeze. The USB was perfectly good until I tried the installation of the Windows 10 on the go, maybe there was a problem with the ISO I downloaded or something

pbatard, 2022-07-31:
> maybe there was a problem with the ISO I downloaded or something

Nah, it's just not possible for software or bad data to damage a flash drive. On the other hand, the flash memory used in pendrives is __A LOT__ more unreliable than people realize and will fail during common operations, regardless of the application being used or what it was writing to the drive.

Veneiro, 2022-07-31:
Ok thanks for your reply and your time, I'll just try to make it work and if not I guess I'll assume it's completely dead

pbatard, 2022-07-31:
Okay. I will close this issue then.

closed 2022-07-31 by pbatard