pbatard/rufus #2007: Near unreadable GUI in HDPI screen (4K)

RafaelLinux, 2022-07-31:
I have a always updated RUFUS version on Windows 10. I bought a 4K monitor and adjust desktop size (icons and fonts) to see clearly all GUI. However, Rufus doesn't resize its GUI according to Windows preferences. 

I searched in Rufus settings but didn't find any that help about.

Please, could you add suport for hDPI monitors???

Thank you

pbatard, 2022-07-31:
That's weird, because I am using a 4K monitor and made sure that Rufus __does__ resize according to the Windows scaling factor.

Here's my version running at 200% scaling factor on a 4 K monitor:
[image]

I really did spend a lot of time making sure that Rufus was HiDPI aware and would scale properly, so I would assert that there must be something off with your environment if you find that Rufus doesn't scale, because you're the first person to complain about such an issue (and Rufus is downloaded by about 2 million people each month, a lot of which will be running it on 4k monitors, so if there was a widespread issue with the application and scaling, it would long have been reported).

I would invite you to double check how you applied your scaling, and whether you don't have something in your environment that prevents application from reading your default scaling factor.

RafaelLinux, 2022-08-03:
You have reason. I don't know what happened. After a Windows reboot, GUI is correctly displayed, so it should be something related to Windows :(

THANK YOU and apologizes for open this thread

closed 2022-08-03 by RafaelLinux