pbatard/rufus #2013: Question: Windows Is Not Activated

changedsoul, 2022-08-06:
This is more a question than an issue, and before I knew about rufus.
I purchased a new laptop that came with windows 11 installed by dell. I booted it checked it out, made sure everything worked, then nuked it and installed linux. Few weeks later I decided I wanted to try and boot windows from an external USB4 SSD for them fringe things I need windows for. 
Cloning the original shipped storage to the external one would never boot. Always tried auto repair ad then failed. Then I found rufus, and since rufus cannot take a clonezilla image and mage it bootable for WindowToGo(this would be an awesome feature by the way)...I downloaded a genuine windows 11 iso and used rufus to make a WindowsToGo bootable external USB4 SSD.

My question, its my understanding windows 8,10, and 11 use digital keys and so booting windows 11 on a laptop that was shipped with windows 11 should be activated should it not? 

Thanks.

Checklist
---------
- [X ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

pineapple63, 2022-08-06:
Modern laptops usually have embedded product keys, the installer should detect the embedded product key and automatically select the correct edition (and it will then use this product key to activate windows)

changedsoul, 2022-08-06:
Yeah thats what I would have thought. I wasnt sure if maybe windows is being picky now because its running off a USB instead of internal? Also, pretty sure it came with home, but I seen N edition in the iso, maybe it came with Windows 11 N. I can redo with that and see if it activates I guess.

JonnyTech, 2022-08-06:
AFAIK, Microsoft only supports Windows To Go for Enterprise and Eductaion editions, therefore your Home licnsee is not valid.

pineapple63, 2022-08-06:
Since the licence that came with the laptop is an OEM licence, it could be Home, Home N, or Home Single Language

It may be the case that Windows to go does not activate with OEM licences (in which case you would need a Retail licence)

There is a trick i know of which you can use to figure out which edition of windows the embedded product key is for (which involves creating installation media and adding an answer file to it to get it to bring up the edition selection screen)

pbatard, 2022-08-06:
> rufus cannot take a clonezilla image and mage it bootable for WindowToGo(this would be an awesome feature by the way)

Not gonna happen. This falls straight into point 2 of [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want).

As to license activations issues, I'm afraid that this is out of scope of this issue tracker, on account that the best you're going to get are __unauthoritative__ educated guesses as to how Microsoft handles its activation. If you want a definitive answer, you should ask on Microsoft user forums, or Microsoft support directly.

closed 2022-08-06 by pbatard