pbatard/rufus #2014: How Rufus create the persistent usb?

yap241986, 2022-08-07:


pbatard, 2022-08-07:
1. Format an extra partition to the `ext3` file system. Possibly also create a `persistence.conf` there for Debian.
2. Copy the Linux ISO files and bootloaders to a main FAT32 (or NTFS) partition
3. Edit the GRUB or Syslinux configuration file to add the `persistent` or `persistence` kernel option, since, even where Debian does use a `persistence.conf`, Linux persistence is __TOO DUMB__ to check for persistent partitions on its own to decide whether it wants to use them without a bloody kernel option. Oh and since Linux distro maintainers __DON'T__ talk to one another, or try to follow what others are doing (Every distro seems to have a serious case of [NIH](https://en.wikipedia.org/wiki/Not_invented_here) for how they handle persistence), we need to __try__ to detect whether the ISO is Debian based or Ubuntu based because the keyword to use for the kernel __is not the same__...
4. ???
5. No profit whatsoever there, because this still won't allow persistence to work for distros that aren't Ubuntu or Debian based... ☹️ 

You may also want to see [here](https://github.com/pbatard/rufus/blob/b2bf29f65263f973d7511212a8b732b019cb98cb/src/iso.c#L340-L366) and [here](https://github.com/pbatard/rufus/blob/b2bf29f65263f973d7511212a8b732b019cb98cb/src/format_ext.c#L457-L484) for some more details.

closed 2022-08-07 by pbatard

yap241986, 2022-08-07:
How can I verify if is a Ubuntu or a Debian ISO? Did Rufus use some system commands?

pbatard, 2022-08-07:
> How can I verify if is a Ubuntu or a Debian ISO?

Ask the maintainers if it's an Ubuntu or Debian derivative.
Changes are that, if the official web site doesn't mention it, it's none of those.

At any rate, your question was on how Rufus creates a persistent USB, which is what I answered.
If your actual question was that you have a problem with persistence for a specific distro, you really should have said that, as well as disclosed the name of the distro you are using, instead of asking how Rufus creates a persitent drive.

> Did Rufus use some system commands?

Not sure what you mean by "system" commands (Rufus being a __Windows__ utility, it obviously cannot run __Linux__ system commands) but I'm going to assume that the answer to your question is no.

yap241986, 2022-08-07:
For example... 
echo, tree but to manage the disks.

pbatard, 2022-08-07:
Rufus is not a script application. It's a compiled program that uses C code to achieve what it needs to do. For instance, [this](https://github.com/pbatard/rufus/blob/b2bf29f65263f973d7511212a8b732b019cb98cb/src/drive.c#L2248-L2576) is what Rufus does, in C, to create a partition.

In other words, Rufus does __not__ call `diskpart.exe` to partition a drive, but instead it does the same as what `diskpart` does internally, and calls on the native Windows programming (not scripting) API.

As such, there is literally no "system" command being invoked. It's way lower level than that and you will be required to be familiar with C programming if you want to follow what Rufus does behind the scenes.

yap241986, 2022-08-07:
I'm creating a version for Mac but I use dd.
I'm using swift programming language

pbatard, 2022-08-07:
I'm afraid I can't help you there. If you want to create a version of Rufus for another OS, I will expect you to be able to first be able to read the C source code.