pbatard/rufus #2015: Windows store version install failed

swift-lee, 2022-08-11:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Windows store version install failed, for both MS Store and appx installer. Tried on Windows 11 22622.450 and 25174.1000. Indicating sr-SP is not an valit language. (0x80070057)
Log
---
```
<FULL LOG>
```
App installation failed with error message: 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt : error 0x80070057 : Cannot register the 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt package because the specified resource language 'sr-SP' is not a valid language. (0x80070057)
[image]

pbatard, 2022-08-11:
Thanks a lot for the report. I've been trying to address this issue for the last 24 hours (see [55ce65af6c464b2d91a882c1c7af80a00d5336bb]), and this is the first actual hint I'm getting as to what might be wrong with the app, so thanks a lot for providing this data.

Is there anything special you did to get to the screen you are showing where the full Reason for the failure was displayed? Because the best I ever got was a generic error screen with the code only, and no further hint as to the cause (and especially no mention about the language code being a problem):
[image]

So if you can detail how you exactly got to the message you show, I would greatly appreciate it, as I'll be able to at least get to some relevant info if an issue with the App Store install happens again...

Oh and you'd think that with a validation and certification process that takes about 6 days (which is how long it took to get the last submission approved), Microsoft would at least try to validate the region codes or, at the very least, try to __actually__ install the app to run a quick check, but it looks like this does __not__ happen. So, when people tell you that all of the Apple, Android and Microsoft App Stores do nothing to actually improve the user experience, but are a pure cash grab as well as a means to exert arbitrary control, you'd better believe them, because this is pure proof that not even basic elementary checks as to the viability of the app are not being performed before publishing it...

At any rate, I have submitted a new package with the country code for Serbian fixed. Hopefully, once it gets "validated" and published by Microsoft, that will fix the issue (and I'm also hoping that this process is not going to take 6 days like last time around). 

And thanks again for being __A LOT__ more helpful than Microsoft at identifying the actual issue so that I can try to fix it!

pbatard, 2022-08-11:
Answering part of my question above after finding [this page](https://docs.microsoft.com/en-gb/windows/win32/appxpkg/troubleshooting). The Windows Event Viewer will actually have the full installation error in `Applications and Services Logs` &rarr; `Microsoft` &rarr; `Windows` &rarr; `AppXDeployment-Server` &rarr; `Microsoft-Windows-AppXDeploymentServer/Operational`.

And PowerShell's `Get-Appxlog` will also provide a proper hint regarding the actual error:
```
PS C:\Users\pete> Get-Appxlog

Time                      ID           Message
----                      --           -------
2022.08.11 09:54:44       603          Started deployment RegisterByPackageFamilyName operation on a package with main parameter 19453.net.Rufus_y8nh7bq2a8dtt and Options
                                       RegisterHighestVersion,FailIfNeedsRemediation and SkipDeploymentOperationRpcCallerIsAdminCheck. See http://go.microsoft.com/fwlink/?LinkId=235160 for help
                                       diagnosing app deployment issues.
2022.08.11 09:54:44       10002        Creating Resiliency File
                                       C:\ProgramData\Microsoft\Windows\AppRepository\f0b86bc0-ba18-493b-8087-1fa6023ac545_S-1-5-21-1749556476-2293787269-1158594612-1001_3.rslc for Register
                                       Operation on Package 19453.net.Rufus_3.20.1929.0_neutral_~_y8nh7bq2a8dtt.
2022.08.11 09:54:44       607          Deployment Register operation on package 19453.net.Rufus_3.20.1929.0_neutral_~_y8nh7bq2a8dtt has been de-queued and is running for user SID
                                       S-1-5-21-1749556476-2293787269-1158594612-1001.
2022.08.11 09:54:44       614          Bundle Uri C:\Program Files\WindowsApps\19453.net.Rufus_3.20.1929.0_neutral_~_y8nh7bq2a8dtt\AppxMetadata\AppxBundleManifest.xml contains packages:
                                       "Rufus_3.20.1929.0_x86.appx": 19453.net.Rufus_3.20.1929.0_x86__y8nh7bq2a8dtt "Rufus_3.20.1929.0_x64.appx": 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt
                                       "Rufus_3.20.1929.0_arm.appx": 19453.net.Rufus_3.20.1929.0_arm__y8nh7bq2a8dtt "Rufus_3.20.1929.0_arm64.appx": 19453.net.Rufus_3.20.1929.0_arm64__y8nh7bq2a8dtt
                                       "Rufus_3.20.1929.0_scale-100.appx": 19453.net.Rufus_3.20.1929.0_neutral_split.scale-100_y8nh7bq2a8dtt "Rufus_3.20.1929.0_scale-125.appx":
                                       19453.net.Rufus_3.20.1929.0_neutral_split.scale-125_y8nh7bq2a8dtt "Rufus_3.20.1929.0_scale-150.appx":
                                       19453.net.Rufus_3.20.1929.0_neutral_split.scale-150_y8nh7bq2a8dtt "Rufus_3.20.1929.0_scale-400.appx":
                                       19453.net.Rufus_3.20.1929.0_neutral_split.scale-400_y8nh7bq2a8dtt.
2022.08.11 09:54:44       854          Successfully added the following uri(s) to be processed:  C:\Program
                                       Files\WindowsApps\19453.net.Rufus_3.20.1929.0_neutral_~_y8nh7bq2a8dtt\AppxMetadata\AppxBundleManifest.xml C:\Program
                                       Files\WindowsApps\19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt\AppxManifest.xml C:\Program
                                       Files\WindowsApps\19453.net.Rufus_3.20.1929.0_neutral_split.scale-100_y8nh7bq2a8dtt\AppxManifest.xml C:\Program
                                       Files\WindowsApps\19453.net.Rufus_3.20.1929.0_neutral_split.scale-150_y8nh7bq2a8dtt\AppxManifest.xml.
2022.08.11 09:54:44       855          Finished resolving action lists. DeploymentRequest action lists: addPackageList: 19453.net.Rufus_3.20.1929.0_neutral_split.scale-150_y8nh7bq2a8dtt
                                       19453.net.Rufus_3.20.1929.0_neutral_split.scale-100_y8nh7bq2a8dtt 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt
                                       19453.net.Rufus_3.20.1929.0_neutral_~_y8nh7bq2a8dtt.
2022.08.11 09:54:44       10000        About to service package 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt. Setting the package state to disabled returned with 0x0.
2022.08.11 09:54:45       5506         19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt : C:\Program
                                       Files\WindowsApps\19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt\microsoft.system.package.metadata\resources.3b4104ed.pri : Created shared merged PRI file
2022.08.11 09:54:45       5505         19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt : C:\Program Files\WindowsApps\19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt\microsoft.system.package.metadata
                                       \S-1-5-21-1749556476-2293787269-1158594612-1001-MergedResources-0.pri : Linked user-specific merged PRI file
2022.08.11 09:54:45       5500         19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt : error 0x80070057 : Cannot register the 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt package because the
                                       specified resource language 'sr-SP' is not a valid language.
2022.08.11 09:54:45       303          error 0x80070057: While preparing to process the request, the system failed to register the windows.MRT extension due to the following error: The parameter
                                       is incorrect.
                                       .
2022.08.11 09:54:45       5507         19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt : C:\Program
[... cut ...]
2022.08.11 09:54:45       404          AppX Deployment operation failed for package 19453.net.Rufus_3.20.1929.0_neutral_~_y8nh7bq2a8dtt with error 0x80073CF6. The specific error text for this
                                       failure is: 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt : error 0x80070057 : Cannot register the 19453.net.Rufus_3.20.1929.0_x64__y8nh7bq2a8dtt package
                                       because the specified resource language 'sr-SP' is not a valid language.
```

Gonna leave this here so that if someone ever gets error code `0x80070057` on Windows Store app install, they can get some diagnostic data to help them.

scriptingstudio, 2022-08-11:
The same here. Not depend on location, locale, etc.
As a fallback it is better to have appx installer on github.

pbatard, 2022-08-11:
> As a fallback it is better to have appx installer on github.

You do realize that the appx installer would have the exact same issue, don't you? The locales are present in the AppManifest and there's no way I'm going to maintain different manifest for the Store and a separate appx installer.

And you need to depend on location/locale so that the app properly reports what languages it supports...

eliezerjrs, 2022-08-11:
🆙 For me this issue only happens to Rufus.

pbatard, 2022-08-11:
Deleted dupe comment. And yes, other applications are not expected to run into this problem since, as identified above, the problem is that our last package submission contained an invalid country code for Serbia.

We have fixed this and submitted a new package about 8 hours ago, but __we have no control whatsoever over the Microsoft package approval process, which can take up to 6 days as was the case with our previous submission...__ 😢

scottbeamer, 2022-08-11:
I just came here to mention I'm having the same problem.  The store said there was an update, but it wouldn't update, so I tried uninstalling and reinstalling, and that gave me an error message as well.

vegetabledog-x86, 2022-08-11:
[image]
the same problem occered to me

pbatard, 2022-08-11:
Unfortunately, saying "me too" in this thread won't accomplish anything, as I have absolutely no means of making the Microsoft package approval process go faster.

The problem has most likely been corrected 10 hours ago already when I submitted a new version, but as long as Microsoft hasn't certified and published the new package, these changes won't appear in the Store.

This should actually serve as a __caution__ on how app stores are allegedly supposed to be better for users, compared to direct downloads. They are not, because they take control __away__ from the developer of the application, in order to put it in the hands of the Google, Apple and Microsoft of this world. So if you want a version of Rufus that just works and that isn't restricted by what Microsoft decides an app should or shouldn't do, just go to https://rufus.ie and get the application there. Oh, and you __will__ get automated updates and everything you'd expect from a Store app. Heck, you will usually get updates faster than from the Store, since I don't have to wait for Microsoft to move their ass and approve an updated package...

scottbeamer, 2022-08-11:
> Unfortunately, saying "me too" in this thread won't accomplish anything, as I have absolutely no means of making the Microsoft package approval process go faster.
> 
> The problem has most likely been corrected 10 hours ago already when I submitted a new version, but as long as Microsoft hasn't certified and published the new package, these changes won't appear in the Store.
> 
> This should actually serve as a **caution** on how app stores are allegedly supposed to be better for users, compared to direct downloads. They are not, because they take control **away** from the developer of the application, in order to put it in the hands of the Google, Apple and Microsoft of this world. So if you want a version of Rufus that just works and that isn't restricted by what Microsoft decides an app should or shouldn't do, just go to https://rufus.ie and get the application there. Oh, and you **will** get automated updates and everything you'd expect from a Store app. Heck, you will usually get updates faster than from the Store, since I don't have to wait for Microsoft to move their ass and approve an updated package to push an update for the regular app...

Why do you bother with a store version then with all these issues?   I actually first installed it from your website, then later by accident found out (via winget) that it was also in the store, so I went with the store version thereafter.
[image]

pbatard, 2022-08-12:
> Why do you bother with a store version then with all these issues?

Because it was gifted to me.

For the longest time, Microsoft did not accept applications that needed to run elevated in the Store, so I didn't really bother with anything Store related except for providing a `.appx` (because _some_ people really wanted Rufus in the form of an app rather than a portable executable).

Then about 1.5 years ago, and unbeknownst to me, Microsoft relaxed their "no-elevated apps in the Store ever" policy and around the same time, somebody try to see if they could build their own version of Rufus and get it accepted to the Store, which it did.

When that happened, that person first tried to monetize it, but they got a slew of bad reviews (_"How dare you trying to monetize something that's free!"_, _"Hey, you're not the official developer!"_, and so on, which is actually exactly [what Microsoft has unsuccesfully tried to weed out recently](https://techcrunch.com/2022/07/19/microsoft-u-turns-on-policy-that-wouldve-banned-commercial-open-source-apps/)), so, to their credit, they decided they wanted to get the app off their hands and got in touch with me to see if I was interested in taking ownership, which I was, since, even if I don't like App Stores, I'd rather have the inevitable version of my app that will appear there under my control instead of somebody else's...

At any rate, the latest Rufus package has been certified about 4 hours ago now and the Store versions should all have been updated, which means that the issue above is now fixed.

closed 2022-08-12 by pbatard

ahmedsabriz, 2022-09-15:
Relevant to @pbatard 's last comment, I thought you may want to know that winget package fails to install too. 

> Found Rufus [Rufus.Rufus] Version 3.20
> This application is licensed to you by its owner.
> Microsoft is not responsible for, nor does it grant any licenses to, third-party packages.
> An unexpected error occurred while executing the command:
> 0x80070032 : The request is not supported.

commit ae839d96 (2022-08-12): [appstore] fix invalid country code for Serbia

* sr-SR is not a country code that Microsoft accepts (and from what I can see
  is not valid, because it should be sr-RS).
* This has the unfortunate effect of preventing the installation of Rufus from
  the Windows Store, which fails with error 0x80070057 (Invalid parameter).
* Fix this by using a country code for Serbia that Microsoft does accept: sr-Latn-RS
* Closes #2015