pbatard/rufus #2017: [Enhancement] Local account creation when internet is connected.

Jai-JAP, 2022-08-12:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Currently Rufus provides a way to create local accounts on win11 22h2 if internet is disconnected. I've found that after adding `BypassNRO` registry key running `taskkill /F /IM oobenetworkconnectionflow.exe`
enables me to create a local account even when internet is connected.
Using this is it possible to implement a feature in rufus to allow local account creation even when internet is connected, without the requirement to create an account and then modify it.

Log
---
```
N/A
```

Jai-JAP, 2022-08-12:
This no longer works in latest 22h2 RP branch.

closed 2022-08-12 by Jai-JAP

pbatard, 2022-08-12:
At this stage, I'll wait for the formal 22H2 release before I alter anything related to local account creation.
Besides, if you want to bypass MSA while being connected to the internet, I will __STRONGLY__ advise you to use the _Set a local account using the same name as this user's_ option of Rufus 3.20, since this will achieve the same thing (and you can super easily change the username if the one proposed by Rufus isn't the one you actually want).

As such I will close this request.