pbatard/rufus #2019: Microsoft Store version error.

Hellements, 2022-08-12:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I installed Rufus from Microsoft Store. When I try to open it, I get this error:
[image]
Then, after a minute, the program will open:
[image]

Log
---
Rufus x64 v3.20.1930 (AppStore version)
Windows version: Windows 10 Pro, 64-bit (Build 19044.1889)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: Windows VDS is unavailable
0 devices found

pbatard, 2022-08-12:
Not seeing this, and nobody else seems to be reporting this issue.

VDS (and `vdsldr.exe`) is a native service from Microsoft Windows, so if you get an error when that starts, I strongly suspect that you have an issue with your Windows installation.

Possibly, a previous instance of VDS crashed and left an `11\vds.exe` remnant in `C:\Windows\System32`, since this is what `vdsldr.exe` appears to be complaining about, and the one issue I can see with a similar error seems to hint at something like this (because it doesn't look like a regular installation of Windows has that `11` directory).

However, since I'm just guessing here, I leave it up to you to decide if you want to try altering files in `C:\Windows\System32`, because, again, VDS and its behaviour is a pure Microsoft opaque box...

pbatard, 2022-08-17:
Considering that this is not a Rufus problem, but a Windows environmental issue that nobody else seems to be experiencing, I am going to close this issue.

closed 2022-08-17 by pbatard