pbatard/rufus #2020: feature request: Prevent Windows version coercion

MSP1978, 2022-08-14:
I use Rufus almost every day and would be very grateful for the integration of the following features.

With Dell notebook the Windows serial number is stored in the bios. So when installing Windows, there is no possibility to choose "Home" or "Professional". If a file "ei.cfg" is inserted in the ISO in the folder "sources" the version selection appears although the serial number is stored in the Bios.

The ei.cfg must have the following content:
[EditionID][Channel]
Retail
[VL]
0

Thanks & greetings
MSP78

pbatard, 2022-08-14:
I don't think there's enough demand for such a feature to warrant adding it to Rufus.

I hope you can appreciate that the vast majority of Rufus users will not be switching licenses, and it makes little sense to add a feature that would allow them to. This is a PowerUser/Sysadmin feature, and I'm afraid that Rufus is not designed towards such uses (See 3. from [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want)). Objectively, switching licenses is just too much of an __uncommon__ operation to want to have Rufus take care of it.

Instead, I will assert that, since it boils down to just adding an extra file on the media that is fairly well documented, users who do need to switch licenses should be able to create the `ei.cfg` on their own.

&rarr; Rejected

closed 2022-08-14 by pbatard

Hrxn, 2022-08-14:
> I use Rufus almost every day and would be very grateful for the integration of the following features.
>
> With Dell notebook the Windows serial number is stored in the bios. So when installing [...]

Wait, so this still works even with serial numbers stored in BIOS?
I've always understood that would be an exception here..

TheHellSite, 2022-11-09:
@pbatard 
Hello Pete, first off: Thank you for all the hard work that you put in this great tool!

> This is a PowerUser/Sysadmin feature

While I also think that is an option that will likely only be used by power users I think you might reconsider adding it.
I mean, you also added the options for Windows 11 (disable online account requirement, disable data collection, ...) which IMO are also only meant for power users.

No offense! Just my opinion. 

Kind Regards
TheHellSite

TheHellSite, 2022-11-09:
Okay, understandable. 👍