pbatard/rufus #2021: error when downloading the ISO of win 11 >>> log below

YagamieXe, 2022-08-15:
Rufus x86 v3.20.1929
Windows version: Windows 10 Home Single Language, 64-bit (Build 19044.1889)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xB359DC7E
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 14.6 GB (15668871168 bytes)
  Start Sector: 2048, Boot: Yes
Downloading https://github.com/pbatard/Fido/releases/download/v1.31/Fido.ps1.lzma
File length: 17.6 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/pt-BR/software-download/
Querying https://www.microsoft.com/pt-BR/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows11&query=&action=getskuinformationbyproductedition&sessionId=227d8d55-0f40-4098-914c-8ba77fad5f0c&productEditionId=2093&sdVersion=2
Querying https://www.microsoft.com/pt-BR/api/controls/contentinclude/html?pageId=6e2a1789-ef16-4f27-a296-74ef7ef5d96b&host=www.microsoft.com&segments=software-download,windows11&query=&action=GetProductDownloadLinksBySku&sessionId=227d8d55-0f40-4098-914c-8ba77fad5f0c&skuId=13479&language=Brazilian Portuguese&sdVersion=2
Exited download script with code: 0
Downloading https://software.download.prss.microsoft.com/dbazure/Win11_BrazilianPortuguese_x64v1.iso?t=6ae98a19-c88a-49d5-9818-f902c7c49b78&e=1660660605&h=4f90a674d14b6bc061982c5c5373d22bc107d6fb8061ad7f42bca050d6d151cc
File length: 4.9 GB
Could not download complete file - read: 57016320 bytes, expected: 5288691712 bytes

sintaxx, 2022-08-15:
Apparently Microsoft has blocked ISO downloads via whatever method Rufus uses. You can still get ISO's from Techbench WZT website.. maybe the dev could make rufus download the ISO's from there.

YagamieXe, 2022-08-15:
I can download the iso from outside the rufus, the problem is this: my notebook does not have tpm 2.0 and when I download the iso on the outside the rufus does not give the option to (no tpm/no secure boot)

YagamieXe, 2022-08-15:
Is there any other way to circumvent tmp and secure boot?

JonnyTech, 2022-08-15:
@sintaxx your statement is not correct, that issue has been fixed in Fido 1.31 which Rufus automatically downloads and uses - and is logged in the issue.

@YagamieXe do you have any security / firewall software that may be interfering? You can also use Rufus on your downloaded ISO as long as it is release 22H2 then the option to bypass restriction will be shown **AFTER** you press the `START` button.

YagamieXe, 2022-08-15:
You really have this option, I'll test it and come back here to give feedback

pbatard, 2022-08-15:
> `Could not download complete file - read: 57016320 bytes, expected: 5288691712 bytes`

That's a problem with your internet connection, not Rufus or Fido.

The only thing I can advise is to use the option to download in the browser, which is proposed to you in the download dialog, as if it's a matter of issuing a retry on a broken connection, a browser may be a bit more resilient. Bur really, if Rufus tells you that it wasn't able to download all the data, the issue is most likely with your connectivity.

pbatard, 2022-08-15:
> I can download the iso from outside the rufus, the problem is this: my notebook does not have tpm 2.0 and when I download the iso on the outside the rufus does not give the option to (no tpm/no secure boot)

Then you are not downloading a retail Microsoft Windows 11 ISO, because Rufus will propose the option to bypass TPM/Secure Boot if you use a retail ISO.

Again, please try to use the option to download in a browser from Rufus, so that you do download a retail ISO, and once it is downloaded, you should be able to accomplish what you want.

closed 2022-08-17 by pbatard