pbatard/rufus #2023: There is no "Download" button. No internet connection?

Ma89jf151a, 2022-08-16:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I've been downloading the newest versions of Rufus for 2 years in a row, but I can't download any ISO image in any of them. I simply don't have such a button.
When trying to find updates, a window appears in which it says that no new updates have been found, but at the same time it says "Error: Internet access is required" at the bottom. My Internet works stably and everywhere, BUT it is your program that does not see it. I have a dynamic IP address, it changes every time after restarting the router. I tried running Rufus on Windows 7 and there is a download button there, BUT I can't download it since it requires PowerShell 3.0. Can you fix it? I would like Rufus to work properly on Windows 10.

Log
---
```
<FULL LOG>
```
Rufus x86 v3.20.1929
Windows version: Windows 10 Pro, 64-bit (Build 19044.1288)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'UFD 2.0 Silicon-Power16G USB Device' (13FE:3600)
1 device found
Disk type: Removable, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1884, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x014413F3
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 14.4 GB (15501099008 bytes)
  Start Sector: 2048, Boot: Yes
Could not open Internet session: Internet is disconnected.
Checking for Rufus updates...
Updates: Unable to connect to the internet
[image]
[image]

JonnyTech, 2022-08-16:
> Checklist
> I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.

https://github.com/pbatard/rufus/wiki/FAQ#Help_I_dont_see_the_ISO_download_button

pbatard, 2022-08-16:
Your problem is this:
```
Could not open Internet session: Internet is disconnected.
```

The <kbd>DOWNLOAD</kbd> button will only appear if Rufus is able to connect to the Internet, and it looks like your machine is preventing regular applications from connecting. It may be a security solution, using whitelisting to only allwo browsers to connect to the Internet. Or it could be something else.

At any rate, it will be up to __you__ to figure out why Rufus can't connect to the Internet on your system, because on a standard installation of Windows, Rufus has no problem doing so, therefore the problem is most certainly on your side.

As such, I will close this issue.

closed 2022-08-16 by pbatard