pbatard/rufus #2025: Where is the choice?

QueenUlyana, 2022-08-17:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
No choice when installing image "installing windows 11 without tpm 2.0 module and secure boot"

Log
---
```
Rufus x86 v3.20.1929
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'UFD 2.0 Silicon-Power16G USB Device' (13FE:3600)
1 device found
Disk type: FIXED, Disk size: 16 ГБ, Sector size: 512 bytes
Cylinders: 1884, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {F58CA837-42CB-4A54-AB35-051AB1A2F9E7}
Max parts: 128, Start Offset: 17408, Usable = 15502113280 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {64D026D4-08E1-4140-8035-946B2A6EBFE3}
  Size: 14.4 GB (15501082112 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 11 ISO (Build 22000.194)
  Size: 4.1 ГБ (Projected)
  Note: File on disk is larger than reported ISO size by 532 КБ...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.1 ГБ)
```
[image]
[image]

pbatard, 2022-08-17:
Did you not get a dialog where you could change options after you pressed `START`.

If your log says:
```
 Detected: Windows 11 ISO (Build 22000.194)
```
Then you should have seen that dialog.

I suspect that you are confused because the option to bypass TPM and Secure Boot no longer appears in the _Image options_ menu as they did in 3.18. See https://github.com/pbatard/rufus/wiki/FAQ#Help_I_no_longer_see_the_option_to_bypass_TPMSecure_Boot_with_Windows_11.

QueenUlyana, 2022-08-17:
> Did you not get a dialog where you could change options after you pressed `START`.
>
> If your log says:
>
> ```
>  Detected: Windows 11 ISO (Build 22000.194)
> ```
>
> Then you should have seen that dialog.
>
> I [...]

I assure you, this window did not go out. I tried several times.

pbatard, 2022-08-17:
Aha, it's because you are using Windows 7:
```
Windows version: Windows 7 SP1 64-bit
``` 
The options will __NOT__ display on Windows 7, and as a matter of fact, [I am no longer supporting Windows 7](https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED).

If you __choose__ to run Rufus on Widows 7, you are 100% on your own.

closed 2022-08-17 by pbatard

ccoenen, 2022-08-29:
I ended up here because I _also_ had not seen that option. I would have expected these options to be present _before_ hitting start. Start, to me at least, means "I'm happy with all my settings, now put it on my drive".

cinnamonmatexfce, 2022-09-12:
> I ended up here because I _also_ had not seen that option. I would have expected these options to be present _before_ hitting start. Start, to me at least, means "I'm happy with all my settings, now put it on my drive".

Idem

pbatard, 2022-09-12:
The problem with doing it the way you assert I should do is that these options only apply to Windows, there's 5 of them (with possibly more to come), and they are different depending on whether the user chooses Windows To Go or regular install.

So, __if__ I were to do it the way you suggest, I'd have to move the choice between Windows To Go or regular install to a prompt that happens as soon as the user select an ISO. Then the user would get another prompt about these options with the ones that apply to the mode they selected with the previous prompt. Then nothing would happen and the interface would go back to the default selection screen, on which the user may very well decide that they didn't want to write a Windows ISO after all, but a Linux ISO, in which case what they were prompted about was pointless. Oh and it also gives the user ample time to forget what options they picked up as they go back to the other options such as partition scheme, target system, etc., so that by the time they press START they might be wondering what the heck they selected when they opened the ISO, which may have happened who knows how long ago, compared to what typically happens when you do state that you are ready to create the media and the application asks you for options that shouldn't be proposed globally on the UI (because these don't apply to all ISOs. Heck, they don't even apply to all Windows ISOs). Which means that these options __must__ happen on a separate dialog prompt, and I have to argue that, when considering objectively whether this separate dialog prompt should occur before or after the user presses the START button, the only choice that makes sense from a UI design perspective, if you are trying to help users, is to have that prompt __after__ the user indicates that they are happy with the global/common options and are ready to create a media for a specific type of ISO.

Similarly, Rufus will not prompt you whether you want to write a Linux ISO in DD mode or ISO mode until __after__ you press the START button. And it will also not prompt you whether you want to download additional Syslinux or GRUB files until after you press the START button. Yet, strangely, nobody has ever complained that they think that these prompts or option selections should happen before the user presses the START button, because they wrongly assume that `START` (which simply means "Start the media creation process", __no more, no less__ and makes __NO IMPLICIT PROMISES WHATSOEVER ABOUT THAT CREATION PROCESS NOT REQUIRING ADDITIONAL USER INPUT__) means something that it is actually unable to provide.

Now, if you want to prove me wrong, please try designing a UI mockup that accomplishes what you say Rufus should be doing (while remembering that Rufus must cater for other ISOs than Windows 11 ones) and see how that goes. Because I am pretty sure that there will be/are some people here under the impression that, no matter how a developer might try to explain things factually, the developer "must" still be at fault here, and "must" be dismissing the suggestion out of spite or laziness or whatnot, and that the only way to convince them that the approach being used by Rufus is actually sound and less clunky for end-users, is to have these people get their hands dirty in actual UI design (which has a lot more constraints a lot less "obviousness" than people give it credit for).