pbatard/rufus #2027: WindowsToGo 11 will not shutdown completely.

changedsoul, 2022-08-19:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Additionally (if applicable):
  - [x ] Plugging the USB into a different port.

Issue description
-----------------
I do not know if this is normal behavior for a WindowToGo windows 11 USB boot, but my laptop will not fully power down when shutting the system down. It seems to get to a good shutdown state, but something with the external USB makes it never fully power down.
I have to hold the power button down to shut it off.

The light on the USB disk blinks blue when active, and when shutting down, its quite active, but just before the screen goes black, the led turns red, then goes dark, and the laptop "looks like its off" but it not not fully powered down.

pineapple63, 2022-08-19:
Im not entirely sure, but i know some computers continue to provide power to the USB ports after shutting them down

pbatard, 2022-08-19:
I'm afraid this is your environment. Nobody else is reporting this, and while testing Windows 11 To Go myself, I've never had issues powering computers down.

From what I gather, you didn't test on any other computer but this specific laptop of yours, so it looks like this is a problem between your laptop and Windows.

Oh and keeping USB ports powered is independent of shutdown state (since shutdown is a CPU state, and you can have a CPU shut down while the computer is still providing power to ports).

Considering that this issue is unlikely to have anything to do with Rufus, I will close it. I would advise you to post on Windows forums or on user forums for your specific brand of laptop.

closed 2022-08-19 by pbatard