pbatard/rufus #2028: Rufus left my USB unusable

franzageek, 2022-08-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I know I am the 20th guy writing this, but I think this case is really weird.
I was using Rufus to flash a win8.1 image on an usb, while it says 
"The E: disk is used by the following process: 
• [Explorer.exe]
Close all the programs and retry. "
I did it, but my usb drive was no longer showing up:
- In Explorer, in "This PC", there was "USB Disk (E:)" grayed out (instead of my USB's name) and when I attempted to format it, it said: "No disk inserted in E: - Insert a disk and retry." ; 
- In Disk Management, there was: "Disk [number]: No support", and I could not do anything (like formatting it or so);
- The same in Diskpart;
- ChkDsk hangs on saying it cannot access directly to the drive,
- I also tried in Disk Utility on my Mac, and the usb was not showing up.

I know very well how the formatting process works, but I really cannot understand why this happened lol.

pineapple63, 2022-08-21:
Seems the USB drive may have failed

franzageek, 2022-08-21:
Yeah I think so, but I really don't know how to fix it

pbatard, 2022-08-21:
I'm afraid I have nothing more to add than [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_Rufus_damaged_my_flash_drive) and [this](https://github.com/pbatard/rufus/wiki/FAQ#Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media).

The flash memory used in consumer USB flash drives is a lot more unreliable than people imagine, and when you use a flash drive they tend to fail.

Rufus simply reported that it was unable to write to the drive (i.e. your drive had already failed) and, because this can __also__ happen when an external process is using the drive and locking it, tried to list processes that were accessing the drive in case this was due to it.

But your further reports show that this is a simple hardware drive failure, where the flash memory failed and, after the controller in your USB drive detected it, the controller put your memory offline.

I'm afraid there is no fix. You will have to replace that drive.

closed 2022-08-21 by pbatard