pbatard/rufus #2029: MAX_DRIVES might be too low causing Rufus to occassionally fail to detect drives

jaybz, 2022-08-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus sometimes is unable to detect any drives due to the device number being too big. I've included the full log below, except for the end with the USB debug information I'm also including log snippets below that appear to be related to the issue. 

I tried 4 devices (3 flash drives, 1 hard drive in a USB enclosure) with the List USB Hard Drives option checked. None of them were initially detected by Rufus. After trying repeatedly to get one of the drives detected (I tried using other USB ports at this point) and restarting Rufus, the issue appears to have resolved itself and all 4 devices could then be detected. Note that all the drives and USB ports I tried previously is now working.

Below is the part of the log wherein the issue somehow resolved itself. It shows a first attempt where the error still occurred and a 2nd attempt with a different drive which did get detected. Note that the detected drive was a drive that I tried to use but was not detected by Rufus due to the device number. I don't think I did anything to actually resolve the issue in between the two times I plugged in a device in the log snippet, other than just unplugging and unplugging the drives shown in the log. In between the attempts, I was probably just taking a peek the source code via Github to see if I can find a clue as to how to solve the issue.

```
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (65) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
0 devices found
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1003, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Detected File System: FAT32
  Size: 7.7 GB (8249923584 bytes)
  Start Sector: 63, Boot: Yes
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
1 device found
```

I tried all the drives I that Rufus was not able to detect and tried creating bootable USBs with the drives and encountered no issues this time. I've included portions of the log below showing the same device that was not detected in the log snippet above.

```
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Using 'autorun.inf' label for drive H: 'CLOVERCD'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 57.3 GB (61529391104 bytes)
[... 498 lines cut ...]
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 57.3 GB (61529391104 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Using 'autorun.inf' label for drive H: 'CLOVERCD'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 57.3 GB (61529391104 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Using 'autorun.inf' label for drive H: 'CLOVERCD'
1 device found

```

closed 2022-08-26 by pbatard

jaybz, 2022-08-27:
I'm pointing at MAX_DRIVES due to these lines of code in drive.c, line 1080 in the GetDriveNumber() function as this appears to be the only place where I can find the error message I found in the log.

```
	if (r >= MAX_DRIVES) {
		uprintf("Device Number for device %s is too big (%d) - ignoring device", path, r);
		return -1;
	}
```

You'd have to scroll to the right on the log snippets to see the error message but it shows that the device number that Rufus was getting was 65 in the attempt where I logged the issue.

jaybz, 2022-08-27:
And as I just found out now, the issue isn't as difficult to trigger as I initially thought. I managed to get a full log of the issue using 3 different drives and 3 different USB ports. I got these 3 drives to work fine when I first encountered the issue and it disappeared, and 2 of those 3 USB ports where what I used then. Note that Rufus is receiving device numbers higher than 100 in this case.

```
Rufus x86 v3.20.1929
Windows version: Windows 11 Pro, 64-bit (Build 22000.856)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (108) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (108) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 2.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
Device Number for device \\?\usbstor#disk&ven_jetflash&prod_transcend_32gb&rev_8.07#eudjelvt&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (109) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (108) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 2.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
Device Number for device \\?\usbstor#disk&ven_jetflash&prod_transcend_32gb&rev_8.07#eudjelvt&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (109) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (108) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 2.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
Device Number for device \\?\usbstor#disk&ven_jetflash&prod_transcend_32gb&rev_8.07#eudjelvt&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (109) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
Device Number for device \\?\usbstor#disk&ven_&prod_usb_flash_drive&rev_pmap#19830b340be3&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (110) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (108) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 2.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
Device Number for device \\?\usbstor#disk&ven_jetflash&prod_transcend_32gb&rev_8.07#eudjelvt&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (109) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
Device Number for device \\?\usbstor#disk&ven_&prod_usb_flash_drive&rev_pmap#19830b340be3&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (110) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (108) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 2.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
Device Number for device \\?\usbstor#disk&ven_jetflash&prod_transcend_32gb&rev_8.07#eudjelvt&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (109) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 2.0 device 'USB FLASH DRIVE USB Device' (1005:B113)
Device Number for device \\?\usbstor#disk&ven_&prod_usb_flash_drive&rev_pmap#19830b340be3&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (110) - ignoring device
A device was eliminated because it didn't report itself as a disk
Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5595)
Device Number for device \\?\usbstor#disk&ven__usb&prod__sandisk_3.2gen1&rev_1.00#0101be68a05a3bcb9cccbdb96c3f22adf96b13410dc50f0b9f8f4e3de3ffcc5#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (108) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
```

pbatard, 2022-08-27:
Ah, I missed that your earlier message was for a different check. Good point.

Now the problem here is that we can't just randomly increase the maximum number of drives we accept for this specific part of the code, because we deliberately offset the drive number by `0x80` as a __basic__ safety measure so that if we ever get a _stray_ drive number being passed to one of our calls (which one is entitled to expect to be under `0x80` on the vast majority of systems), we're not going to start erasing a drive that wasn't the expected one, and especially not the lower Windows drive IDs for `0`, `1` and so on, that are expected to be system drives. See https://github.com/pbatard/rufus/blob/master/src/drive.c#L120-L130

So that means we want an offsetted __limited__ range for our internal drive numbers, and I deliberately chose to limit that range to 64 entries because, of course, the more you limit the range, the more likely you will be able to catch _stray_ drive numbers that are the result of a bug or an improper function call.

This means there needs to be a compromise on the maximum number of drives we want to have in our selective range, that compromise being that it shouldn't be too small so that people run into it too easily (i.e we wouldn't want to have a range of `16` or `32` drives, because systems with that many drives  aren't expected to be that ucommon) but we also don't want the range to be too large so that it defeats the point of being a limited range that gives us a fighting chance to catch values that the application doesn't expect. Oh and we wanted it to be a byte to simplify our handling (otherwise, we'd just have offsetted the Windows value by something like `0xFFFF` and be done with it).

At this stage, considering that you are the very first person in 10 years (!) to report running into our max range limitation, I'm not sure I want to alter anything, because my expectation is that you either have more than 64 drives on your system, or at least Windows is internally detecting more than 64 of them (because as far as I know, the numbers reported by `IOCTL_STORAGE_GET_DEVICE_NUMBER` are incremental starting with `0`). If anything, it seems to point out that our choice of for the offsetted limited rage we use as a basic validation for dealing with Rufus' internal drive numbers was probably right, considering that, so far, you are the only person to report ever running into that limit.

So, unless I'm starting to see other reports of people running into that issue, I'm just going to have to leave people who are trying to run Rufus with more than 64 drives on their system stranded, because, at this stage, I just don't see much worth in investing time trying to increase that limit, or switching to a different check, and __validating that it won't have negative effects on the existing code__.

I may add an entry in the FAQ to let people know that, if Windows lists more than 64 drives on their system, Rufus may not be able to locate the drive they want to use...

jaybz, 2022-08-27:
I've investigated the issue further as I couldn't get Rufus to detect my drives again. I'm happy to report that, at least in my case, I've found a reliable workaround. I only have 5 physical drives and one virtual DVD drive so Windows shouldn't need to use a device number higher than 20. Then I remembered that the Xbox for PC app uses virtual disks for most, if not all, games that are available through it. It uses one virtual disk per game that is packaged that way so installing 63+ games will likely cause Rufus to ignore any USB devices that are plugged in after all the virtual drives are mounted. These virtual drives show up in device manager under Disk drives as devices named "Xvd".

The virtual disks are managed by a service named "Gaming Services". There are two services with that name and they are both safe to stop. I think you only need to stop a specific one to unmount the virtual disks but it's just faster to stop them both than figuring out which is which first. The Xbox App should not be running, however, as it automatically restarts these services. It should be enough to right-click on the Xbox App tray icon and selecting Quit on the context menu before stopping the services.

As far as I can see, the Xbox App only mounts the virtual drives when it needs to, but doesn't unmount them afterwards. I think in my case, the Xbox app updated causing the service to restart and unmount the virtual drives. This will make the issue to go away temporarily. If that was the case, anybody experiencing this issue can just restart both services, even while the Xbox App is running. This should make Windows use lower devices numbers until the Xbox App decides it needs to mount enough of the virtual drives to trigger the issue again.

One thing to note is that there is absolutely no indication that these virtual drives exist unless you look at the drives in device manager or the defrag tool with Advanced View enabled. I wouldn't have realized what was going on until I remembered the Xbox App uses virtual drives. I suspect if the Xbox Game Pass for PC gains enough popularity, a FAQ entry will be necessary. I also expect that modifying the code to address the issue won't be necessary, at least not anytime soon. I think anybody who is comfortable enough to create a bootable USB drive will likely be comfortable with stopping/restarting a couple of Windows services.

pbatard, 2022-08-27:
> Then I remembered that the Xbox for PC app uses virtual disks for most, if not all, games that are available through it. It uses one virtual disk per game that is packaged that way

Oh wow! I  may have to review my assertion that the range limitation may not affect that many users, because this virtual drive creation by the XBox PC app may indeed become a problem...

> As far as I can see, the Xbox App only mounts the virtual drives when it needs to, but doesn't unmount them afterwards.

Okay. So theoretically this should only become an issue for people who have launched many games and haven't rebooted their PC, or if for whatever reason the XBox app decides that it needs to mount a bunch of games for updates or something...

I gotta think about this a little be more, and see how prevalent this might become. I may also alter the error message displayed in the log to let users know that, if they have the XBox app installed, they may want to try to restart it. And I'll definitely add a note in the FAQ about this.

At any rate, thanks a lot for investigating this, and sorry for failing to see the actual error in your log and missing the mark on my first reply. I'm still hoping that not many people should be affected by this issue, but I'll definitely be monitoring if they are.

JonnyTech, 2022-08-27:
> As far as I can see, the Xbox App only mounts the virtual drives when it needs to, but doesn't unmount them afterwards.

Can an external application, such as Rufus, make an eject request for these mounted virual drives? Similar to ejecting physical drives, there may be an API available. With user notification, of course.

jaybz, 2022-08-27:
> Okay. So theoretically this should only become an issue for people who have launched many games and haven't rebooted their PC, or if for whatever reason the XBox app decides that it needs to mount a bunch of games for updates or something...

I suspect the app checks for updates on a schedule, perhaps daily, and it mounts everything, or almost everything, regardless of whether there's actually an update for the game or not. I'm pretty sure I didn't play/install/update more than 20 games since I last restarted before the issue started to re-occur.

> At any rate, thanks a lot for investigating this, and sorry for failing to see the actual error in your log and missing the mark on my first reply. I'm still hoping that not many people should be affected by this issue, but I'll definitely be monitoring if they are.

No worries. I know how easy it is to miss things in logs when there's so much information available. And that's still better than not having enough information.

jaybz, 2022-08-27:
> Can an external application, such as Rufus, make an eject request for these mounted virual drives? Similar to ejecting physical drives, there may be an API available. With user notification, of [...]

Since Rufus already runs in admin mode, I think it will have permissions to stop/restart the services which achieves the same result. It just needs to be able to identify the correct service(s) to stop/restart.

I'm hesitant to suggest using a Windows API to eject the drives directly since, in my experience, the Xbox App is still a buggy mess. Unless the Xbox App itself provides such an API, which I doubt, "surprising it" by ejecting the drives will likely cause other issues.

pbatard, 2022-08-27:
> I'm hesitant to suggest using a Windows API to eject the drives

Same here. Plus, since we don't want to eject drives without warning the users and giving them a choice, that'd mean new API, new translations, and require time that I am not willing to invest on an issue that has _yet_ to affect many users.

There are plenty of corner cases, such as this one, that I could address in Rufus one way or another, but when you put them end to end, you start to spend your life addressing corner cases instead of improving the features and behaviours that most users will actually benefit from, so it's usually not a sound investment.

As indicated earlier, I am not planning to go the extra mile on this specific issue at this stage, besides giving more information to the user about this in the log and in the FAQ (since those are genuinely 5 mins changes to apply, instead of what would most likely be a multi-day investement if I were to add eject, especially when factoring UI + translation requirements).

pbatard, 2022-08-28:
I have now added an extra notice in the app when the error is triggered and created [this new FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Device_Number_for_device_XYZ_is_too_big__ignoring_device).

Vantropix, 2022-09-23:
Can confirm that the Xbox App is the main issue here since when downloading games from the Xbox App it makes a Xvd or Xbox Virtual Disk. This is likely done better support the game and make it easier to run on pc if there's no direct pc port for the game.

Personally i have several hundred games installed and my Xvd amount is 92 at the moment and restarting will not do anything since there is a game installed on each of those, this makes me completely unable to use rufus on my main system which is unfortunate.