pbatard/rufus #2030: Rufus doesnt open window to download windows isos.

randomguyfromAustria, 2022-08-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I have a issue. Whenever i click on "DOWNLOAD", it says that it downloads a file, after it finshed downloading the file, it doesnt open the window to download windows isos.

Log
---
```
Rufus x86 v3.20.1929
Windows version: Windows 11 Pro, 64-bit (Build 22000.856)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.31/Fido.ps1.lzma
File length: 17.6 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Add-Type : (0) : Warnung als Fehler: Ungültiger Suchpfad 'C:\Program Files (x86)\Microsoft Visual Studio\VC98\mfc\lib' 
in 'LIB-Umgebungsvariable' -- 'Das System kann den angegebenen Pfad nicht finden. '.
(1) : using System;
In C:\Users\franz\AppData\Local\Temp\{30126341-8460-47F9-BA00-882EBF8B2111}.ps1:86 Zeichen:3
+         Add-Type -MemberDefinition $code -Namespace Gui -UsingNamespa ...
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidData: (Microsoft.Power...peCompilerError:AddTypeCompilerError) [Add-Type], Except 
   ion
    + FullyQualifiedErrorId : SOURCE_CODE_ERROR,Microsoft.PowerShell.Commands.AddTypeCommand
Exited download script with code: 1

```

pbatard, 2022-08-26:
`Windows version: Windows 11 Pro, 64-bit (Build 22000.856)`

Okay...

`Add-Type -MemberDefinition $code -Namespace Gui -UsingNamespa ...`

Huh? Considering that [this](https://github.com/pbatard/Fido/blob/master/Fido.ps1#L83-L87) is our code, it means that your [`$host` built-in PowerShell variable](https://docs.microsoft.com/en-us/powershell/module/microsoft.powershell.core/about/about_automatic_variables?view=powershell-7.2#host) reports a version lower than 7.0 __which should never be the case on Windows 11__

So this means that either there is a bug with Microsoft PowerShell, or you have altered your system in a manner that makes it report a wrong OS version.

At any rate, this is clearly not a Rufus/Fido problem (because you cannot expect us to take into considerations systems where PowerShell does not report an inaccurate version of the OS installed), but a problem with your environment, I will close this issue.

Tom-EllisEVENTS, 2022-09-23:
I am experiencing the same problem on Windows 10, apparently it fails on line 86:

> Add-Type -MemberDefinition $code -Namespace Gui -UsingNamespace System.IO, System.Text, System.Drawing, System.Globalization -ReferencedAssemblies System.Drawing -Name Utils -ErrorAction Stop

It does **not** fail if I call fido1.ps1 directly from a powershell prompt that is started with standard user rights - so I can download the ISO file and use it later.

But it fails with the same error if I run fido1.ps1 from an elevated powershell prompt (which is also the case when run from the Rufus GUI)

Any idea, what could be the root cause for this?

LOG:

Rufus x86 v3.20.1929
Windows version: Windows 10 Pro, 64-bit (Build 19044.2006)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.33/Fido.ps1.lzma
File length: 17.8 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Add-Type : (0) : Warnung als Fehler: Ungültiger Suchpfad 'N:\LIB' in 'LIB-Umgebungsvariable' -- 'Das System kann den 
angegebenen Pfad nicht finden. '.
(1) : using System;
In C:\Users\********\AppData\Local\Temp\{2E4DDB82-B7A5-49CD-BA78-2729D588332C}.ps1:86 Zeichen:3
+         Add-Type -MemberDefinition $code -Namespace Gui -UsingNamespa ...
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidData: (Microsoft.Power...peCompilerError:AddTypeCompilerError) [Add-Type], Except 
   ion
    + FullyQualifiedErrorId : SOURCE_CODE_ERROR,Microsoft.PowerShell.Commands.AddTypeCommand
Exited download script with code: 1

pbatard, 2022-09-23:
From what I can see from the ` Warnung als Fehler: Ungültiger Suchpfad 'N:\LIB' in 'LIB-Umgebungsvariable' -- 'Das System kann den angegebenen Pfad nicht finden. '` error, one issue you have is that you defined a mapped drive `N:` as a system library search path, which cannot be accessed when running PowerShell elevated. This is most likely due to the fact that your `N:` drive is a network mapped drive, that is only visible to your current user, and will __not__ be mapped for the Administrator user.

See https://github.com/pbatard/rufus/wiki/FAQ#why-can-rufus-not-see-my-network-shares

I would __strongly__ advise you to figure out why you have a __nonstandard__ PowerShell search path that includes network shares and sort that out.

pbatard, 2022-09-23:
By the way, I was wrong with what the `$host.version` variable is for above. This does not identify the Windows host version, but the version of PowerShell being used, and unless you installed PowerShell 7.0, you will indeed code from the second branch executed even on Windows 11.

In both cases however, I'm seeing an issue with an elevated PowerShell not being able to access paths that are defined in its library lookup paths, which seems to indicate that you manually altered your system from vanilla Windows (where the search does not fail).

Tom-EllisEVENTS, 2022-09-23:
Thanks - understood.

N:\LIB is a search path for a programming language, that is mapped by using SUBST.

After mapping N: manually inside the elevated prompt, the script works.

If I remove the mapping afterwards, the script also runs a second time without error... very strange.

```
PS C:\Users\********\Downloads> .\fido1.ps1
Please Wait...
Add-Type : (0) : Warnung als Fehler: Ungültiger Suchpfad N:\LIB in LIB-Umgebungsvariable -- Das System kann den
angegebenen Pfad nicht finden. .
(1) : using System;
In C:\Users\********\Downloads\fido1.ps1:86 Zeichen:3
+         Add-Type -MemberDefinition $code -Namespace Gui -UsingNamespa ...
+         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidData: (Microsoft.Power...peCompilerError:AddTypeCompilerError) [Add-Type], Except
   ion
    + FullyQualifiedErrorId : SOURCE_CODE_ERROR,Microsoft.PowerShell.Commands.AddTypeCommand

PS C:\Users\********\Downloads> subst N: D:\Alaska\XPP554
PS C:\Users\********\Downloads> .\fido1.ps1
Please Wait...
PS C:\Users\********\Downloads> subst N: /D
PS C:\Users\********\Downloads> .\fido1.ps1
Please Wait...
PS C:\Users\********\Downloads>

```

But what I don't understand is why AddType fails at the first place? 

But probably that's a question I should ask Microsoft :-)

Tom-EllisEVENTS, 2022-09-23:
One last comment: If I add "-IgnoreWarnings" to the second Add-Type command (just like in the first one for Powershell 7), it works without error.

pbatard, 2022-09-23:
> If I remove the mapping afterwards, the script also runs a second time without error... very strange.

Probably because of caching.

> But what I don't understand is why AddType fails at the first place?

Well, it pretty much tells you that it wants to be able to load libraries (whether it actually needs them or not) from all the library paths you have defined on your system, and will fail if any of these paths are inaccessible.

That doesn't mean that I don't agree that it should just fail silently or warn if a path is not accessible by default, instead of breaking the script.

> One last comment: If I add "-IgnoreWarnings" to the second Add-Type command (just like in the first one for Powershell 7), it works without error.

Yeah, that's what I was about to ask you to test, since I'm looking at the [`Add-Type` documentation](https://learn.microsoft.com/en-us/powershell/module/microsoft.powershell.utility/add-type?view=powershell-7.2). I guess I'll add this to the script then. Let me reopen this issue, so that I can close it when I do so.

Tom-EllisEVENTS, 2022-09-23:
Great! Thank you very much, have a nice weekend :-)

closed 2022-09-23 by pbatard