pbatard/rufus #2031: Request for linux iso

RyzenDew, 2022-08-31:
Have GPT set as default but if only if your running a uefi OS already or if rufus can see you can do uefi.
this will stop issues with linux users and stop users from using mbr which is useless in this day and age

pbatard, 2022-08-31:
> this will stop issues with linux users 

[Citation needed]

Maybe I'm not looking hard enough, but (as opposed to what has been the case for Windows) I'm not seeing that many issues from Linux users associated to having a dual BIOS+UEFI bootable media. On the contrary, it seems very clear to me that, in the days of ISOHybrid media, Linux distro maintainers still haven't switched to UEFI-only, which means that, as long as distro maintainers appear to disprove your theory that using dual BIOS+UEFI is an impairment, I see very little reason not to follow suit in Rufus, and default to dual BIOS+UEFI media for Linux.

> and stop users from using mbr which is useless in this day and age

[Citation needed]

You do realize that MBR is as much part of the UEFI specs as GPT, do you? Considering that you're unlikely to hit the limitations that might make you want to switch to GPT for bootable removable media, I really don't see what switching away from MBR and dual BIOS + UEFI will bring to users.

&rarr; rejected.

closed 2022-08-31 by pbatard

RyzenDew, 2022-08-31:
Just because you don’t see it doesn’t mean it’s not an issue in linux discords we have to go through it 100 times a day because people go to those who can help so closing it because you are not having this as an issue is near sighted at best. Join more linux discords have a presence in the community. Sent from my iPhone

→ rejected.