pbatard/rufus #2035: Alt linux

trimechee, 2022-09-06:
Thank you so much for this awesome and fantastic software  ! i just have a technical question about rufus please, rufus is famous and good bur surprisingly on the web site of ALT Linux which is a russian linux distro, alt linux says in the installation guide: "

Writing hybrid ISO images to boot media
It's not recommended to use UNetbootin, Rufus or UltraISO as these are too "smart" and actually cripple the resulting media to barely bootable or even unbootable state.

It is recommended to use the following utilities.

USB flash drives
Linux: SUSE Studio Imagewriter (imagewriter package) or direct dd use
Windows: Win32 Disk Imager, HDD Raw Copy Tool and the likes"

so if we want to install a light linux distro like antix linux on a usb  flash drive, we shouldn't use rufus? and if xe want to install windows 7 or windows 10 on a usb  flash drive , we should not use rufus ?!

https://en.altlinux.org/Write

pbatard, 2022-09-06:
It's just distro maintainers putting all their eggs into the "ISOHybrid = DD" basket and [failing to realize that DD is not a panacea](https://github.com/pbatard/rufus/wiki/FAQ#Why_doesnt_Rufus_recommend_DD_mode_over_ISO_mode_for_ISOHybrid_images_Surely_DD_is_better) and that they should __actively__ test and make their media compatible with ISO mode or, at the very least, with EFI File system transposition (see the GRUB link below for an explanation of what _File system transposition_ is).

The Arch maintainers understand this and make sure that their media can boot in ISO mode. The Debian maintainers understand this and make sure their media can boot in ISO mode. The Ubuntu maintainers understand this and make sure their media in ISO mode.

But of course, since doing this properly does require some involvement (such as making sure your bootloader supports FAT32 and its limitation in terms of volume label for instance), it is __much easier__ for _some_ distro maintainers to declare that it's not their distro that's the problem but the utility being used to create the media, and promote the idea that some bootable media creators will "cripple the resulting media".

I will point out however that, open minded distro maintainers (i.e. the ones who actually want to make the experience better for users) usually do understand, once it is explained to them what ISO mode really is, that it'll only prevent boot __if__ the maintainers have only done half their job by thinking a Linux ISO image should been seen as a glorified DD image (because for one thing, an open minded distro maintainer should understand that there are very legitimate usage scenarios where one would just want to extract the ISO content to a FAT32 partition and expect the resulting media to boot on a UEFI system, i.e. in File system transposition mode, as it can open the door to single media OS installation, which ISOHybrid will never achieve) and that there's no reason why Linux should be inferior to what Windows ISOs have been supporting for a very long time indeed. For more on that topic, you may also want to see the points [I also made on the GRUB mailing list](https://lists.gnu.org/archive/html/grub-devel/2022-06/msg00024.html), since I am also trying to address some of the default shortcomings that the current version of GRUB 2.0 has in that respect, and that result in Linux distro maintainers currently having to perform extra work when using GRUB in order to have proper ISO mode support.

I can also point out to the Red Hat distro maintainers, who, on hearing that people who used Rufus were unable to install the OS, were also initially quick to jump on the _"Rufus' default mode cripples the boot media"_ bandwagon but, once it was explained to them what ISO mode really is, [realized that they did actually introduce a bug/regression in their installer](https://github.com/rhinstaller/anaconda/pull/3529#issuecomment-925816048) that they wanted to fix.

However, I genuinely don't have the time to go around each distro trying to explain to their maintainers that, despite how it may seem from a Linux-centric perspective, ISOHybrid does have flaws as well as tell them that should perform __extra work__ (but work that will benefit their end users) to ensure that their media/installer does work in both DD mode and ISO/File System transposition mode, especially when they have already publicly decided on a narrative of "Rufus = bad" to present to their users. In fact, there are distros out there that I will not name, that have been resolutely hostile to that message and stuck to the short-sighted _"Rufus is crippling the media"_ view without even trying to consider how providing an alternate way of creating media besides DD could be hugely desirable for their end user.

So my take on this is: Distro maintainers have more than enough of information at their disposal to help them realize that "ISOHybrid = DD" is reductive and will leave some of their users stranded. But of course, understanding this may require a maintainer to look a bit further than their immediate Linux-centric echo chamber. As such, and since there are only so many ways I can tell a distro maintainer _"You are __not__ actually helping your users there"_, if they want to continue promoting the false idea that Rufus "cripples" their installation media, even after having heard the arguments in favour of __providing users with the choice to use either mode__ or precedents about the alleged crippling usually being introduced by the Linux installer itself, on account that restricting to a single option (DD) does make their own maintainer lives easier, so be it.

PS:, I have not tested Alt Linux yet, so I don't know what the specific issue is in this case. The fact that it's using RPMs leads me to think that it may be Red Hat based and suffer from the RH anaconda installer bug, in which case I may add it to the [list of Red Hat derivatives that are __restricted__ to DD-mode only](https://github.com/pbatard/rufus/blob/master/src/rufus.c#L1260-L1273), but that's only a guess for now... I'll know more once my Alt ISO has finished downloading.

trimechee, 2022-09-06:
Woahh this is very interesting and instructive! thank you very much for these very useful and beneficial technical explanations and of course we love Rufus which makes the lives of millions of people easier and more comfortable for free 🥇 💯 

I asked the question thus Mr Pete has the opportunity to respond to these allegations of this russian distro ! thank you very much and we hope that you will put a sponsor button so that people can express their gratitude and thanks for this great project exceptional unique essential for our world !

closed 2022-09-06 by trimechee

trimechee, 2022-09-06:
@pbatard Just for information, distrowatch says alt linux is Independent (forked from Mandrake), and alt linux has several variants but I think what people are interested in is the version intended for normal use called "Simply Linux" :

https://getalt.org/en/simply/

https://distrowatch.com/table.php?distribution=alt

but I notice that simply linux alt doesn't say that rufus can't install the distro, so maybe alt linux was talking about other versions of alt linux regarding the so called incompatibility:

"Use this guide to write the ISO image onto blank DVD; x86 ISO images are also suitable for USB flash drives (NB: UNetbootin and UltraISO cripple the resulting media rendering it unbootable).
Note that aarch64 rootfs archives presume the use of alt-rootfs-installer utility to handle target platform differences; the guide is currently available in Russian."

https://getalt.org/en/

pbatard, 2022-09-06:
Yeah, I'm still trying to find a bit more, so I am downloading `alt-workstation-10.0-x86_64.iso` for now, which is listed as the latest official Alt release. Unfortunately, both the direct servers and bittorrent peers for this specific image seem a bit slow, so I'm still 1.5 hours from a complete download...

I'll take a look at Simply later. Thanks for the pointers.

trimechee, 2022-09-06:
Great ! personally, if one day  i have to use linux instead of windows 10, it would be antix distro beause it is light distro or linux mint, i see that alt linux weighs 4Gb which is huge compared to other light distro that weigh 1Gb or less, if the download is slow, maybe it will be enough to just contact the developers of alt linux and check if the problem is with alt linux and not Rufus and ask them to remove what they say about Rufus....

in any case thank you so much for this magnificent software and for your precious help :)

pbatard, 2022-09-06:
Well I tested `alt-workstation-10.0-x86_64.iso` (in ISO mode) and while the Live session works fine through Isolinux/Syslinux (i.e. in BIOS/Legacy boot), for GRUB (i.e. UEFI mode) the process complains that it can't find the cdrom media.

All in all, it looks to me like the Alt maintainers somehow choose different options when booting their kernel from GRUB compared to Syslinux and appear to be relying on locating a media with an ISO9660 file system, which of course is not going to happen outside of DD mode.

Now, I am not planning to look at it more closely than this, but I will point out that if a media created by Rufus fails to boot properly in UEFI mode, it means that it also is not going to boot in File system transposition mode, __which anybody can create a media for without the need to use Rufus or any third party tool__ (by creating a FAT32 partition on a USB drive and extracting all the ISO files there because as long as you don't have a single file larger than 4 GB, which Alt hasn't, __that is all UEFI should take to boot a media__) and to me, it will always be a failure of the distro maintainers not to properly support File system transposition, especially, again, as this is something that Windows has supported for even longer than Linux has started to rely on ISOHybrid...

Now, I will grant to distro maintainers that the current GRUB 2.0 makes properly supporting File system transposition require some extra work, that shouldn't be needed in the first place, which is something I am actively trying to get GRUB to fix. But the lack of testing or even remotely understanding while File system transposition support __is__ important to have as a feature (I mean, for all its flaws the one thing UEFI did right was to make the boot process move away from the need to write bootloaders at the block level, to instead write them at the file system level, which somehow the Linux community choose to completely disregard in order to run head first into the first solution that was proposed to them that pretty much ensures that end users must go back to writing boot media at the block level) demonstrates, much more eloquently than anything I can write, that it is not the utility that is being used to write the media that is flawed, but the way in which the media has been mastered, and that is it really up to the distro maintainers to fix their ISO.

trimechee, 2022-09-07:
Ouahh thank you somuch  for testing, this is very interesting and instructive ! the Alt linux developers should read your very important advices and recommandations !! 

@altlinux

https://github.com/altlinux/ALTMediaWriter/issues/37