pbatard/rufus #2039: Unable to write to USB explorer because of a explorer.exe error

Windows10User2, 2022-09-09:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [X] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Every time I try to write ISO files (n this case, Windows 11's installers') to USB flash drives in +3.9 Rufus versions, a window opens during the process stating explorer.exe is being used and Rufus is unable to finish writing to the USB flash drives. Many users complaing about this problem in the internet in the latest versions of Rufus so I don't know why wasn't it fixed yet.

Log
---
```
<FULL LOG>
```

pbatard, 2022-09-09:
> so I don't know why wasn't it fixed yet.

Because it's not explorer but another process that you need to identify yourself.

Rufus tries to provide a list of processes that are accessing the drive when it detects that it's unable to write to it because another application has taken a lock, but there's only so many processes Windows allows Rufus to list.

Explorer is usually __always__ listed as a process, because the Windows file explorer will obviously be displaying the drive, but you will usually see it flagged as `r` which means read-only, and therefore it is not the culprit. But we do list it for completion. The problem is that, unlike what is the case on Linux and other OSes, Windows allows some processes to operate at the system level and not be listed, so you end up ion the situation where, when an application like Rufus tries to help its users try to find out what application is preventing Rufus from writing to a drive, the reality is as follows:

* `explorer.exe (r)` &rarr; The WIndows File Explorer application, which only has the drive as read-only and is __NOT__ an issue
* `". . ." (rw)` &rarr; The __ACTUAL__ application responsible for locking the drive, that operates as the system/kernel level and that Windows will not provide the name of when attempting to list processes.

The result: Users like yourselves think that Rufus is complaining about `explorer.exe` accessing the drive, whereas all Rufus does is give you a __NON EXHAUSTIVE__ list of the processes it can list that are accessing the drive to try to help you the one that is the culprit, it case it can help.

Considering that you chose not to attach a log, I'm going to __assert__ that you ran into [this](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) in which case you probably want to check [this](https://github.com/pbatard/rufus/wiki/FAQ#wiki-Incompatible_Software), paying attention, there again, to the fact that the list provided is __NOT AN EXHAUSTIVE ONE__ instead of jumping to the conclusion that Rufus is doing a bad job.

There's only so much I can do to try to help you when an external application is preventing access to a USB drive, which is what I do when I am trying to list all the processes that have read or write access to the drive (which __at least__ `explorer.exe` almost always has), so, before pointing the finger at the application, you need to understand that there may be more that needs to be interpreted from what is being reported to you, especially when Microsoft has decided that even applications running with elevated privileges should not have the ability to identify all the processes that are running on a computer and that are accessing a drive.

Oh, and especially if you want to complain, the log is __NOT__ optional.

closed 2022-09-09 by pbatard

Windows10User2, 2022-09-10:
> > so I don't know why wasn't it fixed yet.
> 
> Because it's not explorer but another process that you need to identify yourself.
> 
> Rufus tries to provide a list of processes that are accessing the drive when it detects that it's unable to write to it because another application has taken a lock, but there's only so many processes Windows allows Rufus to list.
> 
> Explorer is usually **always** listed as a process, because the Windows file explorer will obviously be displaying the drive, but you will usually see it flagged as `r` which means read-only, and therefore it is not the culprit. But we do list it for completion. The problem is that, unlike what is the case on Linux and other OSes, Windows allows some processes to operate at the system level and not be listed, so you end up in the situation where, when an application like Rufus tries to help its users try to find out what application is preventing Rufus from writing to a drive, the reality is as follows:
> 
> * `explorer.exe (r)` → The WIndows File Explorer application, which only has the drive as read-only and is **NOT** an issue
> * `"? ? ?" (rw)` → The **ACTUAL** application responsible for locking the drive, that operates as the system/kernel level and that Windows will not provide the name of when attempting to list processes.
> 
> The result: Users like yourselves think that Rufus is complaining about `explorer.exe` accessing the drive, whereas all Rufus does is give you a **NON EXHAUSTIVE** list of the processes it can list that are accessing the drive to try to help you the one that is the culprit, it case it can help.
> 
> Considering that you chose not to attach a log, I'm going to **assert** that you ran into [this](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) in which case you probably want to check [this](https://github.com/pbatard/rufus/wiki/FAQ#wiki-Incompatible_Software), paying attention, there again, to the fact that the list provided is **NOT AN EXHAUSTIVE ONE** instead of jumping to the conclusion that Rufus is doing a bad job.
> 
> There's only so much I can do to try to help you when an external application is preventing access to a USB drive, which is what I do when I am trying to list all the processes that have read or write access to the drive (which **at least** `explorer.exe` almost always has), so, before pointing the finger at the application, you need to understand that there may be more that needs to be interpreted from what is being reported to you, especially when Microsoft has decided that even applications running with elevated privileges should not have the ability to identify all the processes that are running on a computer and that are accessing a drive.
> 
> Oh, and especially if you want to complain, the log is **NOT** optional.

I can't also run Rufus twice, i.e., nothing happens after running it for the second time after closing it the first time. I have to download it again and if I want to use it again after running and closing it, I have to redownload it and so on.

pbatard, 2022-09-10:
> Where can I find Rufus' log?

Per the check list that you quoted but chose to disregard:

> - [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.

The Log button can be found left of the `START` button.

> Also, I can't run Rufus twice, i.e., nothing happens after running it for the second time after closing it the first time. I have to download it again and if I want to use it again after running and closing it, I have to redownload it and so on.

Then you are the __only__ person to ever report such behaviour, so chances are that you are missing something, or there's something very specific to your environment or the manner in which you run Rufus.

If you really do find that Rufus doesn't launch the second time, please install and run [DebugView](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview) and copy/paste the log you get from Rufus there (since Rufus also sends its log to the global Windows debug facility) as it should tell precisely what is happening.

Please do not assume that your issues are widespread ones that everybody else is facing, and that we are somehow choosing not to do anything about. Instead, consider that chances are that you are in a __minority__ of people facing these issues and that, if you really want to help us address them, you should follow the check list and especially provide a log when requested, as this is the __ONLY WAY__ we can get enough insight into what is really going on in your environment, and potentially address it.

Windows10User2, 2022-09-10:
> > Where can I find Rufus' log?
> 
> Per the check list that you quoted but chose to disregard:
> 
> > * [ ]  I clicked the 'Log' button or pressed Ctrl-L in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
> 
> The Log button can be found left of the `START` button.
> 
> > Also, I can't run Rufus twice, i.e., nothing happens after running it for the second time after closing it the first time. I have to download it again and if I want to use it again after running and closing it, I have to redownload it and so on.
> 
> Then you are the **only** person to ever report such behaviour, so chances are that you are missing something, or there's something very specific to your environment or the manner in which you run Rufus.
> 
> If you really do find that Rufus doesn't launch the second time, please install and run [DebugView](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview) and copy/paste the log you get from Rufus there (since Rufus also sends its log to the global Windows debug facility) as it should tell precisely what is happening.
> 
> Please do not assume that your issues are widespread ones that everybody else is facing, and that we are somehow choosing not to do anything about. Instead, consider that chances are that you are in a **minority** of people facing these issues and that, if you really want to help us address them, you should follow the check list and especially provide a log when requested, as this is the **ONLY WAY** we can get enough insight into what is really going on in your environment, and potentially address it.

I mean, in Windows, where can I find the LOG? What's the directory? I disregarded because I don't even know where it is in Windows, don't you think? And I did search for this problem and I didn't get results unlike what you stated.

But again, where's Rufus' log in Windows? Why should I use DebugView if I don't even know where Rufus' log is?

Since when I assumed this second problem was widespread, and again, how can I post Rufus' log if I don't even know where it is? That's why I didn't check it in the first place, don't you think?

pbatard, 2022-09-10:
> I disregarded the LOG because I don't even know where it was, don't you think?

You are supposed to go through each step of the check list as you create the issue. According to your description, your issue is something that you can replicate every time, which means that you can access the log from Rufus, either through the `Log` button (there are tooltips over each button that indicate what they are for) or using <kbd>Ctrl</kbd>-<kbd>L</kbd> with your keyboard. So what you are supposed to do is: Replicate the error you describe in Rufus, and once you see the error, access the log as described by the check list, as you will be able to save or copy it, so that you can attach it to the issue.

I really cannot stress enough how there really isn't much I can do if you don't provide the log, and I believe that, even if English isn't your first language, the steps described in the check list should be enough to guide you through accessing and copying the log.

> Since when I assumed this second problem was widespread?

Let me quote what you wrote when you opened the issue:

> Many users complain about this problem in the internet affecting the latest versions of Rufus so I don't know why wasn't it fixed yet.

This pretty much implies that you consider that the problem is widespread, since you assert that if affects "many users" and furthermore, you imply that because you assert that it is a widespread issue, it should have been fixed by now.

There aren't that many ways to interpret your statement above besides concluding that you believe that many people are affected by the issue (in other words, that the problem is widespread) and that you believe that the developer of the application is doing a bad job as a result, because they appear not to have fixed an issue that you do allege is widespread.

> but also english isn't even my native language.

That's fine. From what I am seeing, your english should be good enough to follow the check list and access the log so that you can provide its data. The check list gives you two different methods of accessing the log, and repeats, __in 3 different locations and 3 different ways__ (_"I clicked the Log button..."_, _"The log I am copying..."_, _"`<FULL LOG>`"_) how you are required provide a log, so I'm afraid that, even if your English isn't very good, you should at least get some idea that providing a log is required if you want your issue to be addressed properly.

Windows10User2, 2022-09-10:
> > I disregarded the LOG because I don't even know where it was, don't you think?
> 
> You are supposed to go through each step of the check list as you create the issue. According to your description, your issue is something that you can replicate every time, which means that you can access the log from Rufus, either through the `Log` button (there are tooltips over each button that indicate what they are for) or using Ctrl-L with your keyboard. So what you are supposed to do is: Replicate the error you describe in Rufus, and once you see the error, access the log as described by the check list, as you will be able to save or copy it, so that you can attach it to the issue.
> 
> I really cannot stress enough how there really isn't much I can do if you don't provide the log, and I believe that, even if English isn't your first language, the steps described in the check list should be enough to guide you through accessing and copying the log.
> 
> > Since when I assumed this second problem was widespread?
> 
> Let me quote what you wrote when you opened the issue:
> 
> > Many users complain about this problem in the internet affecting the latest versions of Rufus so I don't know why wasn't it fixed yet.
> 
> This pretty much implies that you consider that the problem is widespread, since you assert that if affects "many users" and furthermore, you imply that because you assert that it is a widespread issue, it should have been fixed by now.
> 
> There aren't that many ways to interpret your statement above besides concluding that you believe that many people are affected by the issue (in other words, that the problem is widespread) and that you believe that the developer of the application is doing a bad job as a result, because they appear not to have fixed an issue that you do allege is widespread.
> 
> > but also english isn't even my native language.
> 
> That's fine. From what I am seeing, your english should be good enough to follow the check list and access the log so that you can provide its data. The check list gives you two different methods of accessing the log, and repeats, **in 3 different locations and 3 different ways** (_"I clicked the Log button..."_, _"The log I am copying..."_, _"`<FULL LOG>`"_) how you are required provide a log, so I'm afraid that, even if your English isn't very good, you should at least get some idea that providing a log is required if you want your issue to be addressed properly.

But like I said, I assumed it was widespread for the first problem (not being abtle to write to USB) but not for the second one (unable to run Rufus).

Anyway, I guess one can find out in the log what causes the first issue but are you sure about the second one?

I didn't post the log because I didn't know where to find it but when I have ir, should I post it in the first post?

pbatard, 2022-09-10:
```
Extracting: D:\autorun.inf (128 bytes)
  Unable to create file: Windows error code 0x00000003 (FormatMessage error code 0x00003AFC)
```

You most likely have a poorly designed security solution (Antivirus, Enterprise security software, etc.) that __wrongly__ considers that all `autorun.inf` files are malicious and therefore prevent them from being created.

Unfortunately, Rufus cannot tell you what it is, but you need to find it, and either disable its `autorun.inf` rule or report to its developers that there are very legitimate reasons to want to create an `autorun.inf` and that they really should not prevent third applications from creating one. For more on this, see [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#why-does-rufus-create-an-autoruninf-file-isnt-that-dangerous).

Note that there is nothing Rufus can do about this, as this specific `autorun.inf` comes from the Windows ISO, so you will have to solve this with the security solution that is actively preventing Rufus from writing the `autorun.inf` file.

>  but regarding the second issue, how can I access the log if I can't even open Rufus?

I explained earlier how you should install DebugView and run it, as Rufus will also send its log there, regardless of whether you see the Rufus UI or not.

Please install and run [DebugView](https://docs.microsoft.com/en-us/sysinternals/downloads/debugview). When you run it, and when you launch Rufus, you will see its log starting with `*** Rufus init ***`. If you copy/paste the rest of the log, it should give us an idea of where Rufus appears to hang.

Windows10User2, 2022-09-10:
> ```
> Extracting: D:\autorun.inf (128 bytes)
>   Unable to create file: Windows error code 0x00000003 (FormatMessage error code 0x00003AFC)
> ```
>
> You most likely have a poorly designed security solution [...]

I don't even have any antivirus solution installed.

How can I use the cheat code?

> ```
> Extracting: D:\autorun.inf (128 bytes)
>   Unable to create file: Windows error code 0x00000003 (FormatMessage error code 0x00003AFC)
> ```
>
> You most likely have a poorly designed security solution [...]

I don't even have an antivirus solution installed (not even Microsoft Defender) so how can I use a cheat code to bypass this explorer.exe error?

After running DebugView, I ran Rufus, it didn't open and no "Rufus" entries showed up in DebugView and if I do this when Rufus is working, I guess DebugView will show no errors.

pbatard, 2022-09-10:
> I don't even have any antivirus solution installed.

It doesn't have to be an antivirus. It can be any type of security rules that were enabled on your computer.

> how can I use a Rufus cheat code to bypass this explorer.exe error?

1. `explorer.exe` has nothing to do with your issue. It's just that, when Rufus tries to help you to list applications that can __potentially__ be responsible from preventing access, it will usually list `explorer.exe` because most of the times, Windows File Explorer will have a letter opened for the USB drive in read-only access.
2. As I explained above, there is no "bypass" that Rufus can apply here. I can guarantee that you have a third party application or set of security rules or whatever that is actively preventing Rufus from creating an `autorun.inf` file, and it is up to you to find it and disable it, because Rufus expects to be able to create `autorun.inf`. I can also guarantee you that on a freshly installed Windows system, this issue will not occur, so you do have something that is different from a freshly installed Windows system, that is preventing third party applications like Rufus from creating the `autorun.inf` file. Rufus expects a system where it can create an `autorun.inf` file. If you cannot provide such a system, then I can't support you, because it is not my job to work around systems that have been hardened in a manner that prevents Rufus from doing its job.

> it didn't open, no "Rufus" entries showed up in DebugView

Then either you didn't configure DebugView properly or you have an issue that is completely unrelated to Rufus (such as your third party security solution or security rules or whatever preventing an application that created an `autorun.inf` file from running again, because it __wrongly__ considers it as malicious). There again, there is nothing I can fix, or that I can tell you how to fix, because if you don't even get the `*** Rufus init ***` message, it means that your system prevented the executable from launching altogether, and, obviously, since Rufus is unable to execute any code in that situation, there's really nothing I can do in the code to prevent it.

All in all,, especially as you are using Windows __Enterprise__, which usually means corporate environment that has been hardened outside of your control, I am seeing very strong evidence pointing to your environment being the source of all your issues, and especially, despite your claim that this is not the case, some security solution or some security rules actively interfering with Rufus operations.

If you want to use Rufus, I will ask you to identify what is interfering with it and either remove it or disable it, because there's absolutely nothing else I can do or advise you to do. Your environment __is__ the problem, not Rufus.

Windows10User2, 2022-09-10:
> It doesn't have to be an antivirus. It can be any type of security rules that were enabled on your computer.
>
>
> 1. `explorer.exe` has nothing to do with your issue. It's just that, when Rufus tries [...]

But I do have this issue after a clean install and without any installed program so maybe it's a policy set by NTLite.

I only said I didn't have an antivirus solution installed, not one was the cause of it, and I didn't even mention policies before because you only talked about antivirus solutions back then.

But like I said, why should I had to use DebugView since like I told you I couldn't even open Rufus? I don't understand why you recommended it in the first place.

pbatard, 2022-09-10:
>  why should I had to use DebugView since like I told you I couldn't even open Rufus? I don't understand why you recommended it in the first place.

Because even if Rufus does not display anything on screen, it will send its output to DebugView. The very first line of code that Rufus executes is to send a `*** Rufus init ***` message to the Windows logging facility, which is what DebugView display.

An application "not opening" (i.e. not showing any window onscreen) does not mean that it isn't running. So the fact that you don't get any output in DebugView tells me that it's not an issue with Rufus window initialisation (which could happen, and could be a problem that may be resolved in the Rufus code), but something external that prevents Rufus from launching altogether, in which case, there is nothing I can do about your problem, because it is cause by your system actively preventing Rufus from running altogether.

By the way, since you appear to indicate that you performed the installation of Windows Enterprise on your own, can you explain how you managed to get a license for Windows Enterprise outside of a corporate environment? My understanding is that Microsoft does not sell individual licenses of Windows Enterprise to end users, unless they are being purchased for a multi-license corporate environment. So the fact that you appear to indicate that you have been performing a fresh installation of Windows Enterprise, on your own, outside of a corporate environment is a bit puzzling...

Windows10User2, 2022-09-10:
> > why should I had to use DebugView since like I told you I couldn't even open Rufus? I don't understand why you recommended it in the first place.
> 
> Because even if Rufus does not display anything on screen, it will send its output to DebugView. The very first line of code that Rufus executes is to send a `*** Rufus init ***` message to the Windows logging facility, which is what DebugView display.
> 
> An application "not opening" (i.e. not showing any window onscreen) does not mean that it isn't running. So the fact that you don't get any output in DebugView tells me that it's not an issue with Rufus window initialisation (which could happen, and could be a problem that may be resolved in the Rufus code), but something external that prevents Rufus from launching altogether, in which case, there is nothing I can do about your problem, because it is cause by your system actively preventing Rufus from running altogether.
> 
> By the way, since you appear to indicate that you performed the installation of Windows Enterprise on your own, can you explain how you managed to get a license for Windows Enterprise outside of a corporate environment? My understanding is that Microsoft does not sell individual licenses of Windows Enterprise to end users, unless they are being purchased for a multi-license corporate environment. So the fact that you appear to indicate that you have been performing a fresh installation of Windows Enterprise, on your own, outside of a corporate environment is a bit puzzling...
> 
> > you only talked about antivirus solutions back then.
> 
> No, I was very careful to talk about "security solutions", and mentioned antiviruses as an example of a security solution.
> 
> My words were:
> 
> > You most likely have a poorly designed security solution (Antivirus, Enterprise security software, etc.)
> 
> "Antivirus" is an example of a security solution, just like "Enterprise security software" (which is different from antivirus) is another. Please don't distort what I have been saying to pretend that I only talked about antivirus, when it can be easily demonstrated that this is not the case at all.

I downloaded it from UUP Dump and I also have this problem in Windows 11 Home and Pro, anyway.

But you also mentioned security rules and I have no security solution. What do you mean by Enterprise security software? Also, I don't understand what does Enterprise have to do with this issue as well as where did I get it from.

pbatard, 2022-09-10:
From your log:

```
Windows version: Windows 11 Enterprise, 64-bit (Build 22000.918)
```

You have been running Rufus on Windows 11 Enterprise.

At any rate, as I have pointed above, the issue is clearly external to Rufus. The log clearly shows that something is interfering with writing an `autorun.inf` file and this is a common enough issue to point to the finger at an external security solution (or security rules if you prefer, though it's hard to pinpoint what the actual culprit is, which is why I prefer to use the generic term "security solution"). As such, I will say once again that the problem is 100% yours to fix, because there is literally nothing I can do in Rufus to solve an issue that exists purely on your system (and, considering that this issue tracker is not full of similar issues __only__ on your system).

Windows10User2, 2022-09-10:
> From your log:
>
> ```
> Windows version: Windows 11 Enterprise, 64-bit (Build 22000.918)
> ```
>
> You have been running Rufus on Windows 11 Enterprise.
>
> At any rate, as I have pointed above, the issue is [...]

I know I've been running Windows 11 Enterprise but like I said, I also have this problem in Windows 11 Home and Windows 11 Pro. 

Do you think disabled AutoPlay may cause this problem?

pbatard, 2022-09-10:
> but like I said, I also have this problem in Windows 11 Home and Windows 11 Pro.

The only thing I can accept is a log. And you will see, from this very issue tracker, that you are the only person running Windows 11 (Home, Pro or Enterprise) to report such an issue. For the record, I am running Windows 11 Pro and have never EVER seen the problem you mention. I have however seen many issues in the past where Rufus failed when writing `autorun.inf` and __EVERY SINGLE ONE OF THEM__ was caused by a security solution.

> Do you think disabled AutoPlay may cause this problem?

No.

Windows10User2, 2022-09-10:
> > but like I said, I also have this problem in Windows 11 Home and Windows 11 Pro.
> 
> The only thing I can accept is a log. And you will see, from this very issue tracker, that you are the only person running Windows 11 (Home, Pro or Enterprise) to report such an issue. For the record, I am running Windows 11 Pro and have never EVER seen the problem you mention. I have however seen many issues in the past where Rufus failed when writing `autorun.inf` and **EVERY SINGLE ONE OF THEM** was caused by a security solution. The fact that Windows Enterprise is often run in corporate environments that use such restrictive security solution is why I am very interested about the fact that your log showed that you were running Enterprise, because that is exactly where I would expect this kid of problem to manifest itself due to corporate security rules being enforced by your corporate sysadmins.
> 
> > Do you think disabled AutoPlay may cause this problem?
> 
> No.

If I also have this problem in Home and Pro, do you think it might be related to UAC?

pbatard, 2022-09-10:
> If I also have this problem in Home and Pro, do you think it might be related to UAC?

No. It __is__ an active security solution or security rules preventing the creation of `autorun.inf`. And this will be my last rely before I lock this thread, because unless somebody else comes on this issue tracker with the same issue on an __unmodified__ official installation of Windows 11, it is 100% your problem to identify and fix, and, considering that this is external to Rufus, I consider that I've already given you enough of my time

pbatard, 2022-12-16:
Also, per #2115, this may actually be due to your computer being infected by a __virus__, since some viruses try to use `autorun.inf` to spread. For instance, it appears that `Virus.Floxif` can cause this error.

And again, I have to stress out that the issue is 100% with the environment, and not with Rufus. If you do run Rufus from a __clean__ installation of Windows, without dodgy software or security solutions, you will not run into this problem.