pbatard/rufus #2043: Error during installation: DRIVER IRQL NOT LESS OR EQUAL

mohsen67t, 2022-09-12:
When installing Windows and after choosing the partition for installation, everything went well, the Windows files were copied and the installation was done, and Windows automatically restarted, but I encountered this error in the second step of the installation.

Error: Your PC ran into a problem and needs to restart.
Stop Code: DRIVER IRQL NOT LESS OR EQUAL
What Failed: rt640x64.sys

[image]

pineapple63, 2022-09-12:
First thing i would do is verify that the ISO you are using is genuine, and has not been modified or gotten corrupted in some way

If you used a media creation tool ISO, then that could be the source of the problem (as i have had strange things happen with media creation tool ISOs)

mohsen67t, 2022-09-13:
I downloaded this ISO file directly from the Microsoft site

This URL: https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewiso
File name: Windows11_InsiderPreview_Client_x64_en-us_25197.iso

And I put the ISO file in the flash memory without any manipulation through the Rufus software

[image]

pbatard, 2022-09-13:
Please post about this issue on a Windows user forum.

Windows installation problems such as this one have nothing to do with Rufus, since Rufus does not alter any files from Windows (not even for the installer).

Therefore, you __will__ encounter the same issue regardless of whether you used Rufus to install Windows or not, which means that this issue should not be reported in the Rufus issue tracker, as you will not obtain the kind of help you are hoping to get, since this is not a Windows installation help forum.

As such I will close this issue as it is off-topic.

closed 2022-09-13 by pbatard