pbatard/rufus #2046: Publish to WIndows Package Manager

vedantmgoyal2009, 2022-09-17:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
With every new release of MediaWriter, use a GitHub workflow to update manifests at https://github.com/microsoft/winget-pkgs repository.

I can create a pull request if you would like. It would help in keeping manifests up-to-date at WinGet with every new release automatically.

Existing projects which use the action: https://github.com/search?q=vedantmgoyal2009%2Fwinget-releaser+path%3A.github%2Fworkflows%2F+language%3AYAML&type=Code&s=indexed&o=desc

Log
---
```
<FULL LOG>
```

pbatard, 2022-09-17:
1. I don't use WinGet as part of any of my workflows, and I don't see that changing anytime soon. So it'll be difficult for me to maintain something that I don't see myself use at all in the near to medium term.
2. WinGet has a [rather __shady__ birth story](https://www.thurrott.com/windows/windows-10/235783/appget-creator-says-microsoft-stole-his-product), which, as an opinionated publisher of Open Source software, makes me quite unwilling to support its microcosm, because I sure don't want to be seen, by endorsing WinGet, as endorsing some less than desirable practices... 
3. Because Rufus runs elevated, I want the final Rufus executable to be signed always, which is not something that can ever be accomplished through GitHub Actions (since there is no way I would ever share my signing key with an automated process and, even if I wanted to, it only resides on a physical USB token, so it can't be shared in the first place). And no, I don't consider a secret winget token, that would reside in the cloud, as safe enough.
4. I already publish Rufus a Windows Store application, and since Rufus is a GUI application, it makes little sense trying to work with a package manager that is aimed at commandline use, when installing from the Store, for those who __must__ have some kind of package manager take care of installation and updates for them (whereas Rufus does have automated update on its own), should be fine in most scenarios.
5. (Since I'd have to manually sign the executable before I can pass it on to a WinGet package generation workflow), I don't want to have to perform extra tasks to an already fairly long list of actions I need to perform for each release.
6. I am not personally seeing that much use of WinGet in the wild. What I do occasionally see are a __limited__ number of people who seem to be almost religious about WinGet, but, unlike what has been the case for the Windows Store or appx packages, I'm really not seeing people claiming that WinGet is _the next big thing_ and that I am going to fail Rufus users unless I start publishing Rufus as a WinGet package.
7. As a self contained __small__ executable with no installer and automated update check and download, for an application that the vast majority of people aren't going to run on a day to basis, I also kinda fail to see the need to repackage the application further. Heck, doing so for the Windows Store already seems like an overkill to me, and I sure don't want to have to do that over and over again for whatever "flavour du jour" package manager comes onto the scene every other year.

As such, I'm afraid I will reject this request.
You are however free to publish and maintain your own WinGet package of Rufus if you wish.

closed 2022-09-17 by pbatard

vedantmgoyal2009, 2022-09-17:
I appreciate your detailed response and the valuable time you've spent writing it. As you've mentioned in the end, I'll add Rufus to my automation which updates several other packages 😃