pbatard/rufus #2049: No Extended Windows 11 Installation Image option in Rufus when selected Win11_22H2_English_x64.iso

palavnik85, 2022-09-20:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here> There is no Extended Windows 11 Installation Image options in Rufus when selected Win11_22H2_English_x64.iso official release.

Log
---
```
<FULL LOG>
Rufus x86 v3.20.1929
Windows version: Windows 11 Pro, 64-bit (Build 22000.978)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'USB Disk 3.0 USB Device' (1F75:0918)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3786, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {66436B71-00D0-4A6B-BF9A-7BE330D4492E}
Max parts: 128, Start Offset: 17408, Usable = 31142672896 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {3FBE5DBE-E513-4FAE-9A56-8FECB73E9704}
  Size: 29 GB (31140555264 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {63138898-2026-4A32-B104-7568F711C104}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60823445, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 22621.1)
  Size: 5.1 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win11_22H2_English_x64.iso (5.1 GB)
```

Swapnanil-Chakraborty, 2022-09-20:
Created iso with the media creation tool 4.15GB. Shows the extended image option.

JonnyTech, 2022-09-20:
Please define "Extended Windows 11 Installation Image options"

ZaneCEO, 2022-09-20:
The answer is here:

Help, I no longer see the option to bypass TPM/Secure Boot with Windows 11! https://github.com/pbatard/rufus/wiki/FAQ#help-i-no-longer-see-the-option-to-bypass-tpmsecure-boot-with-windows-11

closed 2022-09-21 by pbatard