pbatard/rufus #2050: Error Unable to patch/setup files for boot (Win 22H2)

Terule, 2022-09-21:
Checklist
---------
- [ x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ x ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Trying to create a bootable device but it wont work. Error Unable to patch/setup files for boot.  I tried to change USB port, format the drive before using rufus, but got the same error

Log
---
```
Rufus x86 v3.20.1929
Windows version: Windows 10 Pro, 64-bit (Build 19044.2006)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Sony Hard Drive USB Device' (054C:05BF)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 3 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'Sony Hard Drive USB Device' (054C:05BF)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: FIXED, Disk size: 512 GB, Sector size: 512 bytes
Cylinders: 60801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {C1B4CC4B-2635-4ACC-930B-3B76D1A1E061}
Max parts: 128, Start Offset: 17408, Usable = 500107825664 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {D519077C-B78E-42A5-B78C-37A7DC1EF043}
  Size: 465.8 GB (500105700864 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {7AD9A504-3B68-4036-A1AF-194FC734E0FF}
  Size: 1 MB (1048576 bytes)
  Start Sector: 976770995, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 11 ISO (Build 22621.1)
  Size: 5.1 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
[... 212 lines cut ...]
Extracting: G:\sources\dlmanifests\microsoft-windows-fax-server-dl.man (1.8 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-fax-service-dl.man (7.5 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-htmlhelp-dl.man (1.7 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-iasserver-migplugin\iasmigplugin.dll (757.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-iasserver-migplugin\iasmigreader.exe (497.9 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-ie-clientnetworkprotocolimplementation-migration\wininetplugin.dll (37.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-ie-esc-dl.man (1.9 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-ie-feeds-platform-dl.man (2.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-ie-internetexplorer-dl.man (13.0 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-iis-corewebengine-deployment-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-internet-naming-service-runtime\winsplgn.dll (77.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-msmq-messagingcoreservice\mqmigplugin.dll (185.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-netfx35-dl.man (8.8 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-netfx35cdfcomp-dl.man (1.2 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-netfxcorecomp-dl.man (44.1 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-networkbridge\bridgemigplugin.dll (133.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-networkloadbalancing-core\nlbmigplugin.dll (137.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-rasconnectionmanager\cmmigr.dll (97.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-rasserver-migplugin\rasmigplugin.dll (413.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-storagemigration-dl.man (8.0 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-tapisetup\tapimigplugin.dll (93.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-tapisetup-dl.man (2.5 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-terminalservices-clientactivexcore-dl.man (1.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-terminalservices-licenseserver\tlsmigplugin.dll (137.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-textservicesframework-migration-dl\chxmig.dll (169.4 KB)

Terule, 2022-09-21:
Extracting: G:\sources\dlmanifests\microsoft-windows-textservicesframework-migration-dl\imjpmig.dll (213.4 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-textservicesframework-migration-dl\imkrmig.dll (137.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-textservicesframework-migration-dl\msctfmig.dll (161.4 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-textservicesframework-migration-dl\tabletextservicemig.dll (109.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-unimodem-config\modemmigplugin.dll (117.4 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-wab-dl.man (2.5 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-wcfcorecomp-dl.man (4.8 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-winsock-core-infrastructure-upgrade\wsupgrade.dll (241.3 KB)
Extracting: G:\sources\dlmanifests\microsoft-windows-wmi-core\wmimigrationplugin.dll (697.3 KB)
Extracting: G:\sources\dlmanifests\microsoft.windows.com.base-dl.man (1.8 KB)
Extracting: G:\sources\dlmanifests\microsoft.windows.com.complus.setup.dl.man (2.3 KB)
Extracting: G:\sources\dlmanifests\msmq-dcom-dl.man (1.7 KB)
Extracting: G:\sources\dlmanifests\msmq-domain-ic-dl.man (1.5 KB)
Extracting: G:\sources\dlmanifests\msmq-http-dl.man (2.2 KB)
Extracting: G:\sources\dlmanifests\msmq-messagingcoreservice-dl.man (6.6 KB)
Extracting: G:\sources\dlmanifests\msmq-multicast-dl.man (1.5 KB)
Extracting: G:\sources\dlmanifests\msmq-routing-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\msmq-triggers-dl.man (1.7 KB)
Extracting: G:\sources\dlmanifests\mup-dl.man (1.5 KB)
Extracting: G:\sources\dlmanifests\netfx-wcf-http-activation-dl.man (1.2 KB)
Extracting: G:\sources\dlmanifests\netfx-wcf-msmqactivation-registration-dl.man (1.6 KB)
Extracting: G:\sources\dlmanifests\netfx-wcf-pipeactivation-registration-dl.man (1.6 KB)
Extracting: G:\sources\dlmanifests\netfx-wcf-tcpactivation-registration-dl.man (1.6 KB)
Extracting: G:\sources\dlmanifests\netfx-wcf-tcpportsharing-dl.man (1.1 KB)
Extracting: G:\sources\dlmanifests\netfx3-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\netfx4-wcf-client-dl.man (1.5 KB)
Extracting: G:\sources\dlmanifests\netfx4-wcf-extended-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\netfx4clientcorecomp-dl.man (15.5 KB)
Extracting: G:\sources\dlmanifests\netlogon-dl.man (2 KB)
Extracting: G:\sources\dlmanifests\networking-mpssvc-svc\icfupgd.dll (173.4 KB)
Extracting: G:\sources\dlmanifests\ntfs-dl.man (4.0 KB)
Extracting: G:\sources\dlmanifests\ntoskrnl-dl.man (7.8 KB)
Extracting: G:\sources\dlmanifests\odbc32dll-dl.man (4.7 KB)
Extracting: G:\sources\dlmanifests\partmgr-dl.man (1 KB)
Extracting: G:\sources\dlmanifests\printing-spooler-core-dl.man (2.5 KB)
Extracting: G:\sources\dlmanifests\printing-spooler-networkclient-dl.man (2.2 KB)
Extracting: G:\sources\dlmanifests\rights-management-client-v1-api-dl.man (3.2 KB)
Extracting: G:\sources\dlmanifests\rights-management-services-server-dl.man (3.2 KB)
Extracting: G:\sources\dlmanifests\rpc-http-dl.man (1.6 KB)
Extracting: G:\sources\dlmanifests\rpc-http_proxy-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\rpc-local-dl.man (1.8 KB)
Extracting: G:\sources\dlmanifests\rpc-remote-dl.man (1.9 KB)
Extracting: G:\sources\dlmanifests\schannel-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\schedsvc-dl.man (1.9 KB)
Extracting: G:\sources\dlmanifests\security-digest-dl.man (803 bytes)
Extracting: G:\sources\dlmanifests\security-kerberos-dl.man (803 bytes)
Extracting: G:\sources\dlmanifests\security-ntlm-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\security-ntlm-lmc.man (1.1 KB)
Extracting: G:\sources\dlmanifests\shutdown-event-tracker-dl.man (1.3 KB)
Extracting: G:\sources\dlmanifests\smartcardsubsystem-dl.man (3.7 KB)
Extracting: G:\sources\dlmanifests\smbserver-dl.man (13.7 KB)
Extracting: G:\sources\dlmanifests\smss-dl.man (1.7 KB)
Extracting: G:\sources\dlmanifests\smtpsvc-admin-dl.man (1.1 KB)
Extracting: G:\sources\dlmanifests\smtpsvc-service-dl.man (1.2 KB)
Extracting: G:\sources\dlmanifests\speechcommon-dl.man (2.8 KB)
Extracting: G:\sources\dlmanifests\tabletpc-tabbtn-dl.man (1.7 KB)
Extracting: G:\sources\dlmanifests\tabletpcinputpanel-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\tabletpcjournal-dl.man (8.5 KB)
Extracting: G:\sources\dlmanifests\tabletpcplatforminput-core-dl.man (1.1 KB)
Extracting: G:\sources\dlmanifests\terminalservices-appserver-dl.man (1.5 KB)
Extracting: G:\sources\dlmanifests\terminalservices-appserver-licensing-dl.man (1.3 KB)
Extracting: G:\sources\dlmanifests\terminalservices-drivers-dl.man (1.1 KB)
Extracting: G:\sources\dlmanifests\terminalservices-licenseserver-dl.man (2.2 KB)
Extracting: G:\sources\dlmanifests\terminalservices-localsessionmanager-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\terminalservices-rdp-winstationextensions-dl.man (20.4 KB)
Extracting: G:\sources\dlmanifests\terminalservices-remoteconnectionmanager-dl.man (1.4 KB)
Extracting: G:\sources\dlmanifests\terminalservices-sessiondirectory-client-dl.man (2.2 KB)
Extracting: G:\sources\dlmanifests\terminalservices-sessiondirectory-server-dl.man (2.4 KB)
Extracting: G:\sources\dlmanifests\terminalservices-terminalservicesclient-dl.man (1.9 KB)
Extracting: G:\sources\dlmanifests\textservicesframework-migration-dl.man (2.8 KB)
[... 676 lines cut ...]

Found USB 2.0 device 'Sony Hard Drive USB Device' (054C:05BF)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
[... cut ...]
Max parts: 128, Start Offset: 17408, Usable = 500107825664 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {07FB7877-AB9A-4D5E-A4BC-926DA6AE1D92}
  Size: 465.8 GB (500105700864 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {39A92547-1903-4E3E-9012-4D9AA62C59BC}
  Size: 1 MB (1048576 bytes)
  Start Sector: 976770995, Attributes: 0x9000000000000000

```

pbatard, 2022-09-21:
Please reboot your computer and try again.

pbatard, 2022-09-23:
Since there has been no update, I will assume that rebooting solved your issue and close it.

closed 2022-09-23 by pbatard