pbatard/rufus #2051: what  methode of bypass for windows 11 21H2 do you use

glda19, 2022-09-24:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2022-09-24:
https://www.majorgeeks.com/content/page/bypass_tpm.html

pbatard, 2022-09-24:
For more how exactly we do it, see https://github.com/pbatard/rufus/blob/master/src/wue.c#L685-L736. I'm afraid that I am not going to describe this any further, because all methods variations of creating the `LabConfig` registry keys, and you'll find plenty of documentation on how to do that on the internet.

The one thing to note is that these keys only apply to the installation environment. Once Windows is installed, there's absolutely __NO__ difference between what you get with or without the bypass applied.

closed 2022-09-24 by pbatard