pbatard/rufus #2054: "Error: Write Error" "Error: Inflate Error" when writing images into SDCard

IEatCodeDaily, 2022-09-27:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm trying to write a MainsailOS into an 8GB V-Gen SDCard and keep hitting the same write error after ~25% progress. I suspect the error happens when rufus tries to write into the primary partition, as the boot partition was properly written and can be accessed. I managed to write an OctoPi image previously and planning to try MainsailOS.

I've tried using other images (ubuntu, raspbian, octopi) just to test that the image is not the problem. All other images also resulted in the same error.

Log
---
```
Rufus x86 v3.20.1929
Windows version: Windows 10 Home Single Language, 64-bit (Build 19044.1889)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'WD My Book 25ED USB Device' (1058:25ED)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
1 device found
Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
Cylinders: 966, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x067E19D7
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 256 MB (268435456 bytes)
  Start Sector: 8192, Boot: No
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext4
  Size: 4 GB (4336910336 bytes)
  Start Sector: 532480, Boot: No
Scanning image...
ISO analysis:
  'E:\Installer\OS Image\octopi-0.18.0-klipper-20220109154656\octopi-0.18.0-klipper-20220109154656.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: octopi-0.18.0-klipper-20220109154656.img (4.3 GB)
Scanning image...
ISO analysis:
  'E:\Installer\OS Image\MainsailOS-buster-lite-0.7.1.zip' doesn't look like an ISO image
Disk image analysis:
  Image is a compressed bootable disk image
Using image: MainsailOS-buster-lite-0.7.1.zip (1.2 GB)

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Requesting logical volume handle...
Bad Blocks: Checking from block 0 to 15167 (1 block = 512 KB)
Bad Blocks: Using offset 117153 for fake device check
Bad Blocks: Writing test pattern 0x55
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Writing compressed image:
++++++++++++++++++++++++++++++Error: write error: -1
Error: inflate error

Could not write compressed image: -1
Opened \\.\PhysicalDrive3 for shared write access
Remounted \\?\Volume{028f5b2a-3e2e-11ed-8e5a-5c879cc2bc2a}\ as G:
Re-mounted volume as G: after error

Found USB 3.0 device 'WD My Book 25ED USB Device' (1058:25ED)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
1 device found
Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
Cylinders: 966, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x9BE557D4
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 256 MB (268435456 bytes)
  Start Sector: 8192, Boot: No
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext4
  Size: 4.5 GB (4875878400 bytes)
  Start Sector: 532480, Boot: No
Found USB 3.0 device 'WD My Book 25ED USB Device' (1058:25ED)
Device eliminated because it was detected as a Hard Drive (score 23 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
```

JonnyTech, 2022-09-27:
> Error: inflate error

I suspect that your image file is corrupted, please try downloading it again and verify any checksum that it provides.

JonnyTech, 2022-09-27:
>  'E:\Installer\OS Image\octopi-0.18.0-klipper-20220109154656\octopi-0.18.0-klipper-20220109154656.img' doesn't look like an ISO image

The log is not for MainsailOS as your report states.

>  If using an image, I clicked on the (✓) button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Please perform this step that you claim to have done when completing the issue checklist

IEatCodeDaily, 2022-09-27:
The image is not corrupted, I have done the SHA256 checksum and it checks out.

here's computed checksum,

> Computing checksum for 'E:\Installer\OS Image\MainsailOS-buster-lite-0.7.1.zip'...
>   MD5:    939d53116c9bd01160d55639e9f3b502
>   SHA1:   b1d330399ba320c739913dc952a967fa980f83e4
>   SHA256: ce681d82ea21c27b68ab7e3d2a4a4b23aeb03cd560da0c2f88b1ea2cb5fa097c

here's checksum provided by mainsailOS,

> ce681d82ea21c27b68ab7e3d2a4a4b23aeb03cd560da0c2f88b1ea2cb5fa097c  MainsailOS-buster-lite-0.7.1.zip

  
I did use the mainsailOS, it was preloaded with octopi from my previous try and I loaded mainsailOS after.

> Scanning image...
> ISO analysis:
>   'E:\Installer\OS Image\MainsailOS-buster-lite-0.7.1.zip' doesn't look like an ISO image
> Disk image analysis:
>   Image is a compressed bootable disk image
> Using image: MainsailOS-buster-lite-0.7.1.zip (1.2 GB)

JonnyTech, 2022-09-27:
Can you unzip the file with external tools, then use Rufus to write the .img file and post a new log.

pbatard, 2022-09-27:
Not able to replicate your issue so far with `octopi-0.18.0-klipper-20220109154656.zip` (since that is the image reference I saw in your log), though I do get an `Error: bad length` at the end of extraction, which I believe is innocuous but that I'll try to look into

```
Rufus x86 v3.20.1929
Windows version: Windows 11 Pro for Workstations, 64-bit (Build 22621.521)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found VHD device 'Microsoft Virtual Disk'
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x87
Disk type: FIXED, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1044, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive does not have a Boot Marker
Scanning image...
ISO analysis:
  'D:\Imgs\octopi-0.18.0-klipper-20220109154656.zip' doesn't look like an ISO image
Disk image analysis:
  Image is a compressed bootable disk image
Using image: octopi-0.18.0-klipper-20220109154656.zip (1.1 GB)

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive7 for shared write access
Requesting logical volume handle...
No logical drive found (unpartitioned?)
Writing compressed image:
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++Error: bad length

Opened \\.\PhysicalDrive7 for shared write access
Remounted \\?\Volume{067e19d7-0000-0000-0000-400000000000}\ as F:

Found VHD device 'Microsoft Virtual Disk'
1 device found
Disk type: FIXED, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1044, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x067E19D7
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 256 MB (268435456 bytes)
  Start Sector: 8192, Boot: No
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext4
  Size: 4 GB (4336910336 bytes)
  Start Sector: 532480, Boot: No
```

My guess, since I know that Windows can be very anal about anything that look like an ESP, is that you had/have some kind of ESP on your device, and you are fighting Microsoft's moronic _"Oh, but we cannot let users access an ESP... __EVER!!!!__"_ restriction that they have added to Windows.

So my advice would be:
* Use <kbd>Alt</kbd>-<kbd>Z</kbd> in Rufus to zero the drive
* Regardless of whether it completed or not, reboot your computer completely
* Try writing your image again.

IEatCodeDaily, 2022-09-27:
For now, I'm trying @pbatard advice.
So far, it's writing properly, currently at 40%. Fingers crossed.

It's already nighttime here. I'm going to go to sleep first. Will update tomorrow morning (if I remember to).

IEatCodeDaily, 2022-09-28:
I just completed @pbatard advice. Rufus successfully written the mainsailOS image. here's the log,

> Rufus x86 v3.20.1929
> Windows version: Windows 10 Home Single Language, 64-bit (Build 19044.2006)
> Syslinux versions: 4.07/2013-07-25, 6.04/pre1
> Grub versions: 0.4.6a, 2.06
> System locale ID: 0x0409 (en-US)
> Will use default UI locale 0x0409
> SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
> Localization set to 'en-US'
> Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
> Found USB 3.0 device 'WD My Book 25ED USB Device' (1058:25ED)
> Device eliminated because it was detected as a Hard Drive (score 23 > 0)
> If this device is not a Hard Drive, please e-mail the author of this application
> NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
> Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
> 1 device found
> No volume information for drive 0x83
> Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
> Cylinders: 966, Tracks per cylinder: 255, Sectors per track: 63
> Partition type: SFD (Super Floppy Disk) or unpartitioned
> Scanning image...
> ISO analysis:
>   'E:\Installer\OS Image\MainsailOS-buster-lite-0.7.1\MainsailOS-buster-lite-0.7.1.img' doesn't look like an ISO image
> Disk image analysis:
>   Image has an unknown Master Boot Record
>   Image is a bootable disk image
> Using image: MainsailOS-buster-lite-0.7.1.img (4.8 GB)
> 
> Format operation started
> Requesting disk access...
> No drive letter was assigned...
> Will use 'G:' as volume mountpoint
> Opened \\.\PhysicalDrive3 for exclusive write access
> Requesting logical volume handle...
> +++++++++++++++++++++
> Write error at sector 2621440: [0x000001E3] The request failed due to a fatal device hardware error.
> Retrying in 5 seconds...
> ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
> Opened \\.\PhysicalDrive3 for shared write access
> Remounted \\?\Volume{028f5db5-3e2e-11ed-8e5a-5c879cc2bc2a}\ as G:
> 
> Found USB 3.0 device 'WD My Book 25ED USB Device' (1058:25ED)
> Device eliminated because it was detected as a Hard Drive (score 23 > 0)
> If this device is not a Hard Drive, please e-mail the author of this application
> NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
> Found USB 2.0 device 'MXT-USB Storage Device USB Device' (AAAA:8816)
> 1 device found
> No volume information for drive 0x83
> Disk type: Removable, Disk size: 8GB, Sector size: 512 bytes
> Cylinders: 966, Tracks per cylinder: 255, Sectors per track: 63
> Partition type: MBR, NB Partitions: 2
> Disk ID: 0x9BE557D4
> Drive has an unknown Master Boot Record
> Partition 1:
>   Type: FAT32 LBA (0x0c)
>   Detected File System: FAT32
>   Size: 256 MB (268435456 bytes)
>   Start Sector: 8192, Boot: No
> Partition 2:
>   Type: GNU/Linux (0x83)
>   Detected File System: ext4
>   Size: 4.5 GB (4875878400 bytes)
>   Start Sector: 532480, Boot: No

There was a fatal hardware error but it managed to finish writing in the end.
1st partition isn't read as a boot partition, which is weird. I'll plug it into my rasp pi and test whether it works or no.

IEatCodeDaily, 2022-09-28:
Rasp pi booted, web interface and ssh function properly. @pbatard advice worked like a charm.
I suspect it's the fault of crappy SDcard adaptor. I've tried 2 SDcards and 3 adaptors, they all had the same problem of I/O failure/hardware failure when trying to write .img file directly. All of the adaptors are cheapo backwater $1 adaptors which make sense.
after following @pbatard advice, both SDcards managed to be written successfully.

I'll mark this as closed. Thanks for the help @pbatard @JonnyTech !

closed 2022-09-28 by IEatCodeDaily