pbatard/rufus #2055: Write error [0x00000005] Access is denied.

SibilantScream, 2022-09-28:
Failing to load LTSC.2019.X64.ENU.MAY2022 using either MBR or GPT methods.
Win7 x64.
Flash is normally formattable & acessable.

Log
---
```
SearchProcess: Could not open process 4: Access Denied
SearchProcess: Could not open process 4: Access Denied

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Windows volumes from this device:
● \\?\Volume{8b3eef1d-3e5c-11ed-9d6d-68b599e942ea}\ @0
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
SearchProcess: Could not open process 4: Access Denied
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
SearchProcess: Could not open process 4: Access Denied
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
SearchProcess: Could not open process 4: Access Denied
Write error [0x00000005] Access is denied.
Could not reset partitions
Re-mounted volume as E: after error

Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&1b2f9866&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[061]: USB\VID_8087&PID_0020\5&3DD8D5A&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub20#4&2b14a31&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[158]: USB\VID_8087&PID_0020\5&B12D121&0&1
Processing Hub 4:
Processing Hub 5:
  Hub[3] = '\\?\usb#vid_8087&pid_0020#5&b12d121&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[104]: USB\VID_18F8&PID_0F99\6&2EAD5BA9&0&2
  Found ID[001]: USB\VID_8564&PID_1000\32GJ9OMH17PQH8R4
Processing IDs belonging to 'USBSTOR':
  USB\VID_8564&PID_1000\32GJ9OMH17PQH8R4
Processing 'IDE' device:
  Unsupported or disabled by policy
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskJetFlashTranscend_4GB___1100'
  Matched with ID[001]: USB\VID_8564&PID_1000\32GJ9OMH17PQH8R4
  Matched with Hub[3]: '\\?\usb#vid_8087&pid_0020#5&b12d121&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
1 device found
Windows volumes from this device:
● \\?\Volume{8b3eef1d-3e5c-11ed-9d6d-68b599e942ea}\ @0
Disk type: Removable, Disk size: 4.1GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub20#4&1b2f9866&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[061]: USB\VID_8087&PID_0020\5&3DD8D5A&0&1
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub20#4&2b14a31&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[158]: USB\VID_8087&PID_0020\5&B12D121&0&1
Processing Hub 4:
Processing Hub 5:
  Hub[3] = '\\?\usb#vid_8087&pid_0020#5&b12d121&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[104]: USB\VID_18F8&PID_0F99\6&2EAD5BA9&0&2
  Found ID[001]: USB\VID_8564&PID_1000\32GJ9OMH17PQH8R4
Processing IDs belonging to 'USBSTOR':
  USB\VID_8564&PID_1000\32GJ9OMH17PQH8R4
Processing 'IDE' device:
  Unsupported or disabled by policy
Processing 'USBSTOR' device:
  Hardware ID: 'USBSTOR\DiskJetFlashTranscend_4GB___1100'
  Matched with ID[001]: USB\VID_8564&PID_1000\32GJ9OMH17PQH8R4
  Matched with Hub[3]: '\\?\usb#vid_8087&pid_0020#5&b12d121&0&1#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
Found USB 2.0 device 'JetFlash Transcend 4GB USB Device' (8564:1000)
1 device found
Windows volumes from this device:
● \\?\Volume{8b3eef1d-3e5c-11ed-9d6d-68b599e942ea}\ @0
Disk type: Removable, Disk size: 4.1GB, Sector size: 512 bytes
Cylinders: 492, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned

```

JonnyTech, 2022-09-28:
Post the WHOLE log - it should start with Rufus version number.

>  Win7 x64.

https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

SibilantScream, 2022-09-28:
[Is this official?](https://rufus.ie/en/)
Cuz it says 3.20 requires win7.

JonnyTech, 2022-09-28:
> Is this official?

Yes

> Cuz it says 3.20 requires win7.

@pbatard please reword the page