pbatard/rufus #2057: FAQ first hit for Google, misses a reason I could arrive there

linuxdropout, 2022-10-03:
Just a small thing from the FAQ I came to while fixing a non-rufus issue. But one you might want to add to the FAQ as an edge case for that error message.

The error:

ERROR: BIOS/LEGACY BOOT OF UEFI-ONLY MEDIA

Can be triggered by creating a bootable drive with Rufus.
Then formatting the drive to exFAT with windows 10.
Then trying to boot from that drive.

Some context:
- I was trying to use gigabyte q-flash to update my bios version
- The USB I put the bios for q-flash on was previously a bootable Ubuntu drive created with Rufus
- I formatted the drive using windows explorer and copied the bios onto the drive
- Rebooted and attempted to use q-flash, somehow I was thrown onto the screen with the above error message, 99% sure this is a fault with q-flash somehow trying to boot the drive in this edge case.
- I fixed this by reformatting the drive as non-bootable FAT32 using Rufus which was then usable via q-flash as expected.
- The error appeared despite my bios being UEFI and the drive previously being created as UEFI bootable.

pbatard, 2022-10-03:
> Can be triggered by creating a bootable drive with Rufus.
> Then formatting the drive to exFAT with windows 10.
> Then trying to boot from that drive.

I'm sorry but I'm gonna say: _"Well, duh"_ here.

You should know that formatting a drive in Windows does __not__ remove the bootloaders. It just formats the partition.

So if you formatted the drive in Rufus and it installed the BIOS/LEGACY BOOT OF UEFI-ONLY MEDIA MBR bootloader, you __will__ get the message even if you reformat the drive in Windows.

Instead, you need to follow [this](https://github.com/pbatard/rufus/wiki/FAQ#Help_how_do_I_restore_my_USB_to_how_it_was_before_I_created_a_bootable_drive) to __properly__ restore your drive to non-bootable and remove the bootloaders.

I am therefore going to close this issue as "user misconception of what Windows formatting is supposed to do".

closed 2022-10-03 by pbatard

dylanplural, 2022-10-03:
I think you misunderstood what I meant, my bad for not wording it correctly.

But eh, it's not a big deal, just trying to help out future people chasing their tails googling the same error message I got. Hopefully they'll find this GitHub issue instead.