pbatard/rufus #2059: Problems with Windows To Go on usb-sata bridge adapter

willrabbermann, 2022-10-07:
﻿

Checklist
---------
- [ x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ x ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus Version 3.20.1929
(host) gentoo x86-64 (luks, unified kernel image, secure-boot(bypassed for WTG))
Dell XPS 13 9370
512 gb 2.5" SATA external laptop hdd
2.5"sata-usb3.0 bridge adapter

I have created a Windows To Go (WTG) usb device from a genuine windows 10 iso inside of a working windows 10 pro Virtual Machine with Rufus. I have a 512 gb 2.5" hard drive connected to a sata-usb bridge adapter and passed into the VM. Installing to the drive seems successful.

The Problem:
I can boot into the WTG USB through UEFI. Windows loads and says its getting things ready. Then the machine reboots and I am stuck in a loop of waiting for windows to load and then suddenly another reboot. No BSOD. Just loads until it reboots for some reason. 

When I try creating a WTG USB out of a genuine Windows 11 iso file, I again can boot into the usb, wait for windows to load, then a BSOD indicating the boot EFI partition cannot be found. 

There must be a way to get windows (to go); I really don't want to re-partition what I've got and waste space on windows when I have a usable spare laptop hard drive.

Hope this doesn't come off as rude, have a nice day.

Best,
will

Log
---
```
[rufus.log](https://github.com/pbatard/rufus/files/9730842/rufus.log)
```

pbatard, 2022-10-07:
> There must be a way to get windows (to go);

Maybe. But you will have to ask Microsoft, because what you're describing is not a Rufus issue and I can't help you with this, as I don't supersede Microsoft when it comes to providing support for pure Windows issues.

FYI, the way Rufus creates a Windows To Go drive is the same as the one that's described at https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-8.1-and-8/jj721578(v=ws.11) (See _Windows PowerShell equivalent commands_), and you will find that if you do create your Windows To Go drive using this guide, you will run into the exact same issue.

As such, I'm afraid I have to close this issue as off topic, as your problem looks like a specific environmental issue between your PC and Windows To Go that only Dell, Microsoft or other people using the same hardware will be able to answer.

closed 2022-10-07 by pbatard

willrabbermann, 2022-10-07:
Is there any other predefined method to getting a working windows 10 or 11 installation working on a USB device? 

Even when I tried installing to the sata drive from microsoft install media, windows deliberately blocks installing an OS to USB port drives. Its mildly infuriating but this is windows. Oh well it seems windows to go isnt really a rufus issue.

willrabbermann, 2022-10-07:
So for anyone else sharing my particular issue with rufus, I would recommend getting WinToUSB. 

Im using the enterprise edition of WinToUSB and it made a working Windows 11 To Go usb drive out of my drive. No BSOD or issues. First try too, I hope rufus' WTG gets troubleshot more...

willrabbermann, 2022-10-07:
Maybe this is a sign you shouldn't use the **official** WTG?

pbatard, 2022-10-07:
A single failure report for Windows To Go in years?

I count this as a pretty clear indication that it's not really worth wasting my time on this, when I can spend it on matters that can help a lot more people. By all accounts, your specific issue is something that the vast majority of people are unlikely to run into, and I will therefore treat it as such, instead of dropping everything else I am doing, and investing weeks to:
1. Reverse engineer what __closed source__ software is doing
2. Evaluate whether deviating from the __official__ method of creating WTG drive will actually be beneficial
3. Altering the Rufus code
4. Validating the new changes

Again, this is Open Source software, so if you feel that this is important, I'll happily accept the results of your investigation along with a patch that goes with it. But please don't expect __me__ to sink my time into what looks like a very limited and purely environmental issue, especially as you have demonstrated that there exists a workaround (use a non official method of creating a WTG drive) for people who are somehow encountering it.

willrabbermann, 2022-10-07:
enjoy your open source soyware. clearly you cant waste time on it

pbatard, 2022-10-07:
https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want

willrabbermann, 2022-10-18:
i deleted my last comment. there is no need to be so negative about open source software. its not your fault.