pbatard/rufus #2060: Error during "make"

dlbendigo, 2022-10-07:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
"./configure" runs OK.  During "make" I get: "-Werror-implicit-function-declaration -Wbidi-chars=none -Wno-pointer-sign" not allowed on command line
Log
---
```
<FULL LOG>
```
[config.log](https://github.com/pbatard/rufus/files/9731742/config.log)

pbatard, 2022-10-07:
```
uname -s = Linux
```

You're trying to compile Rufus on Linux, as a Linux application.

Rufus is __Windows-only__, so this will not work.

Oh, and before you ask, please [read this entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_port_Rufus_to_LinuxMac_OSSome_other_OS).

closed 2022-10-07 by pbatard

dlbendigo, 2022-10-09:
Sorry, just about anything at SourceForge will compile on an open
source system, although the page recommending Rufus suggested an app
I hadn't heard of to use with Linux.

Years ago, I asked a developer how to compile his app.  He replied:
"You don't understand Open Source.  I give you the code, but
getting it to work is none of my business."  If so, why does code
include an "INSTALL" file, and put any differences in the README file? I
replied to him that if he wasn't willing to support his app, there
must be a problem with it. He had paying customers, who would no
doubt be supported.  But this situation is entirely different.

Regards,

Doug.

On Fri, 07 Oct 2022 04:08:09 -0700 Pete Batard
***@***.***> wrote:

> ```
> uname -s = Linux
> ```
>
> You're trying to compile Rufus on Linux, as a Linux application.
>
> Rufus is __Windows-only__, so this will not work.
>
> Oh, and before you ask, please [read this entry from [...]

pbatard, 2022-10-19:
https://github.com/pbatard/rufus#compilation