pbatard/rufus #2062: Error 17: No Mapping

tasrface22, 2022-10-11:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
After I make a bootable Windows 10 USB using UEFI, and attempt to boot from USB, I always get the error code 17: no mapping, no matter how many times I erase the USB drive and rewrite it with Rufus. What can I do? 

Thanks,
Jake

Log
---
```
<FULL LOG>
```

pbatard, 2022-10-11:
Unless you want your issue to be closed without being looked at, I __STILL__ need to see the log of how you created the drive with Rufus.

Also, can you provide the whole output from UEFI:NTFS with the error you get? A screen capture (with your phone or a camera), attached as an image is fine, as long as the text can be read.

Without these 2 elements, you simply don't provide enough information for your issue to be looked at.

tasrface22, 2022-10-11:
OK, give me a little bit to gather that information for you.

tasrface22, 2022-10-11:
Here is the screenshot of the error...

tasrface22, 2022-10-11:
Here is the full log of when I created the USB drive:

Rufus x86 v3.20.1929 (Portable)
Windows version: Windows 11 Home, 64-bit (Build 22621.608)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'PNY USB 3.2.1 FD USB Device' (154B:1008)
Found USB 2.0 device 'Generic- SD/MMC/MS PRO USB Device' (0BDA:0177)
2 devices found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3766, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {4C1916FB-3F9B-43C3-B9A7-56C6DF813A2D}
Max parts: 128, Start Offset: 17408, Usable = 30979095552 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {3B3BD8C6-773E-43E2-AE90-357D76CF11CE}
  Size: 28.8 GB (30976985088 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {D3049814-9594-496F-A634-AFA23B297DD1}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60503972, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 10 ISO (Build 19041.1)
  Size: 5.5 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Windows10_InsiderPreview_Client_x64_en-us_19045.1826.iso (5.5
GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 28.8 GB)
● Creating UEFI:NTFS Partition (offset: 30978033664, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for shared write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
[... 1448 lines cut ...]
Extracting: E:\sources\replacementmanifests\winsat-cmi-replacement.man (889
bytes)
Extracting: E:\sources\replacementmanifests\wlidsvc.vista-win7.man (4.1 KB)
Extracting: E:\sources\replacementmanifests\wmdmigration-replacement.man (4
KB)
Extracting: E:\sources\replacementmanifests\wmi-core-replacement.man (2.7
KB)
Extracting: E:\sources\replacementmanifests\wmiacpi-replacement.man (1.1 KB)
Extracting:
E:\sources\replacementmanifests\wmpnsservice-migration-replacement.man (1.5
KB)
Extracting:
E:\sources\replacementmanifests\wmpplayer-migration-replacement.man (2.8 KB)
Extracting: E:\sources\replacementmanifests\wpc\wpcmigration.downlevel.dll
(208 KB)
Extracting: E:\sources\replacementmanifests\wpc\wpcmigration.uplevel.dll
(231.5 KB)
Extracting:
E:\sources\replacementmanifests\wsinfra-other-mw-replacement.man (1.1 KB)
Extracting: E:\sources\replacementmanifests\wsinfra-upgrade-replacement.man
(1.5 KB)
Extracting: E:\sources\replacementmanifests\wwansvc-servercfg-repl.man (1
KB)
Extracting: E:\sources\reportgen.dll (52.3 KB)
Extracting: E:\s

tasrface22, 2022-10-11:
<FULL LOG> 
Rufus x86 v3.20.1929 (Portable)
Windows version: Windows 11 Home, 64-bit (Build 22621.608)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'PNY USB 3.2.1 FD USB Device' (154B:1008)
Found USB 2.0 device 'Generic- SD/MMC/MS PRO USB Device' (0BDA:0177)
2 devices found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3766, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {4C1916FB-3F9B-43C3-B9A7-56C6DF813A2D}
Max parts: 128, Start Offset: 17408, Usable = 30979095552 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {3B3BD8C6-773E-43E2-AE90-357D76CF11CE}
  Size: 28.8 GB (30976985088 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {D3049814-9594-496F-A634-AFA23B297DD1}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60503972, Attributes: 0x9000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Detected: Windows 10 ISO (Build 19041.1)
  Size: 5.5 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Windows10_InsiderPreview_Client_x64_en-us_19045.1826.iso (5.5 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 28.8 GB)
● Creating UEFI:NTFS Partition (offset: 30978033664, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for shared write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{d72a3860-ae99-45bd-bffb-c413d084a164}\
Disabling file indexing...
[... 952 lines cut ...]
NTFS Fixup completed.

Found USB 3.0 device 'PNY USB 3.2.1 FD USB Device' (154B:1008)
Found USB 2.0 device 'Generic- SD/MMC/MS PRO USB Device' (0BDA:0177)
2 devices found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3766, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {647CC0AB-344C-4620-A751-0563F93A14A1}
Max parts: 128, Start Offset: 17408, Usable = 30979095552 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {D72A3860-AE99-45BD-BFFB-C413D084A164}
  Size: 28.8 GB (30976985088 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {9E68ED06-F9B7-48F2-80DC-0D9CEAA6996E}
  Size: 1 MB (1048576 bytes)
  Start Sector: 60503972, Attributes: 0x9000000000000000
[image]

pbatard, 2022-10-11:
Thanks. There's something super strange about the output you get on the computer you are trying to boot, because it definitely like the output from UEFI:NTFS, __but__ none of the informational messages (such as the ones that would provide information about your hardware) are displaying.

I actually don't get how that is possible, because the only difference between error and informational messages are that we change the colour of the prefix and that is really it (there aren't different ways of outputting error text and normal text in UEFI, it's all done the same). So I am already leaning towards a __very__ nonstandard UEFI firmware on that machine of yours, which would also explain why you are getting an error.

Can you please tell the exact make and model (and possibly BIOS version) of the computer you are trying to install Windows on?
My guess is that it's one of these UEFI firmwares that provides a half baked NTFS driver, that prevents the NTFS driver from UEFI:NTFS from being installed, and that also prevents the loading of executables such as the Windows bootloaders.

In short, in all likelihood, you're going to have to contact your hardware/BIOS manufacturer and tell them that they need to sort out their UEFI firmware, because if UEFI:NTFS fails to execute the Windows UEFI bootloader, they probably screwed up something there...

tasrface22, 2022-10-11:
I have a Dell Inspiron 13-7370 with BIOS version 1.8.0. I had to downgrade the BIOS from version 1.25.0, because I thought that would fix the issue, but it didn't. Plus, I recorded me booting from the USB using UEFI, and it briefly showed the information message before popping up with the error. I don't think I'm able to upload it here because I'm not sure what the file size limit is when uploading files on GitHub, and I cannot upload the video via Gmail because the file size limit is 25 MB.

tasrface22, 2022-10-11:
Here is the full video of me trying to boot from USB using UEFI

pbatard, 2022-10-11:
Can you transcribe the messages you get from the video?
It should be about 10 lines of text, so hopefully, it shouldn't be too painful to transcribe I hope.

> Dell Inspiron 13-7370

Of course it's a Dell. I'm afraid that's not the first user report I've seen of UEFI:NTFS not working with a Dell UEFI firmware, due to the half-baked NTFS driver they use (which we can't work around as it's not possible to replace a native UEFI driver).

tasrface22, 2022-10-11:
Hang on a second...

pbatard, 2022-10-11:
Okay, then there's no need to post the video, since I won't be able to see more than what you described... 

But really, that error looks like a pure Dell firmware issue, and is not something I can work around, especially if I can't replicate it... I would advise you to contact Dell to see if they are aware that their UEFI firmware may have some issues with booting from an NTFS partition (which a __UEFI-compliant__ firmware with a __UEFI-compliant__ NTFS driver should always be able to do).

tasrface22, 2022-10-11:
I'm currently trying to update the BIOS to the latest version and see what happens

tasrface22, 2022-10-11:
> Okay, then there's no need to post the video, since I won't be able to see more than what you described...
>
> But really, that error looks like a pure Dell firmware issue, and is not something I can [...]

I'll be able to contact them later and see what they are able to do.

pbatard, 2022-10-18:
Since there is not much I can do from this side, I'm going to close this issue. But if you get anywhere with Dell, feel free to report here.

closed 2022-10-18 by pbatard

tasrface22, 2022-10-18:
I have contacted Dell regarding this issue, and they can't help me because
my laptop is out of warranty.