pbatard/rufus #2067: Unused variable in uprintf() call

marcosfrm, 2022-10-15:
https://github.com/pbatard/rufus/blob/b2bf29f65263f973d7511212a8b732b019cb98cb/src/wue.c#L784

Unused `wim_index`.

`WIMUnmountImage()` will print WIM index anyway.

pbatard, 2022-10-15:
That's not unused. Unused would be if we set that variable and didn't pass it to any call, which is very much __not__ the case here.

Also, please bear in mind that `WIMUnmountImage()` may fail, which is the precise reason why we want to to print the index, so that the user (or the developer of the app ;)) gets an idea about what WIM index caused the issue during unmount.

I could also elaborate on why we see it better to print the index in the message we output before we enter a call that might fail, rather than with each error message in the call that fails...

&rarr; Rejected as "not an actual unused variable" (but please don't let that discourage you from looking at the code, coz there are most likely things in there that I overlooked and that could definitely be improved).

marcosfrm, 2022-10-15:
Unused in the `uprintf()` call: there's no `%d` specifier...

pbatard, 2022-10-15:
Oh, I missed that - you're absolutely right. I'll fix it!

closed 2022-10-15 by pbatard

pbatard, 2022-10-15:
Stealth actual commit that closes this is cee21b1981c8c621fae33892c16fac25e748a0c3 because mistyped in your name in the part where I thanked you in the previous one (D'OH!).

commit ff69ab11 (2022-10-15): [wue] prevent 'Administrator' and 'Guest' from being set as username

* These are reserved usernames that are created by default, so we should not use them.
* Also fix missing format specifier in ApplyWindowsCustomization() and make sure we
  print wim_index for both mount and unmount.
* Closes #2067 (with thanks to marcosfm)
commit cee21b19 (2022-10-15): [wue] prevent 'Administrator' and 'Guest' from being set as username

* These are reserved usernames that are created by default, so we should not use them.
* Also fix missing format specifier in ApplyWindowsCustomization() and make sure we
  print wim_index for both mount and unmount.
* Closes #2067 (with thanks to marcosfrm)