pbatard/rufus #2070: Getting: Device Number for device is too big (74) - ignoring device

ozzie2014, 2022-10-18:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
Rufus says Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device and wont find SD card with USB reader.
Log
---
```
Rufus x86 v3.20.1929
Windows version: Windows 10 Home, 64-bit (Build 19044.2130)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 10 ISO (Build 19041.1)
  Size: 4.5 GB (Projected)
  Note: File on disk is larger than reported ISO size by 548 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.5 GB)
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 10 ISO (Build 19041.1)
  Size: 4.5 GB (Projected)
  Note: File on disk is larger than reported ISO size by 548 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.5 GB)
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found
Found USB 3.0 device 'Generic- SD/MMC USB Device' (0BDA:0316)
Device Number for device \\?\usbstor#disk&ven_generic-&prod_sd#mmc&rev_1.00#20120501030900000&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b} is too big (74) - ignoring device
A device was eliminated because it didn't report itself as a disk
0 devices found

```

pbatard, 2022-10-18:
Looks like a duplicate of #2029. Do you have the Xbox App installed? What happens if you reboot your computer and make sure the Xbox App isn't launched?

ozzie2014, 2022-10-18:
> Looks like a duplicate of #2029. Do you have the Xbox App installed? What happens if you reboot your computer and make sure the Xbox App isn't launched?

I swear I searched the FAQ and the other questions for this and didn't find it, sorry for the dupe

closed 2022-10-18 by ozzie2014