pbatard/rufus #2074: NTFS suggested by default (requiring UEFI:NTFS) when FAT32 is possible for UEFI target

marcosfrm, 2022-10-25:
This is a corner case that **only** happens with dual BIOS/UEFI mode enabled.

Dual BIOS/UEFI mode disabled + Windows ISO (UEFI capable) without >4GB file:

|Partition scheme | Target system | File system |
|:--- |:---                |:---                       |
| GPT | UEFI (non CSM)     | FAT32 selected by default |
| MBR | BIOS (or UEFI-CSM) | NTFS selected by default  |

Enabling dual BIOS/UEFI mode and not closing Rufus:

|Partition scheme | Target system | File system |
|:--- |:---                |:---                          |
| GPT | UEFI (non CSM)     | FAT32 selected by default    |
| MBR | BIOS or UEFI       | FAT32 selected by default[1] |

[1] Only if GPT is selected when Alt-E is pressed; if MBR is already selected it will stay as NTFS.

Reopening Rufus ("EnableWindowsDualUefiBiosMode"=dword:00000001):

|Partition scheme | Target system | File system |
|:--- |:---                |:---                       |
| GPT | UEFI (non CSM)     | NTFS selected by default  |
| MBR | BIOS or UEFI       | NTFS selected by default  |

Disabling dual BIOS/UEFI mode and not closing Rufus:

|Partition scheme | Target system | File system |
|:--- |:---                |:---                       |
| GPT | UEFI (non CSM)     | NTFS selected by default  |
| MBR | BIOS (or UEFI-CSM) | NTFS selected by default  |

Minor inconsistencies. Not a big deal.

pbatard, 2022-10-25:
Thanks for the report.

I think this may be linked to #2000 and its following entry:
- [ ] There are some inconsistencies with how we select the default FS. For instance, `tiny10 21H2 x64 beta 1.iso` may pick FAT32 at first but once the media is created, NTFS is selected.

That's something I am planning to look into for 3.21, but I can't make any promises at this stage...

pbatard, 2022-11-16:
This should now be fixed in the commit that closed this issue.

Note however that we may still have default file system selection issues when plugging a new device, as the current code considers that new device arrival, and the file system detected for the new device, may override the current file system selection, and our current logic considers that the current device is always a new device when the device list is refreshed (see the current last entry of #2080), so you can still trigger a file system override if you play with plugging/unplugging devices after the ISO has been scanned. This is something I __may__ try to address in 3.22, but for now, I believe the issue you mentioned should be fixed.

Thanks again for reporting this issue, as it led me to identify another issue, which I fixed (or rather worked around) in a6769e4fb262b590a2550a99ac52c0b1c672a8d5.

commit 9727fd43 (2022-11-16): [core] improve/fix file system selection according to current ISO

* Closes #2074.
* Also harmonize whitespaces.
closed 2022-11-16 by pbatard