pbatard/rufus #2075: Windows 10 To Go no longer works

neuhaus, 2022-10-25:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [-] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [-] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [-] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [-] I also tried one or more of the following:
  - [-] Using a different USB drive.
  - [-] Plugging the USB into a different port.
  - [-] Running Rufus on a different computer.
- [-] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I created a UEFI Windows 10 Pro To Go drive on an external SSD in an USB 3.1 enclosure with Rufus and it worked for a week.
Today when I returned to the PC it booted into the legacy Rufus screen and i can no longer boot the USB SSD in UEFI mode.

I wanted to look at the SSD on another PC but was surprised that i don't see an NTFS partition. Is it normal when creating a Windows 10 To Go with Rufus?

Is there a way to repair the SSD drive and make it bootable (UEFI) again? Or do i have to start over? What can cause this issue?

Mainboard: Gigabyte GA-AB350M-D3V with the latest BIOS Version F51g. I have tried disabling CSM in the BIOS but then i don't see the USB SSD at all among the boot options.

Thanks!

Log
---
```
$ LANG=C sudo fdisk -l /dev/sdc
Disk /dev/sdc: 3.64 TiB, 4000787030016 bytes, 7814037168 sectors
Disk model: SSD Enclosure   
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
Disklabel type: gpt
Disk identifier: x-x-x-x

Device      Start        End    Sectors  Size Type
/dev/sdc1    2048     534527     532480  260M EFI System
/dev/sdc2  534528     796671     262144  128M Microsoft reserved
/dev/sdc3  796672 7814037134 7813240463  3.6T Microsoft basic data

$ LANG=C sudo parted /dev/sdc
GNU Parted 3.4
Using /dev/sdc
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) print                                                            
Model: Kolink SSD Enclosure (scsi)
Disk /dev/sdc: 4001GB
Sector size (logical/physical): 512B/4096B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name                          Flags
 1      1049kB  274MB   273MB   fat32        EFI System Partition          boot, esp
 2      274MB   408MB   134MB                Microsoft Reserved Partition  msftres
 3      408MB   4001GB  4000GB               Main Data Partition           msftdata

```

pbatard, 2022-10-25:
> Today when I returned to the PC it booted into the legacy Rufus screen and i can no longer boot the USB SSD in UEFI mode.

That's either a problem with your UEFI settings (you enabled Legacy boot to have precedence over UEFI boot) or you screwed up your ESP on the SSD (which sounds unlikely since Windows 10 doesn't really allow you to do that). Most likely, your UEFI firmware reverted to prioritizing Legacy boot, and you will have to figure out how to re-enable UEFI boot over BIOS/Legacy. I'm afraid however that, since every UEFI firmware is different, I can not provide any assistance on how to do that. It is up to __YOU__ to check the manufacturer documentation and figure that out.

> was surprised that i don't see an NTFS partition

Main Data Partition __IS__ the NTFS partition. Not sure why parted is unable to tell you that there is an NTFS file system there, but I can guarantee that it should tell you that it's there. At any rate, this is not the cause of your issue, since if it was, you would still be able to boot in UEFI mode and wouldn't get the BIOS/Legacy boot message.

neuhaus, 2022-10-25:
Here is the gdisk output with the partition codes:
```
Disk /dev/sdc: 7814037168 sectors, 3.6 TiB
Model: SSD Enclosure   
Sector size (logical/physical): 512/4096 bytes
Disk identifier (GUID): x-x-x-x
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 7814037134
Partitions will be aligned on 2048-sector boundaries
Total free space is 2014 sectors (1007.0 KiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048          534527   260.0 MiB   EF00  EFI System Partition
   2          534528          796671   128.0 MiB   0C01  Microsoft Reserved ...
   3          796672      7814037134   3.6 TiB     0700  Main Data Partition
```

neuhaus, 2022-10-25:
OK so i am sure it was booting in UEFI mode when it was working and now i can no longer boot it in UEFI mode and i have looked at the BIOS and tried every setting that could help.

Now, perhaps you can ask this question before i give up:
Is it normal that when i connect my Windows 10 To Go USB SSD created with Rufus to another Windows 10 PC that i cannot access the concent of the drive? Because that's what i am seeing. Thanks again.

pbatard, 2022-10-25:
The codes are fine. A UEFI partition code is __NOT__ meant provide any indication as to the file system, so it is 100% expected that you will not see anything mentioning NTFS. But if you try to mount the NTFS partition on a system that has NTFS support, I'm pretty sure you will see that your data is still there.

> and now i can no longer boot it in UEFI mode

There's your issue.

> Is it normal that when i connect my Windows 10 To Go USB SSD created with Rufus to another Windows 10 PC that i cannot access the concent of the drive?

Make sure Windows assigns a letter to the drive, because it may not always do that on its own.
But, yes, unless you really screwed up your disk, Windows 10 should be able to access the NTFS partition from your Windows To Go drive.

Also, please understand that your issue has nothing to do with Rufus, as your system did work, and then you did something, __that did not involve Rufus__ to break it. So, I'm afraid this is as far as I will get to help you, and I will close this issue now.

closed 2022-10-25 by pbatard

neuhaus, 2022-10-25:
OK i realize that it's probably not a Rufus issue but thanks for your input.
I connected the SSD to another PC and that one is also not able to boot it in UEFI mode. I guess there is an issue with the EFI partition and it got corrupted?

pbatard, 2022-10-25:
I'm afraid I cannot guess what happened, but it does look like your drive got corrupted.

neuhaus, 2022-10-25:
I noticed that the EFI partition is completely empty, is there a way to restore it with Rufus?

pbatard, 2022-10-25:
Not with Rufus no.