pbatard/rufus #2076: rufus not detecting USB

Whooshyy, 2022-10-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I am trying to make a bootable usb for my laptop, but rufus is not detecting my usb and is showing this error "A device was eliminated because it didn't report itself as a disk">

Log
---
```
<Rufus x86 v3.20.1929
Windows version: Windows 10 Home Single Language, 64-bit (Build 19044.2130)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
Could not open '\\?\usbstor#disk&ven_generic&prod_flash_disk&rev_8.07#7e2f1029&0#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}': [0x00000005] Access is denied.
A device was eliminated because it didn't report itself as a disk
0 devices found
>
```

JonnyTech, 2022-10-28:
https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied

pbatard, 2022-10-28:
You may also try to turn on enumeration debug in Rufus with <kbd>Alt</kbd>-<kbd>.</kbd>, as it may provide more info about your issue in the log, but, yes, if you are seeing `[0x00000005] Access is denied` right after your device is listed, it simply means that something __external to Rufus__ is actively preventing access to the drive, and there is literally nothing I can do in Rufus to fix your issue, so it is up to you to identify that external something and disable it (starting with making sure that you don't use any of the applications listed [here](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software).

pbatard, 2022-11-02:
No update &rarr; Closed.

closed 2022-11-02 by pbatard