pbatard/rufus #2083: Target volume label on formatting should be kept

powerbroker, 2022-11-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Preserve target volume label when formatting. Avoid forcing user to do extra steps such as manually returning it back.

Log
---
```
Rufus x86 v3.20.1929 (Portable)
Windows version: Windows 11 (Unknown Edition 0xBF), 64-bit (Build 22000.1098)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'JetFlash Transcend 32GB USB Device' (8564:1000)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3801, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {BFFB9266-AEA2-4853-AADD-7C8B8C22AEBA}
Max parts: 128, Start Offset: 17408, Usable = 31264306688 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {159B6386-3AAF-4C5E-A215-315642B3B0E5}
  Size: 29.1 GB (31263275520 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCSA_X64FRE_EN-US_DV5'
  Detected: Windows 11 ISO (Build 22000.1157)
  Size: 4.9 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 22000.1219.221105-0007.CO_RELEASE_SVC_PROD1_CLIENTMULTI_X64FRE_EN-US.ISO (4.9 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive7 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 29.1 GB)
● Creating UEFI:NTFS Partition (offset: 31263241216, size: 1 MB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive7 for shared write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{c3941621-f62a-4a95-a11b-c1bb9226ff4c}\
Disabling file indexing...
Successfully remounted \\?\Volume{c3941621-f62a-4a95-a11b-c1bb9226ff4c}\ as F:
Extracting files...

Comment is too long (maximum is 65536 characters)

There was an error creating your Issue: body is too long (maximum is 65536 characters). 

```

pbatard, 2022-11-10:
I disagree. In the vast majority of cases, people want the label to reflect the ISO that was used to create the media, so that they aren't left wondering what it is they wrote last.

Now, if you are annoyed with Microsoft's way of labelling their ISO, you should complain to Microsoft directly.

closed 2022-11-10 by pbatard

powerbroker, 2022-11-16:
> I disagree. In the vast majority of cases, people want the label to reflect the ISO that was used to create the media, so that they aren't left wondering what it is they wrote last.
> 
> Now, if you are annoyed with Microsoft's way of labelling their ISO, you should complain to Microsoft directly.

why not to offer all the options in editable combobox with pre-selected ISO label?

pbatard, 2022-11-16:
> why not to offer all the options in editable combobox with pre-selected ISO label?

Cost/benefit. Not going to be helpful to as many users as you believe it will. Plus the fact that editable comboboxes, besides being confusing for users (who expect comboboxes to be read-only) are something that are at the very least "tricky" to implement in pure win32 is not going to do any favours for the feature there.

Please stop being under the impression that the developer of an application is sitting idly, with nothing better to do than wait for suggestions to implement so that they can pass the time, and that UI changes like the one you want to see "would take 5 minutes top to add". The developer of an application like Rufus has actually __way__ too many features and improvements in their backlogs (and too many other projects besides Rufus to compete for their time) not to consider the cost/benefit ratio of a feature that is only likely to help a handful of users (because most people don't care about the lable of their bootable media which is intended for immediate one time use) and that they know, from experience with trying to add modern UI elements in pure win32, is likely to take at minimum 2 days of full time work.

Plus you haven't considered how to avoid user confusion with your suggestion. You can't just drop a combobox and be on your way. You have to add tooltips to describe why the user is suddenly being prompted with multiple choice, because some are going to be confused about a label that has multiple choices, which means I now need to handle 37 translations, on top of the other development time, just so that a handful of users, who are merely _irked_ with regards to something that should logically be construed as barely an inconvenience, get somewhat less irked...