pbatard/rufus #2086: Arch Linux 2022.11.01 fails to boot if label is changed, despite patching

Leo40Git, 2022-11-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Changing the label from the default "ARCH_202211" value to something else (I used "Arch Linux 2022.11.01") seems to cause the boot loader to be unable to load the kernel - it cannot find "ARCH_202211".
This is despite the log claiming that references to this label have been patched (which is why I am opening an issue here rather than on Arch Linux - since it seems like a Rufus feature intended to avoid this exact scenario isn't working).

Log
---
```
Rufus x86 v3.20.1929
Windows version: Windows 11 Home, 64-bit (Build 25236.1010)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Using 'autorun.inf' label for drive F: 'Arch Linux 2022.11.01'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000E9287
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 6.6 GB (7040130048 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 8.0 GB (8589934080 bytes)
  Start Sector: 13752302, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre2-11-gbf6db5b4 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'ARCH_202211'
  Size: 766.4 MB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: archlinux-2022.11.01-x86_64.iso (782.3 MB)

Computing checksum for 'C:\Users\Leo\Documents\Linux\archlinux-2022.11.01-x86_64.iso'...
  MD5:    c908f303e24208691ca8cd4424c998e0
  SHA1:   f0173119e849405982d156f76190f29ba95caa09
  SHA256: df6749df55b02cec98e5a9177c7957acfb96fe14d04553b6e4714100a4824f68
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Leo\AppData\Local\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\' for Syslinux installation
[... 142 lines cut ...]
Extracting: F:\syslinux\zzjson.c32 (3.5 KB)
Created: F:\syslinux.cfg → /syslinux/syslinux.cfg
Finalizing, please wait...
Created: F:autorun.inf
Created: F:autorun.ico

Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Using 'autorun.inf' label for drive F: 'Arch Linux 2022.11.01'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x000E9287
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 6.6 GB (7040130048 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 8.0 GB (8589934080 bytes)
  Start Sector: 13752302, Boot: No
```

pbatard, 2022-11-10:
That's probably because you used a space in there. I'll test this, but what happens if you use underscore instead of spaces in your label?

Leo40Git, 2022-11-10:
Will try using underscores instead of spaces shortly.

pbatard, 2022-11-10:
For reference, the problem is that there are a mainstream distros out there (Red Hat, CentOS...) that require spaces in the FAT32 label to be converted to `\x20` sequences, in order to work properly. So we fixed that in [55a5eb07e72d13fa43f057f7efa13818bbe418a8]. But I guess Arch on the other hand does require actual spaces...

If that is the issue, this is going to be __very difficult__ to fix, because we cannot "guess" what type of distro may require spaces and what type of distro may require `\x20` sequences, and trying to maintain a list of "` ` vs `\x20`  distros" would both be a major time sink and be sure to break in the long run as there's a new distro pretty much every other day.

Which means that, as long as it works with the default label proposed by Rufus, or when users don't try to change the label to something that includes spaces, I'm probably not going to do anything about this.

But again, I'll test this when I have a chance to see if this theory is correct...

Leo40Git, 2022-11-10:
Now trying "Arch_Linux_2022.11.01", here's the Rufus log.
Will respond in a bit if boot succeeded...
```
Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Opened \\.\PhysicalDrive1 for shared write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 6.6 GB)
● Creating Linux Persistence Partition (offset: 7041178624, size: 8.0 GB)
Waiting for logical drive to reappear...
Using Debian-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
262143 possible inodes out of 2097151 blocks (block size = 4096)
104857 blocks (5.0%) reserved for the super user
Creating 64 inode sets: [1 marker = 1.0 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 8192 journal blocks: [1 marker = 102.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Created 'persistence.conf' file
Formatting to FAT32 (using IFS)
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for shared write access
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{aea6e976-6124-11ed-bdd3-8c16456ccfcb}\
Installing Syslinux 6.04...
Opened \\?\Volume{aea6e976-6124-11ed-bdd3-8c16456ccfcb} for shared write access
Using existing 'C:\Users\Leo\AppData\Local\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\ldlinux.sys' ✓
Using existing 'C:\Users\Leo\AppData\Local\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted \\?\Volume{aea6e976-6124-11ed-bdd3-8c16456ccfcb}\ as D:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: D:\EFI\BOOT\BOOTIA32.EFI (5.3 MB)
Extracting: D:\EFI\BOOT\BOOTx64.EFI (6.4 MB)
Extracting: D:\EFI\BOOT\grub.cfg (2.2 KB)
  Patched D:\EFI\BOOT\grub.cfg: 'ARCH_202211' ➔ 'ARCH_LINUX_'
Extracting: D:\arch\boot\amd-ucode.img (50 KB)
Extracting: D:\arch\boot\intel-ucode.img (4.9 MB)
Extracting: D:\arch\boot\licenses\amd-ucode\LICENSE.amd-ucode (3.7 KB)
Extracting: D:\arch\boot\licenses\intel-ucode\LICENSE (1.6 KB)
Extracting: D:\arch\boot\licenses\memtest86+\license.txt (17.7 KB)
Extracting: D:\arch\boot\memtest (141.0 KB)
Extracting: D:\arch\boot\x86_64\initramfs-linux.img (64.4 MB)
Extracting: D:\arch\boot\x86_64\vmlinuz-linux (10.6 MB)
Extracting: D:\arch\grubenv (1 KB)
Extracting: D:\arch\pkglist.x86_64.txt (7 KB)
Extracting: D:\arch\version (11 bytes)
Extracting: D:\arch\x86_64\airootfs.sfs (671 MB)
Extracting: D:\arch\x86_64\airootfs.sfs.cms.sig (698 bytes)
Extracting: D:\arch\x86_64\airootfs.sfs.sig (471 bytes)
Extracting: D:\arch\x86_64\airootfs.sha512 (143 bytes)
Extracting: D:\shellia32.efi (908.6 KB)
Extracting: D:\shellx64.efi (1010.3 KB)
Extracting: D:\syslinux\archiso_head.cfg (809 bytes)
Extracting: D:\syslinux\archiso_pxe-linux.cfg (1.3 KB)
[... 74 lines cut ...]
Extracting: D:\syslinux\zzjson.c32 (3.5 KB)
Created: D:\syslinux.cfg → /syslinux/syslinux.cfg
Finalizing, please wait...
Created: D:autorun.inf
Created: D:autorun.ico

Found USB 2.0 device 'SanDisk Cruzer Glide USB Device' (0781:5575)
Using 'autorun.inf' label for drive D: 'Arch_Linux_2022.11.01'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1900, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0022A3DB
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 6.6 GB (7040130048 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Detected File System: ext3
  Size: 8.0 GB (8589934080 bytes)
  Start Sector: 13752302, Boot: No
```

Leo40Git, 2022-11-10:
Does NOT work!
Printed message is:
```
error: no such device: ARCH_202211.
error: file `/arch/boot/x86_64/vmlinuz-linux' not found. 
error: you need to load the kernel first. 
```

pbatard, 2022-11-10:
But it works when you don't change the label right?

I'm just going to point out that there's actually a good reason that Arch use a uppercase label with no spaces and less than 11 chars for their ISOs, so that it is compatible with the limitation of FAT32 and that while I can understand why one would want to change the label, the root of the problem is that there are too many Linux distros out there that use a search by label for the boot media (AFAIK the only ones that evolved from this are the good people of Debian, who realised that this was way too brittle a solution and are looking for a specific file on all the partitions that can be accessed, which is the __MUCH SMARTER__ way to do it), and this is what leads users into exactly the kind of issue you are facing.

Plus it breaks UEFI file system transposition (the ability that UEFI provides of simply copy the content of a boot media like an ISO at the file system level and still end up with a bootable media), which is another problem...

At any rate, I will test this issue when I get a chance, but, yeah, I'm actually not surprise that things do break, as they have repeatedly done in the past, when distros rely on labels to identify the boot media...

marcosfrm, 2022-11-10:
Arch media uses by default `/dev/disk/by-label/...` to mount the device. Hence `\x20` is not the issue probably.

Leo40Git, 2022-11-10:
Now attempting just leaving the stupid label option alone, and editing the `autorun.inf` file manually.

pbatard, 2022-11-10:
> Arch media uses by default /dev/disk/by-label/... to mount the device. Hence \x20 is not the issue probably.

Yeah, there's also a good chance that they have hardcoded a search for `ARCH_202211` in their init script, in which case the minute you change the label it's game over (and I'm obviously not going to attempt to fix the init script in Rufus)...

Another reason why looking for a boot media by label is a __VERY STUPID IDEA__ and why the Debian folks have moved away from it.

Leo40Git, 2022-11-10:
Yep, not changing the label works.

D3vil0p3r, 2022-11-13:
Hello guys, I have the same issue on another OS (Athena OS). The name of the ISO is `athena-2022.11.13-x86_64.iso` and I'm on Windows
[image]

At the end of the process, Rufus assigns the label "ATHENA-2022" to the USB device instead of the label shown in "Volume label" field of Rufus. 

If I remove and replug the USB to Windows, it sees my USB device in "My Computer" as "athena-2022.11.13" and not more as "ATHENA-2022" but "athena-2022.11.13" is a "fake" label, because if I open Disk Management of Windows, the USB device is seen as "ATHENA-2022".

When I reboot the PC for booting Athena OS system, the grub is not able to find the device (as explained by @Leo40Git) because it refers to the "fake" label "athena-2022.11.13" and not the real assigned label "ATHENA-2022".

If I edit the "Volume label as `athena-21234567.11.13`, Rufus will make the label as "ATHENA-2123" so the issue could be related mainly to the length of the label.

It is needed that Rufus must assign as REAL label the one specified in "Volume label" field, that in my case is "athena-2022.11.13", so either force the program to have limited characters in "Volume label" or remove the limited characters (if possible, idk) from the final label name on the USB device.

marcosfrm, 2022-11-13:
> When I reboot the PC for booting Athena OS system, the grub is not able to find the device (as explained by @Leo40Git) because it refers to the "fake" label "athena-2022.11.13" and not the real assigned label "ATHENA-2022".

Rufus should patch the GRUB configuration file, replacing the label. We must find out why it is not working anymore.

pbatard, 2022-11-13:
> At the end of the process, Rufus assigns the label "ATHENA-2022" to the USB device instead of the label shown in "Volume label" field of Rufus.

That's because the FAT specifications do limit labels to 11 uppercase characters with no space.
So Rufus __has no choice__ but to assign a FAT-compliant label to the volume, which `athena-2022.11.13` isn't but `ATHENA-2022` is.

Now, to __hide__ this behaviour and make it look like a non FAT restricted label was assigned, Rufus also creates an `autorun.inf` with the full `athena-2022.11.13` label, which is why, if you use Windows File Explorer (which looks at the `autorun.inf` if one exists), you will see it display `athena-2022.11.13`. But the actual label of the FAT volume, which has to be compliant with the FAT specs, is `ATHENA-2022`

> When I reboot the PC for booting Athena OS system, the grub is not able to find the device (as explained by @Leo40Git) because it refers to the "fake" label "athena-2022.11.13" and not the real assigned label "ATHENA-2022".

Usually, this should be handled by altering the GRUB or Syslinux configuration files, that would have a kernel option that specifies the label that should be looked for to find the boot media. For distros that don't deviate too much from the de-facto norm (i.e. distros that don't place their GRUB/Syslinux config files in weird places), Rufus can do that automatically, and you will see lines such as:
```
  Patched E:\syslinux\some.cfg: 'athena-2022.11.13' ➔ 'ATHENA-2022'
```

Unfortunately, since you did not attach a log, I can't tell what happened in your specific case.

> so the issue could be related mainly to the length of the label.

No, it is related to the FAT specifications, that __PROHIBIT__ the use of a label that is longer than 11 characters or that contains anything but UPPERCASE and a few other characters such as `-`, `_`...

> It is needed that Rufus must assign as REAL label the one specified in "Volume label" field

Again, Rufus CANNOT do that, because FAT labels can only store a label up to 11 characters long, and using uppercase.

You are under the impression we _somehow_ designed Rufus to screw up the label, but we don't go altering labels __unless we don't have any other choice__. That's the reason why Rufus converts the label in the first place and limits it to 11 characters.

Now, if you could provide a full log of how you created the ISO in Rufus, that would help, but it would tell us where the config files are locates and why they might not have been patches by Rufus.

D3vil0p3r, 2022-11-13:
@pbatard in Windows where I can retrieve Rufus logs?

Furthermore, we have this limits due to FAT labels. At this point, cannot you limit also the "Volume label" field at the same 11 chars?

pbatard, 2022-11-13:
> where I can retrieve Rufus logs?

See the check list at the top of this issue.

> At this point, cannot you limit also the "Volume label" field at the same 11 chars?

I won't change a thing, since the label that rufus assigned __is__ the same as you would get with an 11-char limited field. The full field label is __only__ used for `autorun.inf` and has no impact on anything else, so, for all intents and purposes, it's as if the Rufus field was limited to 11 uppercase characters already (we're doing the conversion internally).

I have to stress out that the issue is __not__ with the UI or how Rufus processes the label, but with how we link that processed FAT compliant label with the config scripts that are used to boot the media.

On that note, it's really a shame that Arch derivatives like Athena don't take a __hint__ from what Arch are doing with their ISO labels, and take a minute to wonder why on earth the official Arch maintainers would limit their labels to 11 uppercase characters, which is precisely because they want to make sure that the label is valid for both ISO-9660 and FAT32...

PS: I can only find downloads for `athena-2022.11.04-x86_64.iso`. Can you provide a URL for where you downloaded 2022.11.13?

D3vil0p3r, 2022-11-13:
> On that note, it's really a shame that Arch derivatives like Athena don't take a **hint** from what Arch are doing with their ISO labels, and take a minute to wonder why on earth the official Arch [...]

Don't worry for it, Athena devs can be compliant quickly for labeling in a correct manner within 11 chars and thank you for the suggestion, I was ignoring this limitation.

> PS: I can only find downloads for `athena-2022.11.04-x86_64.iso`. Can you provide a URL for where you downloaded 2022.11.13?

2022.11.13 is still not released because in testing phase. The latest one is `athena-2022.11.04-x86_64.iso`. Meanwhile, I try to get logs.

pbatard, 2022-11-13:
> Athena devs can be compliant quickly for labeling in a correct manner within 11 chars

Sounds good. Hopefully, if the issue (as is my current __guess__ right now) is that they set the Syslinux/GRUB config files into different places or under different names than the ones Arch uses (since Rufus needs to have some idea where it should look in order to patch the kernel option to insert the actual FAT label), they can also revert to not deviating to what Arch does.

Especially, Rufus should produce something similar to what OP reported in their log above:
``` 
Extracting: D:\EFI\BOOT\grub.cfg (2.2 KB)
  Patched D:\EFI\BOOT\grub.cfg: 'ARCH_202211' ➔ 'ARCH_LINUX_'
(...)
Extracting: D:\syslinux\archiso_pxe-linux.cfg (1.3 KB)
  Patched D:\syslinux\archiso_pxe-linux.cfg: 'ARCH_202211' ➔ 'ARCH_LINUX_'
(...)
Extracting: D:\syslinux\archiso_sys-linux.cfg (875 bytes)
  Patched D:\syslinux\archiso_sys-linux.cfg: 'ARCH_202211' ➔ 'ARCH_LINUX_'
```
When dealing with Athena ISOs if you want to have any hope of finding the boot media (which, I am aware per the original issue still doesn't seem to work if you change the label, but which, for most distros, should be enough, even with a user-specified label).

D3vil0p3r, 2022-11-13:
Here the Rufus logs:
```
Rufus x86 v3.20.1929 (Portable)
Windows version: Windows 11 Home, 64-bit (Build 22621.819)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1911, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x7EB17A97
Drive has a Windows 7 Master Boot Record
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre2-11-gbf6db5b4 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'athena-2022.11.13'
  Size: 2.1 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: athena-2022.11.13-x86_64.iso (2.1 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Antonio\Downloads\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.6 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive2 for exclusive write access
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{9df42b13-6382-11ed-a489-48a472bde056}\
Installing Syslinux 6.04...
Opened \\?\Volume{9df42b13-6382-11ed-a489-48a472bde056} for shared write access
Using existing 'C:\Users\Antonio\Downloads\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\ldlinux.sys' ✓
Using existing 'C:\Users\Antonio\Downloads\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\ldlinux.bss' ✓
Successfully wrote 'ldlinux.sys'
Successfully wrote Syslinux boot record
Successfully remounted \\?\Volume{9df42b13-6382-11ed-a489-48a472bde056}\ as E:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: E:\EFI\BOOT\BOOTIA32.EFI (5.3 MB)
[... 90 lines cut ...]
Extracting: E:\syslinux\vesa.c32 (2.9 KB)
Extracting: E:\syslinux\vesainfo.c32 (2.1 KB)
Extracting: E:\syslinux\vesamenu.c32 (26.1 KB)
Extracting: E:\syslinux\vpdtest.c32 (1.8 KB)
Extracting: E:\syslinux\whichsys.c32 (2.4 KB)
Extracting: E:\syslinux\zzjson.c32 (3.5 KB)
Created: E:\syslinux.cfg → /syslinux/syslinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using 'autorun.inf' label for drive E: 'athena-2022.11.13'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1911, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002ABB25
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 14.6 GB (15723659264 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2022-11-13:
Thanks. So Rufus did properly patch the kernel options, but we still weren't able to find the media.

I guess this is indeed the same issue as OP's with recent Arch ISOs, where there must be some remnant option somewhere (not in the config files Rufus patched) that references the original ISO label. Except of course, in OP's case, the issue was brought forward by the user actively altering the label, whereas this time, it's because Rufus has not choice but to alter the label to make it FAT compliant...

I still haven't have a chance to look at it more closely, but I'll do so when I get a chance, coz I'd really like to know where, when the kernel options should be the only elements that specify the label to look for, there might be a residual label lookup in the Arch boot sequence.

Oh, and just so it helps with testing, can you indicate whether you tried to boot in UEFI mode (in which case `E:\EFI\BOOT\grub.cfg` would be used) or in BIOS/Legacy mode (in which case it'll be `E:\syslinux\archiso_sys-linux.cfg`).

D3vil0p3r, 2022-11-13:
> Oh, and just so it helps with testing, can you indicate whether you tried to boot in UEFI mode (in which case E:\EFI\BOOT\grub.cfg would be used) or in BIOS/Legacy mode (in which case it'll be E:\syslinux\archiso_sys-linux.cfg).

Currently I only tried in UEFI mode.

> When dealing with Athena ISOs if you want to have any hope of finding the boot media (which, I am aware per the original issue still doesn't seem to work if you change the label, but which, for most distros, should be enough, even with a user-specified label).

Athena ISO is created by using Archiso, and the ISO label is defined in [profilede.sh](https://github.com/Athena-OS/athena-iso/blob/main/archiso/profiledef.sh) that defines several variables that will be used in syslinux and grub config files (i.e., `%ARCHISO_LABEL%` variable). So, I think it is enough only to change `iso_label="athena-2022.11.13"` by another label... It could be `athena-os`. It should be good also for Rufus. I will submit a PR and I start to test by creating a new ISO and using `athena-os` string as ISO label. It will require around 30 min.

D3vil0p3r, 2022-11-13:
Here when I use the new ISO with the new assigned label on Rufus:
[image]

It failed again because GRUB (or better, Linux environment) reads string in case-sensitive. `athena-os` as label does not work, I added manually `ATHENA-OS` and it worked. So, I will update the `iso_label` as `iso_label="ATHENA-OS"`.

Here the logs:
```
Rufus x86 v3.20.1929 (Portable)
Windows version: Windows 11 Home, 64-bit (Build 22621.819)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using 'autorun.inf' label for drive E: 'athena-2022.11.13'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1911, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x002F6D5B
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 14.6 GB (15723659264 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/syslinux/syslinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/pre2-11-gbf6db5b4 (from '/syslinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'athena-os'
  Size: 2.1 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
Using image: athena-2022.11.13-x86_64.iso (2.1 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from 'C:\Users\Antonio\Downloads\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\' for Syslinux installation

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.6 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive2 for exclusive write access
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Syslinux MBR
Found volume \\?\Volume{9df42b13-6382-11ed-a489-48a472bde056}\
Installing Syslinux 6.04...
Opened \\?\Volume{9df42b13-6382-11ed-a489-48a472bde056} for shared write access
Using existing 'C:\Users\Antonio\Downloads\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\ldlinux.sys' ✓
Using existing 'C:\Users\Antonio\Downloads\Rufus\syslinux-6.04\pre2-11-gbf6db5b4\ldlinux.bss' ✓
[... 97 lines cut ...]
Extracting: E:\syslinux\vesa.c32 (2.9 KB)
Extracting: E:\syslinux\vesainfo.c32 (2.1 KB)
Extracting: E:\syslinux\vesamenu.c32 (26.1 KB)
Extracting: E:\syslinux\vpdtest.c32 (1.8 KB)
Extracting: E:\syslinux\whichsys.c32 (2.4 KB)
Extracting: E:\syslinux\zzjson.c32 (3.5 KB)
Created: E:\syslinux.cfg → /syslinux/syslinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
Using 'autorun.inf' label for drive E: 'athena-os'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1911, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0069C2A5
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 14.6 GB (15723659264 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2022-11-13:
Nah, the issue is that Arch recently added a new:
```
search --no-floppy --set=root --label ARCH_202211
```
line to their GRUB config script, and since it's new, and we don't replace everything with a label willy-nilly, we [don't yet have code in Rufus to patch the search](https://github.com/pbatard/rufus/blob/4cfe1c9947b437a0a274ca18c1d28afa8a160a80/src/iso.c#L371-L377). 

This means that, as soon as you don't have a FAT compliant label there, boot is going to fail.

I have a fix for this which I will commit shortly.

pbatard, 2022-11-13:
This should be fixed now (though I would still advise to follow Arch's example and use a default FAT compliant label for the ISO).

Feel free to test with the artefact from https://github.com/pbatard/rufus/actions/runs/3456902660 once it's built.

commit 165127d2 (2022-11-13): [iso] fix GRUB boot for newer Arch and derivatives

* Arch recently added a "search --no-floppy --set=root --label <LABEL>" into their
  grub.cfg, which we didn't have provision for patching, which means that, as soon
  as the user changed the label or used an Arch derivative with a label that isn't
  compliant with FAT (e.g. Athena Linux), search, and therefore boot would fail.
* Also alter the code so that we modify for more than one token if needed.
* Closes #2086
closed 2022-11-13 by pbatard