pbatard/rufus #2087: Is it possible  to have boot-able UEFI installation win10 usb flash with single partition without the second UEFI partition?

loa111, 2022-11-14:
Is it possible to have boot-able UEFI installation win10 USB flash with single partition without the second UEFI partition?
Can you place all folders and files on uefi partition?

JonnyTech, 2022-11-14:
You can do it yourself: partition your drive with a single FAT32  or NTFS partition, than copy/extract all of the files from the Windows 10 ISO into that partition and boot your computer from that drive.

But there are a few caveats, the main one being that your computer's UEFI must support booting from a NTFS partition, otherwise you need to use FAT32 which has a 4Gb file size limit so you need to split the big `install.wim` file so that it can be written to the filesystem.

Please see other issues, ie #1647 and similar.

pbatard, 2022-11-14:
@JonnyTech has already provided the answer.

But my question is: In what manner do you believe that having the 1 MB UEFI:NTFS partition at the end of the drive is a problem?
Because I am not aware of any issue linked to having a second partition, so if your problem is that it irritates you to have 2 partitions instead of one, I hope you can appreciate how this can be perceived as a rather frivolous and pointless pursuit...

Thereby, unless you can point to a technical issue linked to having two partitions rather than one, I will close this request.

closed 2022-11-14 by pbatard

loa111, 2022-11-14:
It's because the second USB stick was ruined. Stick had firmware lost. I know it's because the stick was used between multiple windows 7 and windows 10 machines, and sporadic fixes that can occur will mess with the partitions. After that, only the rufus is able to format it back to a single non-bootable partition.
Personally, I know that a single partition is safer or just disabling in windows option of fixing the file structure on the USB. The worst thing is when both partitions don't mount in the proper time, and windows offer to format the drive. If you choose to format at that moment, firmware is lost.

It's crazy how USB firmware were fragile.

pbatard, 2022-11-14:
> It's because the second USB stick was ruined.

First time you mention the existence of a second USB drive.

>  I know it's because the stick was used between multiple windows 7 and windows 10 machines

I can guarantee you it's not. The only thing that makes USB sticks fail if bad flash (or repeated write operations, or a buggy firmware from its manufacturer).

> I know that a single partition is safer

Citations needed, because it goes against what every specialist will tell you.

There is no such thing as "special" data blocks on a flash based media. As far as the internal flash controller is concerned, because that is how every single block based disk media is concerned, the flash memory blocks used for partitioning are no different than the flash memory blocks used for data.

So I am calling bullshit on your assertion, because, while it may look like a possibility when you don't know how flash based media work internally, it's a completely ludicrous hypothesis when you do. Flash based media are 100% data agnostic, which means that, again, creating zillions of partitions on them, or having the OSes you mount them to access these zillions of partitions, has absolutely no bearing on the reliability of the drive.

> It's crazy how USB firmware were fragile.

Yes, flash memory used in consumer USB sticks is A LOT more unreliable than people realize and __will__ fail usually A LOT sooner than people hope. And that is the only reason your drive failed because it's physically impossible, due to the data agnosticity of the embedded controller that governs how the flash media blocks are written, that extra partitions or multiplying accesses between Windows 7 and Windows 10 can damage your drive. Oh, and these wouldn't even result in damaging extra read access (if you want to assert that the simple fact of reading the data that describes the extra partition content would damage a drive), because that would account for only a handful of blocks, and, even if you were to mount and unmount the drive every 5 minutes, if your drive can't handle that minimum amount of extra reads (which will be dwarfed by the amount of reads it will issue when accessing the content of a medium sized file), then your drive has a major problem.

Again, you are looking at the wrong place when it comes to the cause of your issue, and are, without any evidence, trying to point the finger at something (extra partitions) that any person who knows the internals of flash based media will point out can simply not be the cause of your issue.