pbatard/rufus #2093: GPT FAT32 USB flash drive boots to Windows from hard drive instead of starting installer

Topkek-190, 2022-11-24:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
When I tried to boot a Windows FAT32 GPT UEFI usb pendrive formatted by Rufus, instead of booting the installer, it boots to Windows from my hard drive. I've tried booting it using the BIOS menu as well as Windows Recovery but it still boots to Windows

Log
---
```
Rufus x86 v3.20.1929
Windows version: Windows 10 Enterprise LTSB, 64-bit (Build 17763.3650)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1988, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {61ABB2B2-FB1E-47BC-BA9C-EF0ED76C4B8C}
Max parts: 128, Start Offset: 17408, Usable = 16357751296 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {1F2933DA-95D7-40E1-A044-74EB7B9BD480}
  Size: 15.2 GB (16356720128 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CES_X64FREO_EN-US_DV9'
  Detected: Windows 10 ISO (Build 19041.1)
  Size: 4.5 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: en-us_windows_10_iot_enterprise_ltsc_2021_x64_dvd_257ad90f.iso (4.5 GB)

Full Log: https://dpaste.org/zJUXd/raw
```

pbatard, 2022-11-25:
That's a UEFI/BIOS issue, not a Rufus issue.

First of all, are you 100% sure that your target computer has a 64-bit UEFI firmware?
Because the only bootloader present on your ISO (`en-us_windows_10_iot_enterprise_ltsc_2021_x64_dvd_257ad90f.iso`) is `efi\boot\bootx64.efi` (per your log about), if you happen to be trying to boot on a machine that has an x86 32-bit UEFi firmware (which can happen __even if__ you have an x86 64-bit CPU), then you will have exactly the kind of issue you report.

Or, considering that you are using IOT, you may be using a completely different CPU architecture on your target system, such as ARM (in which case  `efi\boot\bootx64.efi` will never work).

At any rate, since you don't actually need Rufus to create your media on account that the `sources\install.wim` from your ISO is smaller than 4 GB, I can guarantee that if you simply create a FAT32 partition on your USB, and extract the whole content of your ISO using something like 7-zip, you will see the exact same issue, whereas, if you are actually using an ISO that is compatible with your system, this should just work.

Therefore, unless you can demonstrate that when you extract your ISO onto a FAT32 partition it boots, whereas when you use Rufus it doesn't, I have no choice but to dismiss your issue as something that has nothing to do with Rufus.

closed 2022-11-25 by pbatard

Topkek-190, 2022-11-25:
It seems to be a UEFI issue but Windows's ESP has EFI\Boot\bootx64.efi and Windows boots normally so I can't understand why it wouldn't boot from the USB.
As for the architecture thing, my PC is an x86-64 Intel and the iso is  `en-us_windows_10_iot_enterprise_ltsc_2021_x64_dvd_257ad90f` which is for the x64 architecture so it's definitely not an architecture mismatch

pbatard, 2022-11-25:
Again, please create a FAT32 USB with a `efi\boot\bootx64.efi` and try to make that boot. As long as you can't demonstrate that, when you do that, your USB boots, but not when your USB is created with Rufus, then your issue clearly is unrelated to Rufus.

Most likely, if you do have a `efi\boot\bootx64.efi` for Windows, then you didn't set up your UEFI firmware settings so that USB drives will have precedence over internal drives (there's pretty much __ALWAYS__ a setting, and sometimes multiple ones, in your UEFI firmware that governs this). Those settings a specific to the motherboard and UEFI firmware you use, so you will need to read your manual, because noone but people who have the exact same motherboard will be able to tell you precisely what you need to do.