pbatard/rufus #2094: MSYS2 and error: unrecognized command-line option ‘-Wbidi-chars=none’

noloader, 2022-11-27:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

(Most of the steps in the checklist are not applicable since I am trying to build the app. I have not gotten as far as running the app).

Issue description
-----------------
I'm working on Windows 10 x64, fully patched. I installed MSYS2 from www.msys2.org. All msys2 packages are up to date.

`configure` runs Ok, but `make` fails with:

```shell
$ make V=1
Making all in src
make[1]: Entering directory '/home/Jeffrey Walton/rufus/src'
Making all in ../.mingw
make[2]: Entering directory '/home/Jeffrey Walton/rufus/.mingw'
/usr/bin/sed "s/@.*//" dwmapi.def >dwmapi.def64
dlltool --input-def dwmapi.def64 --output-delaylib dwmapi-delaylib.lib --dllname dwmapi.dll
/usr/bin/sed "s/@.*//" wintrust.def >wintrust.def64
dlltool --input-def wintrust.def64 --output-delaylib wintrust-delaylib.lib --dllname wintrust.dll
rm wintrust.def64 dwmapi.def64
make[2]: Leaving directory '/home/Jeffrey Walton/rufus/.mingw'
Making all in bled
make[2]: Entering directory '/home/Jeffrey Walton/rufus/src/bled'
gcc -DPACKAGE_NAME=\"rufus\" -DPACKAGE_TARNAME=\"rufus\" -DPACKAGE_VERSION=\"3.21\" -DPACKAGE_STRING=\"rufus\ 3.21\" -DPACKAGE_BUGREPORT=\"https://github.com/pbatard/rufus/issues\" -DPACKAGE_URL=\"https://rufus.ie\" -DPACKAGE=\"rufus\" -DVERSION=\"3.21\" -D_GNU_SOURCE=/\*\*/ -I.    -DWINVER=0x601 -D_WIN32_WINNT=0x601 -D_WIN32_IE=0x800 -D_FILE_OFFSET_BITS=64 -D_OFF_T_ -D_off_t=off64_t -Doff_t=off64_t -Doff32_t=long -DUNICODE -D_UNICODE -UNDEBUG -DCOBJMACROS -D__USE_MINGW_ANSI_STDIO=0 -std=gnu99 -Wshadow -Wall -Wformat-security -Wundef -Wunused -Wstrict-prototypes -Wno-restrict -Wno-array-bounds -Werror-implicit-function-declaration -Wbidi-chars=none -Wno-pointer-sign -I./.. -Wno-undef -Wno-strict-aliasing -g -O0 -c -o libbled_a-bled.o `test -f 'bled.c' || echo './'`bled.c
gcc: error: unrecognized command-line option ‘-Wbidi-chars=none’
make[2]: *** [Makefile:333: libbled_a-bled.o] Error 1
make[2]: Leaving directory '/home/Jeffrey Walton/rufus/src/bled'
make[1]: *** [Makefile:511: all-recursive] Error 1
make[1]: Leaving directory '/home/Jeffrey Walton/rufus/src'
make: *** [Makefile:287: all-recursive] Error 1

$ gcc --version
gcc (GCC) 11.3.0
Copyright (C) 2021 Free Software Foundation, Inc.
```

I seem to recall GCC removed tri-graph support in GCC 13. I am surprised to see the error under GCC 11. Also see the [GCC 13 Change log](https://gcc.gnu.org/gcc-13/changes.html). Maybe there needs to be a specific Autoconf test for support of `-Wbidi-chars`?

Here are a couple of related bug reports that were dismissed as "wrong platform": [Issue 1939](https://github.com/pbatard/rufus/issues/1939) and [Issue 2060](https://github.com/pbatard/rufus/issues/2060). It looks like they were portents of things to come.

Log
---
[config.log.zip](https://github.com/pbatard/rufus/files/10097373/config.log.zip)

pbatard, 2022-11-27:
> `gcc (GCC) 11.3.0`
> `Copyright (C) 2021 Free Software Foundation, Inc.`

If you were really using an up to date msys2, it should report 

```
$ gcc --version
gcc.exe (Rev6, Built by MSYS2 project) 12.2.0
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

> All msys2 packages are up to date.

I seriously doubt that. Or you have a gcc override somewhere that you forgot about.
Oh, and the __clean__ MinGW build, from 5 days ago, [using GitHub actions](https://github.com/pbatard/rufus/actions/runs/3523880406/jobs/5908571667) showed no issue. Considering that I just updated my plaform and saw no issue either, and compounding this with the gcc version you report, the probability that msys/MinGW addeda breaking change is much lower than the probability that this is a pure environmental issue on your end.

noloader, 2022-11-27:
>> All msys2 packages are up to date.
> I seriously doubt that. 

Well, they were installed fresh tonight. I also ran `pacman -Suy` several times to update the system. I stopped running it when there were no more updates. Maybe the package manager is serving old packages?

Here are the steps I took: I visited https://www.msys2.org/, and I downloaded [`msys2-x86_64-20221028.exe`](https://github.com/msys2/msys2-installer/releases/download/2022-10-28/msys2-x86_64-20221028.exe). Once installed a MinGW terminal automatically launched. In the terminal I updated with `pacman -Suy` several times to update the system. I then installed `mingw-w64-x86_64-toolchain base-devel autotools git upx` as instructed. I also installed `gcc clang subversion zip unzip`. (And then I cloned, configured and tried to make).

When I run `gcc --version`, I see version 11.3.0.

Perhaps you get a slightly different environment on Windows 11?

> so, no, these were never in any way shape of form "portents of things to come". 

The compile error was present months ago. It is surfacing again. It doesn't take an apothecary to read the tea leaves.

pbatard, 2022-11-27:
> `When I run `gcc --version`, I see version 11.3.0.

When you run `gcc --version` you __DON'T__ see `Built by MSYS2 project`. And you __DON'T__ see `2022` as the copyright. Does that not give you a hint that, __maybe__ you are not actually invoking the `gcc` you are supposed to invoke.

> Perhaps you get a slightly different environment on Windows 11?

Right, and you believe that the msys/MinGW folks did not update `gcc` for Windows 10 in the past 11 months?
Windows 11 is just Windows 10 revamped. There are not enough differences to have the msys2 folks create a different version for Windows 10 and Windows 11 (after all, the main selling point of Windows is its backwards compatibility).

> It doesn't take an apothecary to read the tea leaves.

Apparently it does.

From issue #1939:
```
gcc version 11.2.0 (Ubuntu 11.2.0-19ubuntu1) 
```
And from issue #2060:
```
uname -s = Linux
```

As I said, people trying to invoke Linux's __NATIVE__ gcc, that can only produces ELF code, to compile a Win32 application, that should be PE32.

Again, all signs point to the `gcc` you are invoking not being the `gcc` that should actually be invoked for MinGW. Look into that. For one thing, your `PATH` may not be what you think it is, and may be picking up an alternate `gcc` somewhere along the way.

pbatard, 2022-11-28:
https://github.com/pbatard/rufus/actions/runs/3566821954, i.e. the MinGW GitHub Actions build for today's release, that __ALWAYS__ used the most up to date msys2/MinGW further demonstrates that the problem is purely on your side. As such I will close this issue.

closed 2022-11-28 by pbatard