pbatard/rufus #2096: I run rufus and no windows appears, but the process shows rufus is running?

Jibun-no-Kage, 2022-11-27:
﻿I run rufus and no windows appears, but the process shows rufus is running?  Can't click on anything because nothing is visible, have to kill rufus by killing process.  Completely useless.  Help!

Feathered-Serpent, 2022-11-28:
Does Rufus appear in the task bar? If yes, try clicking onto it (so it would be the active window) then press and hold WINDOWS + SHIFT buttons, then use the LEFT/RIGHT arrow buttons to see, if you can move the window onto your active display.

pbatard, 2022-11-28:
Thought I had already replied to OP, but I guess I got sidetracked with other things and closed the browser window with my reply.

1. Disconnect all USB devices and try to launch Rufus again.
2. Please check that you are not using any of the known software that is incompatible with Rufus and listed [in this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software). Especially, since you seem to be experiencing slow startup time, please make sure that you are not using Sophos Antivirus, as it is [known to cause the exact kind of issue you are experiencing](https://github.com/pbatard/rufus/issues/1597#issuecomment-723462063).
3. If that still doesn't work, please install an run [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview) as Rufus does send its log messages there __even if there is no UI window appearing__. You should see at least a line like:  
    ```
    [13076] *** Rufus init *** 
    ```  
    followed by a bunch more log lines. If you copy/paste that log, it should tell us where Rufus appears to be stuck.

Jibun-no-Kage, 2022-11-28:
Actually I figured it out, turns out it was a bad USB thumb drive.  It would appear on the USB bus, but would hang, so causing all kinds of issues with system IO, so rufus would run, but never fully appear.  Even if I killed the process, and restarted, would not appear.  Then multiple applications were doing it, soon after, but when when the bad thumb drive was inserted.

closed 2022-11-28 by Jibun-no-Kage