pbatard/rufus #2099: "Allow the provision of an arbitrary local account password" option

SlyCooperReloadCoded, 2022-12-01:
Issue description
-----------------
When using the new "Allow the provision of an arbitrary local account username" option for a Windows 10 install, which is a really nice option by the way, it does speed up the install process.  However, upon logging out and logging back in, it'll say that the password needs to be set and lets you do so.  I work for a nonprofit that refurbishes computers and installs Windows 10 on them en masse, so we need USB installers for lots of computers that all have the same account name and no password, so an option to set password (or left blank to disable password) would eliminate this issue and be a nice addition for people in any other situation.  I see #1954 mentions local username and password but the password part doesn't seem to be added yet.  Are there plans for this?

pbatard, 2022-12-01:
> Are there plans for this?

No.

The reason is that the only goal of Rufus' customisation is to avoid regular users having to jump through hoops when installing Windows, and this is aimed at a single installation by a single user.

What you want to accomplish is actual customisation, for sysadmin purposes, and, per [the following FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want), Rufus will never be designed as a sysadmin/power-user tool.

Thus you will need to sort out your `unattend.xml` with your Base64 encoded password yourself (you shouldn't have much trouble to find heaps of online documentation on how to do that), as Microsoft does provide means and extensive documentation for sysadmins like you who want to customize their installation and automate some of the setup tasks, such as creating a default account.

Plus, one thing you need to realize is that we purposefully set a blank password so that the user gets logged in automatically and so that setup gets finalized without user input, which is our explicit goal with our customisation options. But if you add a non blank password, you will not get that streamlined first install.

So, again, I will have to respectfully ask that, if you plan to carry our sysadmin operations, you look at using an actual sysadmin toolbox for that job (which is crafting your own `unattend.xml`) and not expect Rufus to do that for you, as it is not what it is designed for.

closed 2022-12-01 by pbatard

SlyCooperReloadCoded, 2022-12-01:
If it purposefully sets a blank password, then why does it force you to change your password when you log in a second time?  Installs started doing this after I started to use the option, and that's why I suggested this, so it seems like something else is going wrong.

pineapple63, 2022-12-02:
I can only guess, but Windows might be "expiring" the blank password once it has logged you in the first time, and so it would force you to change it the second time

pbatard, 2022-12-02:
> Why does it force you to change your password when you log in a second time? 

Because allowing users to keep a blank password is terrible security-wise. A blank password is fine for first logon, but you __REALLY__ don't want to keep it for longer than needed. And again, we need blank on first logon for automatic logon & install finalization, so that the install process behaves in the same manner as if the user had created an account during the setup phase (because when you do create an account during setup, Windows will reuse the typed in password for first logon, but, for some reason that only Microsoft knows, it doesn't do that with an `unattend.xml` password).

Our goal is to have an installation that behaves as close as possible to a standard installation, but without the nagging or the interruptions, i.e. what we assert a Windows installation _should_ be if Microsoft weren't prioritizing their welfare over the user's...

> Windows might be "expiring" the blank password once it has logged you in the first time

Nah, this is not automated behaviour. We purposefully [add a command to force a password change on next logon](https://github.com/pbatard/rufus/blob/master/src/wue.c#L167-L173).

Maybe when you look at it from a sysadmin's perspective, it doesn't make sense, but when you look at it from the end user's perspective, who just wants to get to a running Windows that they can use and customize without being interrupted, i.e. once the install has started, you should be able to go grab a coffee and come back to a fully installed Windows, and __NOT__ to a prompt asking you to customise this or that or a logon prompt. And that's why the _Windows User Experience_ of Rufus is designed the way it is because, with a couple clicks, it will give you just that. But it won't be able to give you that if it lets you enter your own password...

pbatard, 2022-12-02:
Also, I should point out that this is not the only element that went into our decision not to allow the user to enter a password, because, if we did, you can bet that there would be some users who would insist that Rufus might include some malicious code that sends their typed in password to a remote server or something.

In other words, you'd be surprised at the amount of mistrust people have about an application, especially if that's one that asks them to type in a password. If I were to do that, I'd never see then end of having to justify the fact that, no, Rufus isn't collecting your passwords for nefarious purpose and try to demonstrate to folks that, even though the code is all there laid our in black and white, I haven't somehow been a genius at obfuscated the malicious behaviour (because, yes, I have actually been accused of that before). I get enough crap from Antivirus X suddenly reporting that Rufus contains malware, so boy do I not want the perfect storm of someone pointing to Antivirus X, then pointing to Rufus letting people enter their Windows password, and jumping to the conclusion that Rufus _must therefore_ be malware...

SlyCooperReloadCoded, 2022-12-02:
The computers we refurbish end up getting donated to charities and similar organizations that will inevitably set up a password themselves.  I do see all the points against it though.  We're not a very high-budget nonprofit, but I'll look into that stuff you mentioned.  This is all very informative.