pbatard/rufus #2106: Versión 3.21 provision of a local account username in Windows

DroVerGaDe, 2022-12-07:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Versión 3.21 provision of a local account username in Windows>

Log
---
```
<Sorry for my english... I use this new version of Rufus 3.21 and I found the ability to put a local account username. The problem is I restarted the system and now is asking for a password, so my question is what password Ihave to use???>
```

DroVerGaDe, 2022-12-07:
Sorry for my english... 
I use this new version of Rufus 3.21 and I found the ability to put a local account username. 
The problem is I restarted the system and now is asking for a password, so my question is what password do I have to use???
Thanks in advance

pbatard, 2022-12-07:
It's a blank/empty password.

Just press enter.

pbatard, 2022-12-10:
No update &rarr; Closing

closed 2022-12-10 by pbatard