pbatard/rufus #2107: Multi partition USB

ECAsh001, 2022-12-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Older hardware, Not so small USB, Multi partitioned USB, Wont let me choose the partition

No log unless you need me to RE- multi partition my 32 gig USB.  Then try and get Rufus to only open an ISO into 1 of them.  It dont like it at all, and tells be its going to format the Whole thing.
Why?  Im not using the USB to run the machine, Im installing to.
I was using the portable version of Rufus.  It did see that My USB had more then 1 partition.  Am I going to need to File copy it, then Mark the partition as bootable?  Would rather have all 3 on the USB bootable, le the Computer get confused and ask me Which one to boot.

Edwardchalesash@gmail.com
```
<FULL LOG>
```

JonnyTech, 2022-12-10:
https://github.com/pbatard/rufus/wiki/FAQ#Why_does_Rufus_need_to_delete_all_partitionsvolumes_when_creating_a_drive

pbatard, 2022-12-10:
Yup, this gets asked on a regular basis, and, per the link above, the goal of Rufus is to create a USB that boots, and that requires starting from a known state, which you can't achieve with trying to preserve partitions (plus, if, say you partitioned your drive as GPT, then there is __no way__ Rufus will be able to use that drive to boot on BIOS machines, even if it tried to support partition preservation).

As your question has been answered, I will close this issue.

closed 2022-12-10 by pbatard