pbatard/rufus #2108: Rufus failed to create .iso with autorun.inf [The system cannot find the path specified]

willzyx-hub, 2022-12-10:
Hello Devs my rufus failed to create iso with autorun.inf in this case i used Debian .iso which have autorun.inf :) here is my log

Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 11 Pro, 64-bit (Build 22621.521)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {B7FE567B-C33B-4570-93D5-F60186E5DBE9}
Max parts: 128, Start Offset: 17408, Usable = 15669885440 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Microsoft Basic Data'
  Detected File System: FAT32
  ID: {0B8E4ECA-4203-46A8-BA30-61B7511ED522}
  Size: 14.6 GB (15668854272 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20200816 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Debian testing amd64 1'
  Size: 3.8 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: debian-testing-amd64-DVD-1.iso (3.7 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.6 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for exclusive write access
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{e68be478-de56-43aa-ae7c-a4d5cde6ba33}\
Successfully remounted \\?\Volume{e68be478-de56-43aa-ae7c-a4d5cde6ba33}\ as E:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Rock Ridge extensions (if present)
Extracting: E:\autorun.inf (146 bytes)
  Unable to create file: [0x00000003] The system cannot find the path specified.
Created: E:\syslinux.cfg → /isolinux/isolinux.cfg
Error: Can't open file 'E:\md5sum.txt'

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1905, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {34C77DB0-20CA-4AA6-92CC-DF5879EB8DA0}
Max parts: 128, Start Offset: 17408, Usable = 15669885440 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {9256DA24-E36D-42CD-B699-834370834E85}
  Size: 14.6 GB (15668854272 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000

! Rufus can't be runned on Second time it has no response when i opening on Second time so i need to redownload.

pbatard, 2022-12-10:
This is 100% your security solution.

See https://github.com/pbatard/rufus/wiki/FAQ#List_of_hardware_and_software_Rufus_isnt_compatible_with as well as https://github.com/pbatard/rufus/wiki/FAQ#why-does-rufus-create-an-autoruninf-file-isnt-that-dangerous (but at any rate, the `autorun.inf` came from your ISO and not Rufus in this case).

Nothing I can do about it. Please contact your AV vendor and tell them that their antivirus sucks big time if it treats all `autorun.inf` as malicious files.

closed 2022-12-10 by pbatard

willzyx-hub, 2022-12-11:
> This is 100% your security solution.
>
> See https://github.com/pbatard/rufus/wiki/FAQ#List_of_hardware_and_software_Rufus_isnt_compatible_with as well as [...]

But there is NO ANTIVIRUS INSTALLED EVEN I REMOVED THE WINDOWS DEFENDER COMPLETELY