pbatard/rufus #2115: > If I also have this problem in Home and Pro, do you think it might be related to UAC?

Windows10User2, 2022-12-16:
> If I also have this problem in Home and Pro, do you think it might be related to UAC?

No. It __is__ an active security solution or security rules preventing the creation of `autorun.inf`. And this will be my last reply before I lock this thread, because unless somebody else comes on this issue tracker with the same issue on an __unmodified__ official installation of Windows 11, it is 100% your problem to identify and fix, and, considering that this is external to Rufus, I consider that I've already given you enough of my time

_Originally posted by @pbatard in https://github.com/pbatard/rufus/issues/2039#issuecomment-1242754338_
 
It looks like the culprit wasn't a modified Windows 11 installation but a virus called Virus.Floxif which infects EXE files from installers and from already installed programs. I have some installers and EXE files stored in an external HDD and I always use them after installing Windows. Malwarebytes couldn't remove this virus and after running an infected installer or EXE files, the problem always came back, so I had to reinstall Windows and download and replace installers and EXE files (I scanned each and every one of them using VirusTotal to know which ones were infected) to get rid of this virus. Since then, I can finally use Rufus.

closed 2022-12-16 by Windows10User2

pbatard, 2022-12-16:
Thanks for this very interesting report. I added a note about this to #2039 and I will also try to mention this in the FAQ.

Windows10User2, 2022-12-16:
> Thanks for this very interesting report. I added a note about this to #2039 and I will also try to mention this in the FAQ.

It doesn't infect autorun.inf files like you said in #2039 but EXE files like I said here. Since Rufus is an EXE file, I also had problems with it.