pbatard/rufus #2117: Found what caused #839 and it is important !

Svetoslav92, 2022-12-18:
Seems that guy  was really mad and couldn't make his points clear but he has right.
I found out what's the issue.
My pendrive is 128gb and it got corrupted really badly after Rufus with no chance of diskpart etc as the mentioned issue.
It turns out that if you mount your pendrive with custom driver (mine was so that the drive is detected as sata drive not usb drive), Rufus finds a hidden partition (usually not exposed when running generic usb driver) and deletes it. This partition is possibly holding the firmware for the pendrive itself ! Of course after formatting it, there's no way making this pendrive work except finding a possibility to reflash the firmware.
This should be tested and people should be warned!

Thank you :)

pbatard, 2022-12-18:
> but he has right

I'm sorry but without actual evidence, especially when the guy said he used 3 drives from different brands and they all failed, I beg to differ. It's jut impossible for 3 drives of different brands to be as ill conceived as what you describe.

> My pendrive is 128gb

<Sigh> I'm so tired of people who want to provide an example of a failed drive,. and yet __consistently__ fail to provide the brand and model of their drives.

You do realise that it's a lot easier for everyone (and not just the developer) to validate your information if you __actually__ provide verifiable data.

Manufacturer and exact model of your drive, __PLEASE__.

 >  pendrive with custom driver 

Again, you are withhodling data here, which makes it VERY DIFFICULT to validate your __claim__. If you installed a driver, there must be a download URL somewhere. How about you provide that?

> Rufus finds a hidden partition

If your drive is as ill conceived as using a not-so-hidden partition, with a custom driver that only exists to hide the partition in Windows so that it doesn't get deleted, that's 100% an issue with the drive, not Rufus, because you do realise that, if that is the case, you just have to plug that same drive on Linux or Mac and repartition it to replicate the issue.

If that is true, then this is __exceedingly__ poor drive design that needs to be reported to the manufacturer (especially as they can easily craft a firmware that hides the partition internally rather than externally so as to avoid the _alleged_ issue you describe).

> This should be tested and people should be warned!

And how on earth is one supposed to do that when:
- Do __not__ provide the brand and manufacturer or your drive
- Do __not__ provide a URL to the custom Windows driver you installed

I'm sorry but, considering how extraordinary your claim appears to be, how idiotic from the manufacturer it would be to use a custom Windows driver that is going to result in a drive failure the minute someone attempts to use it on Linux or Mac, how easy it is for a drive manufacturer to hide a partition internally (just reserve a few blocks and then tell the USB host that the only blocks accessible are the ones after that, and use custom USB vendor commands to write data to the hidden partition if needed), and how you _strangely_ chose to withhold any element of data that could back up your claim, I have no choice but call bullshit on this report or, if there's any element of truth to what you describe, deflect the full responsibility of the failure to the drive manufacturer having done an incredibly poor job with the design of their device.

If your idea that drives with an "issue" like the one you report are __common__ and are the reason why people experience failures when using Rufus, prepare to be disappointed, because I can tell you with absolute certainty, due to how extraordinary your claim is, and how it would have become public knowledge if true (on account that Linux and Mac folks would have to be sitting on a mountain of dead USB flash drives, and would long have been reporting about this), what you are asserting here is exceedingly unlikely to be the reason why one person would end up with 3 dead drives from 3 different brands.

So you're just going to have to do better than this is you want to try to pin the blame on Rufus for drive failures. Just like Linujx or Mac, Rufus does not have the capability to destroy properly designed USB Flash Drives period.

closed 2022-12-18 by pbatard

Svetoslav92, 2022-12-18:
You just closed it as I was going to give more info. I'm not here to put any blame on Rufus or onto you. You seem paranoid on this convo, which is strange. I like the product. I'm just telling that people should be warned.
Flash drives are stupid by design and yes, I guess you put custom driver and the madness happens.
I was here to give you clues and you call this bullshit. The pendrive is hama rotate 128gb usb 2.0 .
And I'm sure about it because I got a rufus warning that there's hidden partition and it will be deleted ! I clicked OK ! Afterwards everything was stating errors like corrupted MBR and write-protected drive.
After this happened the pendrive wasn't usable until I reflashed the firmware. 
The custom driver - > https://www.getusb.info/usb-hack-turn-a-usb-stick-into-a-hard-drive-or-local-disk/
A german guy has done it for 64 bit systems (google it if interested) or ask me to provide you the driver itself.

Anyway I've left enough clues for you to test and I was here just to help, not to blame you or Rufus...

JonnyTech, 2022-12-18:
@Svetoslav92 that is a serious bit of hackery and not something that a casual (Rufus) user would contemplate to attempt. Obviously you have a niche user case for your needs as a somewhat advanced user so, as such, the responsibilty for attempting such a risky procedure is entirely in your hands. Although we appreciate your report, please note that Rufus is meant for regular Windows users with a standard environment. Adding your findings to the FAQ may confuse users and potentally lead them to the **false** impression that Rufus is dangerous or unsafe.

On a side note, to satisfy my curiousity, why not simply use a SSD drive with a USB adaptor? They are cheap, reliable, fast and plentiful nowadays.

Svetoslav92, 2022-12-18:
@JonnyTech  I agree that's not usual setup. But if you have different driver people should be warned of the consequences and how to avoid. You know those autodriver/update installers for windows ? They infact can install something different than the generic usb driver for windows (heard that there are intel ones or who knows what else) and if that was the case for #839 the pendrive may experience the same.

JonnyTech, 2022-12-18:
@Svetoslav92 as I understand, you are requesting that @pbatard adds a check within Rufus to check for the specific driver installed on the system and rufuse (or warn) to continue if it is found? Whilst a resonable request, if your issue is independently confirmed to be accurate, let me state again that it is something that a regular Rufus user would do and would take up far too much time for the sole developer of this application. As with any open source solution, you are free to fork the code and create a pull request for the workaround. But again, as an advanced user, you could simply not use the custom driver with paritioning software.

pbatard, 2022-12-18:
> And I'm sure about it because I got a rufus warning that there's hidden partition and it will be deleted !

There's no such warning in Rufus.

Rufus does not produce any warnings about "hidden" partitions. The only warning it produces is about _multiple_ partitions, which is absolutely fine on any drive and will never ever render a drive inoperable if deleted, but there's nothing about "hidden" partitions.

Therefore, I would greatly appreciate if you stopped inventing bullshit and clutching at straws to put forward your pet theory of what _might_ have happened, when it is clear that you actually don't know what actually happened, and therefore do not have nearly enough "evidence" to try to pin the blame on Rufus, __especially__ when you are unwilling to try to replicate your issue to, at the very least, confirm that, in your specific environment, what you experienced wasn't actually a fluke.

> I was here to give you clues and you call this bullshit. 

Well, the "clues" from your initial report were as good as _"The culprit may have been a man... or maybe a woman... I did get a very good look at it, and I could describe it in details (i.e. I have been using the specific driver I am claiming makes Rufus go haywire), but I'm not going to provide you with these details because all you need to know is that it was a human person"_

So I have been calling bullshit because you attempted to report an issue without providing enough details (I still fail to understand why you would leave out the most critical piece of information such as the type of driver and its URL), as well as made a __huge__ assumption, that you cannot validate in any way shape or form, that the person from #839 must have been using a custom driver similar to this.

I also see it as an impossibility that a driver would be able to render a drive inoperable in the manner your describe, unless it issues __USB vendor commands__ to the flash drive.

Furthermore, Rufus is designed not to list hard drives or local disks, so if your driver actually works as designed and changes the type of the drives as listed by Windows, then they will not be listed in Rufus.

You are also doubling down on bullshit such as _"autodriver/update installers (that can) infact (...) install something different than the generic usb driver for windows"_, with an example that _intel_ might do it, whereas a quick search on the internet should comfirm that intel have never produced USB flash drives (they produce SSDs and other storage media but not flash drives, and you can't go on amazon or anywhere to buy an "intel flash" drive). I am also not aware of consumer USB flash drives that would need a custom USB driver, outside of encrypted ones (but then you would think that people reporting failures when using Rufus _would_ have reported that), especially as the generic USB mass storage drivers, which is what Windows, Linux and Mac use, are engineered precisely so that, unlike what might be the case for other devices, the user should never ever have to install a custom driver. So here again, another "clutching at straw" bullshit claim, where you cannot provide an actual example that would be relevant to our case, if you want to assert that some users who experience drive failures when using Rufus must have had a custom mass storage driver installed through Windows Update.

It also appears pretty clear that while you are making a claim for reproducible behaviour, you yourself only ever experienced it once, and are unwilling to back the claim by replicating the issue in your environment even as you indicate that you can recover your drive.

Finally, I had a quick look at your thread, and the driver you describe is a filter driver that is designed to make a drive with the REMOVABLE flag set be seen as a FIXED drive. This was designed with the purpose of allowing cheap USB drives to be used for Windows To Go. Nowhere does it mention anything about _"hidden partitions"_, which is what you claim the driver did create, and if this is all the driver does, then I can guarantee that it is still impossible for Rufus to repartition/reformat your drive in a manner that will make it look dead, because Rufus handles FIXED and REMOVABLE drives the same way.

Thus, considering all the approximations and misreporting, all I am seeing here is clear intent of pinning the blame on Rufus by flogging things into the air (hidden partitions, automated driver installation, mischaracterised custom driver, since it doesn't "make a drive look as SATA", etc)  without the decency to try to at least __replicate__ the matter on your side to see if there's any credence to what you claim.

If you want to make a claim that Rufus renders a drive next to inoperable in some circumstances (but not inoperable since you were able to recover it), and you are adamant that you do have all the evidence you need to prove it (which strangely we had to almost exhort from you), then you should have no problem trying to make me here look like an arrogant fool by replicating what you did for a second time, and providing us with the log from Rufus, as it should contain a lot of insightful data that might prove invaluable if someone else attempts to replicate what I describe and __fails__ to do so.

At this stage, considering that half of what you state is far from even remotely receivable because it appears to be just stuff that you wish was true (hidden partitions, custom drivers that are installed automatically, a driv [... cut ...]
If you wanna make us believe you, please replicate the issue again, and send us the log from Rufus (since it should also contain some insightful information about this alleged hidden partition that appears to be crucial to the failure). You can always access the log using <kbd>Ctrl</kbd>-<kbd>L</kbd> or the small button left of <kbd>START</kbd>.
Or, if you haven't used Rufus since the failure, you should be able to find the last log in `C:\Users\<your_user_name>\AppData\Local\Rufus\`, which should also give us everything we need.

pbatard, 2022-12-18:
Bullshit claim is bullshit.

Please come back to us when you can describe how you installed that `cfadisk-x64-1.zip` from [here](https://www-hardwareluxx-de.translate.goog/community/threads/win-7-x64-removable-als-fixed-laufwerk-anzeigen-lassen-gel%C3%B6st.797059/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en-US&_x_tr_pto=wapp#post-16546825) on Windows 7 SP1 or later (with test signing enabled).

For one thing, one has to go through the process of editing the inf and creating the cat then signing it for test signing. And even after you do that, you can't just right click on the driver, but need to invoke command line.

But even if you do all that, it doesn't seem that the filter driver got applied to USB flash drives anyway, because I could not see it listed as a driver for the device in device manager and the drive was still being reported as `Removable` in Rufus (whereas the purpose of the filter driver is to make the drive be reported as FIXED).

And of course, no matter how many partitions I have on the USB Flash Drive, what type/brand of USB Flash Drive I use, and how many times I try to reformat them in Rufus, __NONE__ of them exhibited the issue you describe.

Therefore, I can only conclude, as I alluded from the very start, that your claim is complete and utter bullshit and, unless you do provide data on how exactly you installed the driver on your platform, with some evidence that it did properly alter the FIXED/REMOVABLE attribute, I will also allege that you __never__ actually used the driver you claim you are using, because it just doesn't seem to work at all on an up to date installation of Windows 7 and, at any rate, if the __only__ driver that purports to support your claim is that `cfadisk-x64-1.zip` one you pointed out (which, by the way, appears to be aimed purely at CompactFlash card devices), then, considering the hoops one needs to go through to install it, and how __no__ modern version that can be applied to regular Windows 10+ appears to be found anywhere, I am pretty confident that the number of people using that driver in conjunction with Rufus is so low that, even __if__ there was an actual issue when using it, it just wouldn't be worth doing anything about it.

So, as I __rightfully__ asserted, your claim is bullshit and all you accomplished with this bullshit claim is made me waste 2 hours of valuable time on a Sunday where I had __much__ better things to do.

But hey, feel free to come back and try to prove me wrong with __actual verifiable evidence__ where you do report each step you have taken, including how you went though the installation of the driver, along with Rufus logs, preferably from a test that you actually ran more than once.

Because until then, I will continue to call your claim as complete invalidated bullshit.

Svetoslav92, 2022-12-19:
Bullshit claim is bullshit.

Please come back to us when you can describe how you installed that cfadisk-x64-1.zip from [here](https://www-hardwareluxx-de.translate.goog/community/threads/win-7-x64-removable-als-fixed-laufwerk-anzeigen-lassen-gel%C3%B6st.797059/?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en-US&_x_tr_pto=wapp#post-16546825) on Windows 7 SP1 or later (with test signing enabled).

I never thought Rufus maker was such a disrespectful and bipolar person. I guess now I know. Just reboot, press F11 or whatever your system recovery button is, Troubleshoot -> Startup Settings and on this screen press 7
[image]
 
Install the driver.
Speechless.. Don't want to have any more convo in here with you. Do whatever you want.

JonnyTech, 2022-12-19:
> Please come back to us when you can describe how you installed that ... (with test signing enabled).

> Troubleshoot -> Startup Settings and on this screen press 7

Dude, by pressing 7 you have disabled driver signing, exactly the opposite of what was requested...

pbatard, 2022-12-19:
> Just reboot, press F11 or whatever your system recovery button is, Troubleshoot -> Startup Settings and on this screen press 7

Which accomplishes the same as using bcdedit to enable test signatures. This is not the problematic part, though, as I pointed out, you'll have to concede that if that cfa driver is the __only__ example you could find to _allegedly_ substantiate your claim, hardly anybody is going to be using this on the versions of Windows that Rufus supports, since it requires to disable signatures altogether on a permanent basis (which Microsoft makes very difficult on Windows 7, and impossible on later platforms) or using test signing, where you will get a permanent watermark on the desktop.

So, yeah, I'm calling your example bullshit from the get go, due to the jumping through hoops it requires on Windows and how nobody with a standard installation of Windows will be using that.

> Be sure that the inf file has your flash drive address replaced with yours as described here https://www.getusb.info/usb-hack-turn-a-usb-stick-into-a-hard-drive-or-local-disk/

Ah yes, you mean that article that is aimed primarily at Windows XP users, which Rufus has stopped supporting long ago.
Even if you try to apply this to Windows 7, you're going to have a lot of trouble making it work. I know I spent another hour trying and getting nowhere. But please, by all means, and this is addressed to __anyone__ who wants to give credence to this bullshit claim, feel free to prove me wrong.

> I'm talking about auto driver installer as all this crap for example and people install it often -> https://www.lifewire.com/free-driver-updater-tools-2619206

Would work A LOT better if you could provide an example of an auto driver installer replacing the Windows native USB mass storage driver. Come back when you can. Otherwise, as with all your "examples" so far, it's just another instance of something that only _looks_ like it can substantiate your claim, but doesn't work at all to substantiate it when you actually look at it closely.

> It was up to you to extract this knowledge and possibility to notify people.

Not when the "knowledge" is UTTER AND COMPLETE BULLSHIT.

Your driver cannot destroy a mass storage drive in the way you describe. That's because a flash drive controller simply present an area of flash data in a completely agnostic manner (meaning that you can write complete garbage or "special" data to any of the flash blocks that are presented to the host, and you will NEVER EVER be able to make the drive malfunction).

That's just how it is. There are no "hidden" blocks of data or "special" partitions that can make a drive fail.

And your driver does not send the __only__ thing that could make a drive fail, which are proprietary USB vendor commands (which are the commands that can talk to the proprietary flash drive controller firmware and make it perform advanced operations). It just acts as a filter driver, on top of the SCSI base class disk driver (you can see this plainly in the `.inf` by the way, which should also give you an idea of how Windows does not treat USB disks, and therefore their partitions or attributes any differently than other disks) to change the FIXED/REMOVABLE attribute that is reported to Windows. That's __all__ it does.

There's nothing to notify people, because your "claim" does not hold water.

> The moment you try Rufus and see that Warning that there's another partition you'll be f*cking surprised I guess.

Then you, __or anyone else__, who would really like to see me taken down a peg because, _"surely, I can not be that 100% sure that what @Svetoslav92 claims cannot be true"_, f*cking surprise me, by sending me a log of Rufus, after you've done through this bullshit installation of a mere filter driver, that shows that:
1. Rufus is for some reason seeing another partition when on a disk driven by a filter driver that _should only have toggled the REMOVABLE attribute to FIXED
2. Rufus is unable to complete its operations
3. The drive produces an error

All I am noticing, again, is that instead of sending me a Rufus log to demonstrate what you claim, which you should have __no trouble__ replicating if what you state is true, you're still asking others to jump through hoops to _try_ to back you up, whilst pretending that if they do so, they will surely experience the results that you claim.

It should be oh so simple for you to send me a log to f*cking surprise me and shut my arguments down, yet you're still not doing that because...?

So, once again, if anyone wants to give substance to this, then invest your own time, and come back when you have actual data elements that appear to validate the claim. Because until then, and because it should be obvious to anyone with a modicum of knowledge of the internals of Windows that installing a filter driver does not magically make the Windows Hardware Abstraction Layer go awry and will not suddenly damage a flash drive in a manner where it has to be low level formatted or worse, I will continue to affirm that this claim is total and utter bullshit.

PROVE. ME. WRONG.