pbatard/rufus #2118: Clone data and apps to new OS feature

danp5648, 2022-12-18:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

danp5648, 2022-12-18:
Clone data and apps to new OS feature

pbatard, 2022-12-18:
I'm not planning to add such a feature.

You should be able to find other software that does that, and the prime purpose of Rufus is to create bootable drives from installation images, not to clone existing OSes.

Now, if you want to clone a USB drive, so that you can restore it exactly as it was, you can use the "Save to VHD" feature of Rufus, per: https://github.com/pbatard/rufus/wiki/FAQ#Saving_an_existing_drive_to_VHD

closed 2022-12-18 by pbatard

danp5648, 2022-12-18:
Rufus has something like make a user for windows so I was thinking one step
more and port-in browser apps
because the process now is a pain. Maybe drivers also which is another step
that is done after a new OS is installed.
I was thinking about all steps afterward and how cool that would be. Is not
really cloning, it is infusing all the steps in an
 automated way that people do after a new Install. It would be so cool...
Thanks

pbatard, 2022-12-18:
https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want

danp5648, 2022-12-18:
The link on that page doesn't work.
Why can't you just add the feature I want
<https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want>
does
nothing at least for Opera//

pbatard, 2022-12-18:
Yes I know. GitHub broke anchor links __again__ (not the first time this happened) and it's a pain in the ass to fix those links manually. You can find the paragraph I pointed to near the top.

pbatard, 2022-12-18:
https://github.com/pbatard/rufus/wiki/FAQ#why-cant-you-just-add-the-feature-i-want if you want a link that works..

danp5648, 2022-12-18:
What would happen if you added the feature? Would the earth blow up? Am I
just trying to make things better!
Anyway thanks for the replies. Dan

pbatard, 2022-12-18:
> What would happen if you added the feature?

I'd have to  magically make some time materialise out of nowhere.

For crying out loud, stop assuming that, because the application is free, my time also is free and that I am sitting on my ass, day and and day out, playing games or watching youtube or whatnot while wondering what I should work on next.

Even without taking __any__ outside requests from users, I have a line up of what I'd like to do with Rufus that can easily keep me busy 10 yours a day, 7 days a week, __for years__. I am not kidding.

And Rufus is only one of the many projects I develop for.

So if I were to add the feature you request, I'd have to delay other features, that are __much more beneficial to Rufus users__, just so you can have your cake. That's not going to happen. What you request will __not__ actually benefit as many users as you believe it will, it is __much more complex and time consuming__ to add than you think it is, and it is not actually a blocking element when you can use ninite installer, backup solutions, clone solutions and other stuff. Plus, it is pure [feature creep](https://en.wikipedia.org/wiki/Feature_creep) as far as I am concerned, because it's well outside the mandate of what Rufus is and should be designed for.

> Would the earth blow up?

My earth would blow up, yes, on account of forced sleep deprivation.

> just trying to make things better!

Then invest __your own__ time to do so.

Anybody who says he's trying to improve something, yet actually relies on __others__ to do that job for them (even more so if it's without remunerating them) is 100% lying to themselves. If you were trying to make things better (i.e. if you _actually_ believed that what you propose is worth it) then you'd learn programming if needed, fork the project and add the feature yourself, because that's how it actually works, and that how you actually demonstrate to the world that you do believe that the feature you are proposing is so beneficial as to be worth one's time investment (be it yours or mine).

Open Source is not _"one person doing all the work, while a myriad of folks chime in to give him or her some 'orders' on what to work on next"_. 

If you actually want something from Open Source software, because you believe it will benefit the world that much, then you should either invest your own time into it or shut up.